MarlinFirmware/Marlin #3891: `opt_enable` in Travis is extremely sucsessful

AnHardt, 2016-05-26:
`opt_enable` in Travis is extremely successful in enabling `Z_PROBE_ALLEN_KEY`.
It also activates all options beginning with `Z_PROBE_ALLEN_KEY` from the commented examples.

```
0.02s$ opt_enable AUTO_BED_LEVELING_FEATURE Z_PROBE_ALLEN_KEY
The command "opt_enable AUTO_BED_LEVELING_FEATURE Z_PROBE_ALLEN_KEY" exited with 0.
9.99s$ build_marlin
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Xlib:  extension "RANDR" missing on display ":1.0".
Initializing packages...
Preparing boards...
Verifying...
In file included from sketch/Marlin.h:44:0,
                 from sketch/M100_Free_Mem_Chk.cpp:43:
sketch/Configuration.h:721:0: warning: "Z_PROBE_ALLEN_KEY_DEPLOY_1_X" redefined [enabled by default]
     #define Z_PROBE_ALLEN_KEY_DEPLOY_1_X 35.0
 ^
sketch/Configuration.h:698:0: note: this is the location of the previous definition
     #define Z_PROBE_ALLEN_KEY_DEPLOY_1_X 30.0
 ^
sketch/Configuration.h:722:0: warning: "Z_PROBE_ALLEN_KEY_DEPLOY_1_Y" redefined [enabled by default]
     #define Z_PROBE_ALLEN_KEY_DEPLOY_1_Y 72.0
 ^
sketch/Configuration.h:699:0: note: this is the location of the previous definition
     #define Z_PROBE_ALLEN_KEY_DEPLOY_1_Y DELTA_PRINTABLE_RADIUS
 ^
sketch/Configuration.h:724:0: warning: "Z_PROBE_ALLEN_KEY_DEPLOY_1_FEEDRATE" redefined [enabled by default]
     #define Z_PROBE_ALLEN_KEY_DEPLOY_1_FEEDRATE (HOMING_FEEDRATE_XYZ)/10
 ^
...
```
Just a optical problem - if you know what is going on.

jbrazio, 2016-05-27:
Thanks for spotting.

thinkyhead, 2016-05-27:
Hmm, in order to have the regular expression handle this nicely, I need to grab the next character after the match (`[^a-zA-Z_]`) and include it in the replacement. Hopefully this won't break it:
```bash
for opt in "$@" ; do
  eval "sed -i 's/\/\/ *\(#define *$opt[^a-zA-Z_]\)/\1/g' Marlin/Configuration.h"
done
```

jbrazio, 2016-05-30:
Fixed by #3906.

closed 2016-05-30 by jbrazio