MarlinFirmware/Marlin #3892: Auto Bed Leveling - Z Height cannot be adjusted after probing

psavva, 2016-05-26:
I have been trying for the last several hours to level my bed using RCBugFix (latest as of a couple of hours ago)
Configuration File:
http://pastebin.com/kig5cBGL

After homing, I will do Auto Level.
After Auto Level, the Z height is not correct. It's too high.
I have to adjust several options when trying to set the Z offset.

I'm using the BLTouch probe.

1)    #define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]
I tried set this to values -1, 0 and even 1, which none of them made any difference to the height after probing.

2) #define Z_MIN_POS 0
I set this to values -1.
When setting Z MIN POS, it will actually allow me to move the Z height with the dial to "-1".
You'd expect that after homing, Zero should be your minimum position possible.

I also tried to set this on the Z Offset on the LCD, which again, after Auto Leveling, it was again "lost".
btw. which value in the firmware does Z Offset correspond to in the configuration file?  I couldn't find it...

Any help would be appreciated

jbrazio, 2016-05-27:
How are you configuring the ABL ?

thinkyhead, 2016-05-27:
`Z_PROBE_OFFSET_FROM_EXTRUDER` should be set to the distance between the bed and the nozzle at the time the probe triggers. So if your nozzle is 2.5mm above the bed when the probe gets triggered, you would set `Z_PROBE_OFFSET_FROM_EXTRUDER` to `-2.5`.

`Z_MIN_POS` sets the permanent physical lower-Z limit for your machine, and also applies to the Z "software endstop." If you don't need to move below the bed (where Z=0), then you can just leave it set to `0`.

psavva, 2016-05-27:
Thank you @thinkyhead, I will try to do this again tonight.
Based on your response, we should probably mark this request as support or invalid.

I'll test it again and report back.

psavva, 2016-05-27:
@jbrazio, my config file is attached.

psavva, 2016-05-28:
Setting Z_PROBE_OFFSET_FROM_EXTRUDER did resolve the problem.

closed 2016-05-28 by psavva

mmakam2, 2018-07-18:
How did it resovle your problem? I am having the same issue, I have tried literally everything, cannot set the z offset with my EZABL sensor.