MarlinFirmware/Marlin #3893: missing Thermometer symbol

bstott, 2016-05-26:
Problem:
Somehow the thermometer symbol for the text lcd is changed/lost. It is showing as "dy." in the 20162404 release and as "y." in the 20161604 release.

I did not see how to fix.

Firmware version 1.1.0-RC6.0
LCD: Gadgets3D 
Settings selected for the installed LCD:
#define DISPLAY_CHARSET_HD44780_JAPAN        // this is the most common hardware
#define ULTRA_LCD   // Character based
#define G3D_PANEL

Thanks!

thinkyhead, 2016-05-26:
I haven't seen this problem with my `G3D_PANEL`. Do you see the same thing if you use the current `RCBugFix` code?

CleoDarius, 2016-05-27:
Hi,
Easy fix. The "dy" is from the last line on your display which is too long. Make sure that your CUSTOM_MACHINE_NAME is not too long since "ready" is added on and sent to the display. Anything over 20 characters wraps around to the first line.

bstott, 2016-05-27:
@CleoDarius  --- Ha, ha... Give the Man a Prize! Too long of a name. That was the solution. No Bug.
 Thanks! 
And to thinkyhead - I'll load the RCBugFix since I now have it.
Cheers.

closed 2016-05-27 by bstott