MarlinFirmware/Marlin #3896: MBL: Mesh area must lay in positive X&Y quadrant.

epatel, 2016-05-27:
First noticed and mentioned in this discussion by @psavva

I will look into this this weekend.

boelle, 2016-05-27:
what if people define center of bed at 0,0 ?

epatel, 2016-05-27:
@boelle Good question. That's why I will look at it for MBL. The rest of Marlin I have no clue about and I will not verify. I consider this a problem similar to the gimbal-lock problem in 3D graphics/simulators. If you are not sure you should avoid doing certain things.

epatel, 2016-05-27:
@boelle By looking at all example configurations all use zero except the delta printers, and MBL hasn't been available for deltas, yet. @Roxy-3DPrintBoard

boelle, 2016-05-27:
yes, but if i define 0,0 as the center of my bed (200*200 mk2a) i will have -100 in one end and +100 at the other end.

boelle, 2016-05-27:
// Travel limits after homing (units are in mm)
#define X_MIN_POS -100
#define Y_MIN_POS -100
#define Z_MIN_POS 0
#define X_MAX_POS +100
#define Y_MAX_POS +100
#define Z_MAX_POS 200

epatel, 2016-05-27:
@boelle Yes, I understand. That is why I will be looking at this (for MBL). Deltas has a totally different coordinate system and path from "model/g-code space" to "printer/stepper space"...and it is very well tested so I believe that should be safe. I would not say the same for "Cartesian" or "normal" coordinate systems printers as I don't believe so many actually use it like that, or? example_configurations don't indicate that. The errors could be small so you do not actually notice them and could perhaps think it was caused by something else, slicer, print speed, filament etc

boelle, 2016-05-27:
i like 0,0 in the center for the reason that it automatic keeps most part where the bed temp is most accurate.... actual temps are lower the further away from the center

epatel, 2016-05-27:
@boelle But you are not using MBL I guess. So are you printing outside the positive quadrant? (with the positive quadrant I mean when the X>0 and the Y>0) Maybe even crossing the zero axes? If it works great!

boelle, 2016-05-27:
i will when i get the last bits for my printer. mine was made in MDF and i have rebuilt it in dibond sheets.

and yes with 0 in the center i will print in both positive and negative

jbrazio, 2016-05-27:
@epatel you didn't link the discussion.
I'm marking this as a possible bug until further notice how we should proceed.

epatel, 2016-05-27:
@jbrazio Oh, I missed that. Pretty sure I copied the link...must have missed pasting it in. Will do.

WheresWaldo, 2016-05-29:
Sorry to jump in on this discussion so late, but even though I understand what @boelle is doing, I can't see why this is a firmware issue and not a slicer issue! 

By your own statement you use this to keep your model centered on the bed
> i like 0,0 in the center for the reason that it automatic keeps most part where the bed temp is most accurate.... actual temps are lower the further away from the center

Can you explain your rationale for needing this in firmware and not commanding your slicer to put your model in the middle of the bed?

thinkyhead, 2016-05-30:
Merged #3903 which is a smart fix for the issue.

Alex9779, 2016-05-30:
On RCBugFix #3903 was recently merged to fix this but I have a problem with this.

I have a dual nozzle setup in Y direction on one carriage and the printer is designed so that every nozzle can move out of the print area. My right nozzle is defined as extruder 0, left is extruder 1.
Extruder 1 has a negative offset from extruder 0.
When I am at X0 the right nozzle is at X0 of the bed, left nozzle is left of the bed at about X-38.

Since Marlin does not make a difference between print area and move area I have to define my X max to the max the print head can move but when it is at X max the right nozzle is not above the bed.
So I have to set X max to 338 to get the left nozzle to X300.

In current RC i defined my mesh min max points as follows:
```
#define MESH_MIN_X 15
#define MESH_MAX_X (285)
#define MESH_MIN_Y 15
#define MESH_MAX_Y (185)
```

With RCBugFix this is not possible for me with hacking a non-configuration file because with
```
#define X_MAX_POS 338
```
and
```
#define MESH_INSET 15
```
my left spot will be X323 where my right nozzle is not above the bed.

This is a problem because I don't wanna level the bed on the one side to the left and on the other side to the right nozzle because when I level the bed I have the nozzle not always levels to each other to the bed level would be useless.

And maybe some can explain the purpose of this fix because I don't get it why this should move the limit to positive values if I set the X min value to a negative value. A greater negative value plus a little inset is still a negative value...

Alex9779, 2016-05-30:
I just realised that I have a problem with the Y axis too.
I made a little drawing to show what the dimension of my printer are:
[image]
Black frame is the bed, the coloured frames are the limits for the nozzles, all coordinates are based on the bed.
So you see there is the so called "dock" in the back of the machine.
My Y max is actually 240 for that certain area and Marlin has:
```
#define X_MAX_POS 338
#define Y_MAX_POS 235
```
while the bed is only 300x200.

epatel, 2016-05-30:
@Alex9779 Interesting! Happy to see that you got MBL working for you, I'll see if I can tweak it to your bidding. Did not see this coming :) I was mainly focusing on ease of use...

epatel, 2016-05-30:
@Alex9779 What would you think of putting the `MESH_MIN/MAX_X/Y` in `Configuration_adv.h`? Sounds reasonable for your use case?

Alex9779, 2016-05-30:
@epatel that would be great.
Don't misunderstand me, I would have no problem editing the mesh_bed_leveling.h directly but like I described I wanna stay on standard as much as possible.
I thought about doing a PR myself but then I thought is will be better to bring this up here than fighting with #if and stuff.
If you wanna make Configuration.h easier ok but make it possible to stay within the standard config files to make specific adjustments.

Still,not related to my issue, I do not understand how this should help to move the mesh area to positive coordinates...

epatel, 2016-05-30:
@Alex9779 I totally understand you.

> Still,not related to my issue, I do not understand how this should help to move the mesh area to positive coordinates.

I think the opposite was happening, some were afraid of editing the macros (like you did) and i.e. `MESH_MIN_X` was a fixed number not related to `X_MIN_POS`, and then `MESH_MAX_X` mixing that up. Old sins. Tried to move that away and clean it up, which then hid the part where you could do you thing.

```
#define MESH_MIN_X 10
#define MESH_MAX_X (X_MAX_POS - (MESH_MIN_X))
```

But, I'll look into moving them to `Configuration_adv.h`

Alex9779, 2016-05-30:
👍

epatel, 2016-05-30:
@Alex9779 See this https://github.com/MarlinFirmware/Marlin/pull/3918

psavva, 2016-05-30:
One last thing to mention :) 
Documentation probably needs to be updated to allow people to understand how to configure the Mesh Bed Leveling, considering the 3 cases we saw here.

1) Cartesian Printer, where travel goes into the negative quadrant
2) Dual Nozzle Setup - Advanced Configuration possibility 
3) Delta Printer - 0,0 is bed center

We would probably need to document the meaning of the "MESH_INSET" and how to set it.

jbrazio, 2016-05-31:
@psavva [will you contribute](https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_articles/howto/bed-auto-leveling.md) ?

psavva, 2016-05-31:
@jbrazio, I would love to.

thinkyhead, 2016-06-01:
Aside from the documentation part, this seems to be resolved. Please re-open if anything is still amiss.

closed 2016-06-01 by thinkyhead