MarlinFirmware/Marlin #3897: LCD menu homing and manual moves

maukcc, 2016-05-27:
A few years ago we threw homing and manual moves out of the LCD menu as it is totally useless and you could manually move after boot (with no homing) and that would move your axis beyond it scopes.
So the only manual move I left in the menu was the extruder(s)
Lately we got questions from customers to re-implement it because other printers do it as well (great reason to do anything. If they jump of a cliff, .....). Never the less it got me thinking on how to do this.
1st I thought was: lets make a homing at boot, but that does not comply with CE certification.
I now have a manual extrude menu high up in the prepare menu(as this is relevant), and a homing menu at the bottom (as I still think it is useless) 
The homing menu homes the X,Y and Z axis first and then presents you with a menu to manually move and/or set the origin.
This way you will always have a homing done before you can manually move

below my prepare menu ( I am not a coder so it can probably be more elegant)
And of coarse the language file needs adapting, and a couple of 'static void' add ons to the top of the file.

static void lcd_prepare_menu() {
  START_MENU();
  MENU_ITEM(back, MSG_MAIN, lcd_main_menu);
  #if defined( SDSUPPORT ) && defined( MENU_ADDAUTOSTART )
    MENU_ITEM(function, MSG_AUTOSTART, lcd_autostart_sd);
  #endif
  MENU_MULTIPLIER_ITEM_EDIT(int3, MSG_FAN_SPEED, &fanSpeed, 0, 255);
  MENU_ITEM(function, MSG_PREHEAT_PLA_N MSG_H1, lcd_preheat_pla0);
  MENU_ITEM(function, MSG_PREHEAT_PLA_N MSG_H2, lcd_preheat_pla1);
  MENU_ITEM(submenu, MSG_MOVE_EXTR, lcd_move_extruder_menu);
  MENU_ITEM(function, MSG_COOLDOWN, lcd_cooldown);
  MENU_ITEM(gcode, MSG_DISABLE_STEPPERS, PSTR("M84"));
  MENU_ITEM(function, MSG_AUTO_HOME, homing_axis); 

  #if HAS_POWER_SWITCH
    if (powersupply)
      MENU_ITEM(gcode, MSG_SWITCH_PS_OFF, PSTR("M81"));
    else
      MENU_ITEM(gcode, MSG_SWITCH_PS_ON, PSTR("M80"));
  #endif

  #if defined(MANUAL_BED_LEVELING)
    MENU_ITEM(submenu, MSG_LEVEL_BED, lcd_level_bed);
  #endif
	
  END_MENU();
}

#ifdef DELTA_CALIBRATION_MENU
static void lcd_delta_calibrate_menu()
{
    START_MENU();
    MENU_ITEM(back, MSG_MAIN, lcd_main_menu);
    MENU_ITEM(gcode, MSG_AUTO_HOME, PSTR("G28"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_X, PSTR("G0 F8000 X-77.94 Y-45 Z0"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_Y, PSTR("G0 F8000 X77.94 Y-45 Z0"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_Z, PSTR("G0 F8000 X0 Y90 Z0"));
    MENU_ITEM(gcode, MSG_DELTA_CALIBRATE_CENTER, PSTR("G0 F8000 X0 Y0 Z0"));
    END_MENU();
}
#endif // DELTA_CALIBRATION_MENU

float move_menu_scale;
static void lcd_move_menu_axis();

static void _lcd_move(const char *name, int axis, int min, int max) {
  if (encoderPosition != 0) {
    refresh_cmd_timeout();
    current_position[axis] += float((int)encoderPosition) * move_menu_scale;
    if (min_software_endstops && current_position[axis] < min) current_position[axis] = min;
    if (max_software_endstops && current_position[axis] > max) current_position[axis] = max;
    encoderPosition = 0;
    #ifdef DELTA
      calculate_delta(current_position);
      plan_buffer_line(delta[X_AXIS], delta[Y_AXIS], delta[Z_AXIS], current_position[E_AXIS], manual_feedrate[axis]/60, active_extruder);
    #else
      plan_buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS], manual_feedrate[axis]/60, active_extruder);
    #endif
    lcdDrawUpdate = 1;
  }
  if (lcdDrawUpdate) lcd_implementation_drawedit(name, ftostr31(current_position[axis]));
  if (LCD_CLICKED) lcd_goto_menu(lcd_move_menu_axis);
}
static void lcd_move_x() { _lcd_move(PSTR("X"), X_AXIS, X_MIN_POS, X_MAX_POS); }
static void lcd_move_y() { _lcd_move(PSTR("Y"), Y_AXIS, Y_MIN_POS, Y_MAX_POS); }
static void lcd_move_z() { _lcd_move(PSTR("Z"), Z_AXIS, Z_MIN_POS, Z_MAX_POS); }

static void lcd_move_e() {
  if (encoderPosition != 0) {
    current_position[E_AXIS] += float((int)encoderPosition) * move_menu_scale;
    encoderPosition = 0;
    #ifdef DELTA
      calculate_delta(current_position);
      plan_buffer_line(delta[X_AXIS], delta[Y_AXIS], delta[Z_AXIS], current_position[E_AXIS], manual_feedrate[E_AXIS]/60, active_extruder);
    #else
      plan_buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS], manual_feedrate[E_AXIS]/60, active_extruder);
    #endif
    lcdDrawUpdate = 1;
  }
  if (lcdDrawUpdate) lcd_implementation_drawedit(PSTR("Extruder"), ftostr31(current_position[E_AXIS]));
  if (LCD_CLICKED) lcd_goto_menu(lcd_move_extruder_menu);
}

static void lcd_move_menu_axis() {
  START_MENU();
  MENU_ITEM(back, MSG_MOVE_AXIS, lcd_move_menu);
  MENU_ITEM(submenu, MSG_MOVE_X, lcd_move_x);
  MENU_ITEM(submenu, MSG_MOVE_Y, lcd_move_y);
  MENU_ITEM(submenu, MSG_MOVE_Z, lcd_move_z);
  if (move_menu_scale < 10.0) {  // Can not delete this
    //MENU_ITEM(submenu, MSG_MOVE_Z, lcd_move_z);
    //MENU_ITEM(submenu, MSG_MOVE_E, lcd_move_e);
  }
  END_MENU();
}

// for extruder
static void lcd_move_menu_extruder_10mm() {
  move_menu_scale = 10.0;
  lcd_move_e(); // lcd_move_menu_axis();
}
static void lcd_move_menu_extruder_1mm() {
  move_menu_scale = 1.0;
  lcd_move_e();
}

//for axis
static void lcd_move_menu_10mm() {
  move_menu_scale = 10.0;
  lcd_move_menu_axis(); // lcd_move_menu_axis();
}
static void lcd_move_menu_1mm() {
  move_menu_scale = 1.0;
  lcd_move_menu_axis();
}
static void lcd_move_menu_01mm() {
  move_menu_scale = 0.1;
  lcd_move_menu_axis();
}

// homing sequence preferably in configuration.h
static void homing_axis(){
  enquecommand("G28 Y");
  enquecommand("G28 X");
  enquecommand("G28 Z");
  enquecommand("G1 Z10 F1200");               // move Z down so you will not bump into anything
  lcd_goto_menu(lcd_homing_menu , 0, false);  // go to homing menu
}

static void lcd_homing_menu() {
  START_MENU();
  MENU_ITEM(back, MSG_MAIN, lcd_main_menu);
  MENU_ITEM(submenu, MSG_MOVE_AXIS, lcd_move_menu);
  MENU_ITEM(function, MSG_AUTO_HOME, homing_axis);
  //MENU_ITEM(function, MSG_SET_HOME_OFFSETS, lcd_set_home_offsets);
  MENU_ITEM(gcode, MSG_SET_ORIGIN, PSTR("G92 X0 Y0 Z0"));
  END_MENU();
}

static void lcd_move_menu() {
  START_MENU();
  MENU_ITEM(back, MSG_BACK, lcd_homing_menu);
  MENU_ITEM(submenu, MSG_MOVE_10MM, lcd_move_menu_10mm);
  MENU_ITEM(submenu, MSG_MOVE_1MM, lcd_move_menu_1mm);
  MENU_ITEM(submenu, MSG_MOVE_01MM, lcd_move_menu_01mm);
  END_MENU();
}

static void lcd_move_extruder_menu() {
  START_MENU();
  MENU_ITEM(back, MSG_PREPARE, lcd_prepare_menu);
  MENU_ITEM(submenu, MSG_MOVE_10MM, lcd_move_menu_extruder_10mm);
  MENU_ITEM(submenu, MSG_MOVE_1MM, lcd_move_menu_extruder_1mm);
 // MENU_ITEM(submenu, MSG_MOVE_01MM, lcd_move_menu_01mm);
  MENU_ITEM(gcode, MSG_EXT_0, PSTR("T0"));
  MENU_ITEM(gcode, MSG_EXT_1, PSTR("T1"));
  MENU_ITEM(gcode, MSG_EXT_2, PSTR("T2"));
  MENU_ITEM(gcode, MSG_EXT_3, PSTR("T3"));
  //TODO:X,Y,Z,E
  END_MENU();
}

maukcc, 2016-05-27:
Here is also my language file

[language.txt](https://github.com/MarlinFirmware/Marlin/files/286067/language.txt)

maukcc, 2016-05-31:
This can be closed as it is fixed in RC6
apart from my personal view that extruding and moving axis are 2 different things / menu items.

closed 2016-05-31 by jbrazio

thinkyhead, 2016-05-31:
We're always open to new ideas to make the LCD menus better. It's always a little tricky to know what to put into those privileged 3 or 4 lines at the top of the menu, or what to make into sub-menus.