MarlinFirmware/Marlin #3898: M Codes

chevellebro1, 2016-05-27:
I have created a custom M Code for laser on/off. When I use the terminal, the codes work great. However when I run it in gcode, the laser turns on for a split second and then off. I have to manually send the code again to turn it on. And then when it reaches the end, the code to turn it off doesn't work. I have to again manually enter that code. Any idea what would cause this?

jbrazio, 2016-05-27:
You should post a bit of code so we can help you out.

chevellebro1, 2016-05-27:
```
case 10: // M10 - Laser On
       pinMode(LASER_BEAM, OUTPUT);
       digitalWrite(LASER_BEAM, HIGH);
      break;

case 11: // M11 - Laser Off
       pinMode(LASER_BEAM, OUTPUT);
       digitalWrite(LASER_BEAM, LOW);
      break;
```
Here are the M Codes that are controlling laser on and off. As I said when I manually send M10 the laser stays on but when I send it in a gcode file it flashes on then back off. Here is a simple test file I've been using to cut a square.

```
(PROGRAM NAME: CALIBRATION.GCODE)

(LASER)
G21
M62
G0 X0 Y0 Z0 F250

M10

G1 X10 F150
G1 Y10
G1 X0
G1 Y0

M11
```

jbrazio, 2016-05-27:
Take this `pinMode(LASER_BEAM, OUTPUT);` out of the loop, find the `setup()` function and put it there. Are you sure that `LASER_BEAM` PIN is not used elsewhere ?

chevellebro1, 2016-05-27:
I moved the `pinMode` to the `setup()` function. I checked the pin.h file and pin 58 is not used anywhere else besides the `LASER_BEAM` function.

jbrazio, 2016-05-27:
May I suggest the following [existing] `MCode` ?

```cpp
/**
 * M42: Change pin status via GCode
 *
 *  P<pin>  Pin number (LED if omitted)
 *  S<byte> Pin status from 0 - 255
 */
```

If the PIN is a digital pin do
- `M42 P58 S1` for ON
-  `M42 P58 S0` for OFF

If the PIN is an analog pin do
- `M42 P58 S255` for ON
-  `M42 P58 S0` for OFF

chevellebro1, 2016-05-27:
I'm using the Ramps 1.4 board so according to the pinout diagram it is both analog and digital so which would I use? It's `Digital 58` or `Analog 4`

chevellebro1, 2016-05-27:
I just tried this code with the same result. The laser flashes on briefly before turning back off

chevellebro1, 2016-05-27:
So for some reason Marlin is reading ahead for the code to turn the laser off. I removed the `M11` command from the file and the laser stays on. But for some reason it is reading ahead and turning the laser off immediately after turning on.

jbrazio, 2016-05-27:
Can you try a G-Code file with more command between the ON and OFF function like example:
```
; go to a know pos
G1 X0 Y0 F0
; fire it up scotty
M42 P58 S1
; do a square
G1 Y10
G1 X10
G1 Y0
G1 X0
; redo the square
G1 Y10
G1 X10
G1 Y0
G1 X0
; turn off laser
M42 P58 S1
```

When does the laser get off ?

chevellebro1, 2016-05-27:
Still turns off immediately after turning on.

bobc, 2016-05-27:
I thought G codes were placed in planner queue bu M codes were actioned immediately? Is there a way to wait for queue to empty?

jbrazio, 2016-05-27:
That's the reason why I was tried to fill the command buffer queue.

jbrazio, 2016-05-27:
But for a command like `M42` this should execute in sync with the G-Code.. otherwise is useless.

chevellebro1, 2016-05-27:
I tried another piece of code without `M11` and it works but it is not staying in sync with the G-Code. 
```
G21
G28 Z ; home Z
G0 X0 Y0 Z119.00 F750 ; move to laser focal point
M42 P58 S255 ; laser on
G1 X10 F500 ; square
G1 Y10
G1 X0
G1 Y0
```

The laser turned on in the middle of the `G0 X0 Y0 Z119` command before it had finished the movement to the given Z depth.

bobc, 2016-05-27:
Ah, here it is "M400 - wait for all moves to finish".

chevellebro1, 2016-05-27:
So would I need to add that `M400` at the end of every movement before I want the laser to turn off?

bobc, 2016-05-27:
Yes, so 
```
(PROGRAM NAME: CALIBRATION.GCODE)

(LASER)
G21
M62
G0 X0 Y0 Z0 F250

M10

G1 X10 F150
G1 Y10
G1 X0
G1 Y0
M400 (wait for sync)
M11
```

jbrazio, 2016-05-27:
> Ah, here it is "M400 - wait for all moves to finish".

I was unaware of this, great tip.

chevellebro1, 2016-05-27:
Works great! Thank for the help

thinkyhead, 2016-05-30:
To explain, the G1 command only tells the planner to queue the movement. The movement will start later. Meanwhile, commands to turn the laser on and off are executed right away. So M400 is needed to wait for the prior move to complete before toggling the laser.

closed 2016-05-30 by jbrazio