MarlinFirmware/Marlin #3900: Collaborators:  Please Read!

Roxy-3DPrintBoard, 2016-05-28:
Dear Collaborators,

We have had a serious security breach.  We have had our repositories damaged.   Our teams have been deleted and members of those teams removed.   We have also had branches of our trees deleted.  We are in the process of trying to recover right now.

We need to lock down access to the repositories until we can sort through things.   The good news is RC6 and RCBugFix are unaffected and you can use them with confidence.

If any Outside Collaborator needs a change merged please notify us and we will try to prioritize that so we don't affect your project's activity or schedules.

Thank You!

boelle, 2016-05-28:
I got a shock when I got the removed msg

ghumphr1, 2016-05-28:
hi, can you remember the thread about linux and rtos, could you post the link for 'https://xenomai.org/' as the lock down has locked me out...Thanks ghumphr1

Roxy-3DPrintBoard, 2016-05-28:
That thread was in the MarlinDev repository.   It should exist in the regular Marlin repository now.  Please try searching there.

CONSULitAS, 2016-05-28:
@Roxy-3DPrintBoard 
Hi Roxy,
thanks for this shocking information!

It seems to me, that i have lost my collaborator status. 😢 

Is there a way to find out - perhaps via the abuse button - what has happened in detail and perhaps get help by github to get a backup of the last sane state?

Greetings Jochen

CONSULitAS, 2016-05-28:
Perhaps we should name people we have in mind als collaborators to make sure that they receive the message:

@thinkyhead 
@jbrazio 
@Wurstnase 
@fmalpartida 
@epatel 

Anybody wrong or missing?

Roxy-3DPrintBoard, 2016-05-28:
@CONSULitAS  The Collaborator team is one of the teams that got waxed.  But you still have Write access to the Marlin Repository.

CONSULitAS, 2016-05-28:
@Roxy-3DPrintBoard 
Thanks for the feedback.

I thought so, but better inform one time to often...

Please do not hurry. It is better to be careful and make sure that the root cause can be eliminated first.

ErikZalm, 2016-05-28:
Can you give me full control over the MarlinFirmware repo?
Also add Daid. We should always have full control over this repo.

Roxy-3DPrintBoard, 2016-05-28:
@ErikZalm   Would it be possible to exchange email address.  It would be helpful to exchange some information.

Roxy-3DPrintBoard, 2016-05-28:
@ErikZalm Got it!  Trying to get a Google Chat initiated.

Grogyan, 2016-05-29:
I too got an email about my Collaboration being cancelled.  I really have no idea what that even means.

boelle, 2016-05-29:
cant help to think of this as the use of an security exploit to get rid of those people that are unwanted

Erik Zalm could have been added just with the use of his github username... Why does @Roxy-3DPrintBoard insist on keep communication off line so others cant see what goes on?

*snif* *snif* I smell BS and big ego's

Roxy-3DPrintBoard, 2016-05-29:
@Boelle The users are here with issues.  They don't need to see all the dirty laundry.

thinkyhead, 2016-05-30:
I hope things are getting sorted out. A lot of confusion around these changes. I don't think I was affected at any point. I can still do the things I need to do.

boelle, 2016-05-30:
well i was totally waxed out and have not been added back

boelle, 2016-05-30:
I was totally waxed out and not added back

and someone is deleting my posts.... Stop doing that... i can be contacted on hangouts if need be

boelle, 2016-05-30:
i was waxed too and not added back in

@ErikZalm has been notified that someone is deleting my posts, and he have made note of who in the log. 

VERY sad way of dealing with things

jbrazio, 2016-05-30:
@boelle It's me, please speak with Roxy.

PS: It's not "your" posts.

boelle, 2016-05-30:
roxy does not speak to me... not on email and not on hangouts

jbrazio, 2016-05-30:
She had some issues do to local floods with power/internet outage, I believe since yesterday things are stable. Just to ping her: @Roxy-3DPrintBoard.

ErikZalm, 2016-05-30:
Please stop bullying each other and respect other contributors.

If this does not resolve I have no other option then take back full control.
I do not want this because I do not have the time to handle all posts and pull-requests.

boelle, 2016-05-30:
@Roxy-3DPrintBoard 

open hangouts.... i have sent an invite for chat... i did so many weeks ago and yet you have not spoken to me

boelle, 2016-05-30:
@ErikZalm 

I have a plan that could fix this....

1) you and daid are the only one with admin role
2) so 2 teams need to be created... Admin/core with you and Daid in it with role set to admin for all 3 repros.. The other team could be called development and have scott in it with write role and promoted to team maintainer to all 3 repro's
3)everyone else will be outside collab... scott will then merge PR's

jbrazio, 2016-05-30:
@ErikZalm the only missing part is to @boelle to speak to @Roxy-3DPrintBoard, but I already explained the issues she was having. So this should be fixed soon.

boelle, 2016-05-30:
well part of the issue is also that more than really needed have to much control

boelle, 2016-05-30:
i could make do as outside collab.... you would then ask why i keep going on

because not everyone that was inside got added back in.. that can only say that a selected few are interested in having power

if that is not the case please tell me without deleting anything.

jbrazio, 2016-05-30:
I deleted only your posts which were not part of Marlin's support, all your travis related posts are there as an example. I don't want to lock this thread, so please be patient and speak with Roxy, she will answer all your questions and the outcome will be different from what you think, but you **need** to speak first with her.

boelle, 2016-05-30:
oki... will wait... email and hangout are open all the time

boelle, 2016-05-30:
btw... do you know why scott have not replied to hangouts or sms on the phone?

Blue-Marlin, 2016-05-30:
@boelle 
There is a high possibility mail servers currently do not accept any further mails from you.
At the 25.5.2016 13:22 - 14:55 (93 minutes) our mail server received 1188 mails with your name associated. Two of our automatic security systems triggered. The spam filter and the warning for a 'denial of service attack'. It might last some time until your spam factor shrinked a bit, so you are received from everybody again.

boelle, 2016-05-30:
looool....  that is not me at least

boelle, 2016-05-30:
could you in same way share info on this?

since you have mail server access you should be able to write a direct email to me ?

boelle, 2016-05-30:
@Blue-Marlin written you a mail... so you can compare where they originate from

what timezone is the date and time you mentioned?

Roxy-3DPrintBoard, 2016-05-30:
@boelle   @ErikZalm When I went to bed last night I was in the middle of trying to organize an email exchange so we can all work together and discuss Marlin's near term objectives.   Please give me a few hours to get that written and out to everybody.

Specifically, the open issues (at least in my mind) are two-fold.   What can we do to accelerate the Golden Release (and do we want to do that)?    And then the second part is we know we want to use that as the base for the development branches.  The more I think about it, we can't have 3 open development branches because the clean up topic is going to interfere with the other two areas.   We might be forced to prioritize different development branches over each other.   

The clean up branch is going to be a big effort, but it is also the one that has the most interest from the users.  There will be simplification that can be done using things like Z-Probe Objects and Extruder objects.   But most of the clean up will come from splitting the code up and organizing it in folders.   We will need to recruit some people to help manage and guide this effort.  The good news is there are lots of programmer types here that are very interested in the topic.  It maybe I'm worrying too much about the complexity of this topic.   

Please give me a few hours to get the first draft out and we can start trying to discuss in greater depth very soon.

boelle, 2016-05-30:
@Roxy-3DPrintBoard could you open hangouts ??

boelle, 2016-05-30:
there should be an invite from me there

Roxy-3DPrintBoard, 2016-05-30:
I can in a few minutes.  But I believe Joao is trying to invite you to the organization.  Maybe that will be done by the time I can open the chat.

thinkyhead, 2016-05-31:
> do you know why scott have not replied to hangouts or sms on the phone?

Not very much in the mood right now.  I have other fish to fry. I'm not quite understanding Bo's credentials. @boelle You seem to think you have an important role to play. Can you describe your background and skills, and what you hope to bring to the project?

Roxy-3DPrintBoard, 2016-05-31:
It would also be good and helpful to know what types of things bring you satisfaction.  For me, I like putting new foundational pieces in place that extend the architecture and can be built upon by later pieces of code.  ThinkyHead likes dealing with bug fixes and all of the very necessary 'day to day' stuff.   Joao seems to like fillling holes that nobody else is doing.  

If you can help us understand what types of contributions bring you satisfaction, we can custom design a role for you.  In the first email of the discussion we have going I identified at least 5 leadership roles that we would love to have somebody say "That's me!  I'll take that one!"    

But really...  This discussion is better handled off line.

closed 2016-06-15 by jbrazio