MarlinFirmware/Marlin #3902: Probe Servo Triggers During Print?

timeshell, 2016-05-29:
I just upgraded to 1.1.0-RC6 and the first two prints I had about an hour or so in seems could be triggering an event on the Z probe servo.  I have not had this issue before so I'm looking for guidance on what is causing this.  The probe is a BLTouch and aprox an hour into the print it seems to be getting triggered and it goes into an error condition.  The probe should just remain in its up position during the print, but the pin is caused to drop.  I don't know if it is the probe itself or 1.1.0-RC6 is somehow triggering the error condition by triggering the servo.

Grogyan, 2016-05-30:
The BL Touch does not consume current to keep the pin retracted, ie Red LED stays on
 I would suggest looking at other reasons as to why the probe goes into an error state, maybe because there is a MaxTemp trigger?  This is what has happened to me, and have fixed the problem, where it was the thermocouple reading through the SPI that was getting the wrong reading.

Can you please elaborate on the problem, and also please include the configuration.h and configuration_adv.h files

Sniffle, 2016-05-31:
BLTouch should not use the servo deactivation delay, this can cause the probe to run into issues.  I've got 2 printers running with it and the only issue i have had is the pin needed to be removed and cleaned after close to 400 hours of printing.

thinkyhead, 2016-06-01:
Perhaps we should add a new probe type (ulp!) named `BLTOUCH` and if it is selected, disable the `SERVO_DEACTIVATION_DELAY` for you. Otherwise it will be hard for users to discover this. I assume, @Sniffle, that users should still leave `DEACTIVATE_SERVOS_AFTER_MOVE` enabled…?

timeshell, 2016-06-01:
@Grogyan why in the world would a error condition on the BLTouch be related to a MaxTemp issue?  The BLTouch blinks its own LED when it is in error.  This should be completely unrelated to MaxTemp.

@thinkyhead I like that idea.  I'd personally think the BLTouch could become a big standard for 3D probing and deserves its own set of instructions.  It could automagically then set the servo codes 10, 90 and add menu item for resetting the probe, and other related specifics for the probe.

Grogyan, 2016-06-01:
@timeshell I was getting clarification from you how the BL Touch is going into an error state.

BL touch could very well be the new defacto standard in the future when it comes to Bed Probing, over capacitive.
Not sure how you would encompass that sensor over others with simple defines in the configuration.h file. Perhaps this is something to look at going forward as there are so many different probes, it gets confusing to the novice what they need to set for their particular probe.

I haven't need to touch anything in the conditionals.h relating to the BL touch

Sniffle, 2016-06-01:
@thinkyhead BLTouch probes should not use the servo deactivation delay or
whatever it is called now that sends the probe to a low state to prevent
twitching.

Bltouch uses an attiny(i think) chip to interpret servo pulses and a hall
effect sensor to have the chip send a swich press.  The pin is moved via an
electromagnet that is switched on and off via the chip.  By sending the
probe to a low state you are effectively depowering the chip and making it
lose its position.  Should it randomly get a power pulse it will do its
default power on routine which is to cycle the pin up/down 2 or 3 times
before storing the pin.  So low or no power is a bad idea for the bltouch.

That being said the interpretted angles by the chip and their actions are
as follows.

10 - deploy probe light is off
90 - retract probe light is on(deploy/retract could be swapped im pulling
from memory, its set as 10,90 in config)
120 - self test as far as i know cycles the pin endlessly until another
command is given
160 - reset error state. If the bltouch is ever in a state that it knows it
shouldnt be in. Ie the pin gets stuck and cant deploy the bltouch blinks
until the problem is fixed or the error is cleared using the above angle
with m280

Commands are give to the bltouch using standard servo commands.  Ie M280 Px
Sxxx

timeshell, 2016-06-01:
@Sniffle According to an email I received from ANTCLABS, this is the option that should be enabled for the BLTouch. 

DEACTIVATE_SERVOS_AFTER_MOVE

Sniffle, 2016-06-01:
I had issues when i first set mine up until i disabled that.  But, if thats
what antclabs says go for it... Ill leave mine disabled as the only issue
ive had since disabling it is a dirty pin that needed to be cleaned.

thinkyhead, 2016-06-01:
> should also test if the probe is enabled before doing a Z probe

Well… if Marlin is explicitly _enabling and disabling_ it, then we should already "know" when the probe is enabled.

timeshell, 2016-06-01:
In theory.  However, when I first started playing with the disable features I seem to recall something wasn't working properly and I ended up get crashes.  Since the BLTouch is able to report its pin position any way by a servo command, is an extra sanity check going to hurt?  At worst it's a few extra lines of code.  At best it can save someone some money.

thinkyhead, 2016-06-01:
> is an extra sanity check going to hurt?  

So… After we _enable_ it, ask it if it's _really_ enabled. Keep trying to enable it a few times, and fail if it never enabled. After disabling it, ask it if it's _really_ disabled, and keep trying until it works.

timeshell, 2016-06-01:
Before probing, ask it its current state, regardless if it has been enabled or disabled or whatever.

thinkyhead, 2016-06-01:
> ask it its current state

I see. So you're saying it might enable or disable itself without our knowledge.

timeshell, 2016-06-01:
See updated comment above.

Sniffle, 2016-06-01:
How can it do that?  It interprets signals and carries out commands the
same as a servo does.  The only thing it sends to the board is via the
endstop.  The servo code doesnt interpret the position of the servo it just
sends pulses to the servo to make it move.  If you prevent a normal servo
from moving it just grinds the gears trying to move the number of pulses
sent.  Its in effect a dumb actuator with no communication between control
board and chip.  It just stores the last command sent so that it itself
will know when it is in an improper state and gives a physical(blinking
light) warning.

By sending a check you are basically just sending a 2nd retract or deploy
command which wont do anything unless the probe loses power at which point
you are giving the probe power again sending it through its boot sequence
of deploy/retract 2-3 times and then defaulting to a retracted position.

timeshell, 2016-06-01:
You query the Z ENDSTOP status on the probe.  If the pin is up on the probe, it is in error.

thinkyhead, 2016-06-01:
I will have to read the datasheet and programming guidelines for this probe. I'm about to Google, but feel free to post any helpful links.

timeshell, 2016-06-01:
They post a lot of their information here:

https://plus.google.com/communities/109126963511513081214?iem=4&gpawv=1&hl=en-CA

Sniffle, 2016-06-01:
Querying the endstop wont do you any good.  Its a swipe across a hall
effect sensor.  The endstop is triggered via a pulse that is sent for, if i
remember the documentation correctly, .5 seconds.

@thinkyhead. My son is finally asleep ill go pull up what i can and post it.

timeshell, 2016-06-01:
Specs here:

https://plus.google.com/113792662927481823969/posts/femibaRv54Y?hl=en-CA

thinkyhead, 2016-06-01:
[image]

timeshell, 2016-06-01:
From their Indiegogo page:

How to release the Alarm :

    - Power ON after Power OFF 

    - Connect after Disconnect

    - M280 P0 S160

How to get status of endstop using M119 :

Step 1 : Power ON or Port Connect

Step 2:  G28          or  HOME

Step 3 : M119

                x_max: TRIGGERED

                y_max: TRIGGERED

                z_min: TRIGGERED

                z_max: TRIGGERED

Step 4 : M280 P0 S10

Step 5 : M119

                x_max: TRIGGERED

                y_max: TRIGGERED

                z_min: open

                z_max: TRIGGERED

Step 6 : M119 after ALARM

                x_max: TRIGGERED

                y_max: TRIGGERED

                z_min: TRIGGERED

                z_max: TRIGGERED

thinkyhead, 2016-06-01:
Hmm. So to detect the alarm, we just have to detect the ZMIN Endstop being triggered when it should not be triggered… So to deal with the ALARM, we should check the ZMIN endstop at the point when probing is about to start (immediately after moving to the "deployed angle") and keep doing `servo[servo_endstop_id[Z_AXIS]].move(160)` (and delaying a moment) until the Z min endstop goes to "open."

timeshell, 2016-06-01:
A reset command may be required before trying to extend the pin.  When it is in error state I don't believe it will respond until it is reset.

G-Forge, 2016-06-01:
@thinkyhead Can this be revisited if other BLTouch specific options are being added.
https://github.com/MarlinFirmware/Marlin/pull/3329

thinkyhead, 2016-06-01:
@G-Forge Maybe. If a menu item is actually needed, we could add one. But since we can both (a) detect an ALARM condition and (b) automatically reset it, then we won't need an option to manually reset it.

timeshell, 2016-06-01:
You can reset it assuming that the BLTouch is functioning correctly.  If the pin is stuck, it will continue to return to a error condition.  There should probably be a retry limit.

Sniffle, 2016-06-01:
Just so it is known, this is copied directly from my octoprint output.

Send: M280 P0 S90
Recv: ok
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: TRIGGERED
Recv: z_min: open
Recv: ok
[...]
Send: M280 P0 S10
Recv: ok
[...]
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: TRIGGERED
Recv: z_min: open
Recv: ok

I know what the indiegogo says, and they may have modified it to be so, but
previous bltouch configurations do not act in the manner stated on the
indiegogo.

Sniffle, 2016-06-01:
and yes @timeshell the probe will not respond until an error state is
resolved.

thinkyhead, 2016-06-01:
> the probe will not respond until an error state is resolved

Is there any error state that will not respond to `M280 P0 S160`, or is this "guaranteed" to resolve it?

timeshell, 2016-06-01:
It will resolve it, but if the pin sticks it will go into error again once a command to drop the pin is sent and is unsuccessful.

thinkyhead, 2016-06-01:
@timeshell So Marlin might be able to suggest the problem is: "BLTouch stuck pin!"

Sniffle, 2016-06-01:
that is correct from my experience as well, if the pin is stuck, M280 P0 S160 will resolve any error always but it will reoccur if the error was not fixed

timeshell, 2016-06-01:
I would say yes, that would be a safe assumption.

thinkyhead, 2016-06-01:
> if the pin is stuck…

Good to know. Is there a noticeable delay before the "fixed" error goes back into ALARM (or TRIGGERED) state?

Sniffle, 2016-06-01:
between a half second and full second after it attempts to move the pin to what should be its last known position

thinkyhead, 2016-06-01:
I see. Assuming this is a "rare" occurrence, I think it's safe to toss in a 1s delay after `M280 P0 S160` before checking the condition.

Sniffle, 2016-06-01:
I'm still having a hard time wrapping my head around how you are checking its condition.  because as I've demonstrated above, it doesnt act the way the documentation on indiegogo says it does, and this cant be verified until someone gets one from teh current indiegogo campaign and tests to see that it does indeed do as they say it does

and yes it is a rare occurance, in 3 months of use, it has happened once, unless i forced it into an error state to test it

timeshell, 2016-06-01:
@Sniffle are you sure it tries to return to last known position?  I would think a reset would put it back into a closed/pin up position by default.  That being said, I do t know for sure.

thinkyhead, 2016-06-01:
@Sniffle Is that OctoPrint log showing the BLTouch output when it is in an ALARM condition?

timeshell, 2016-06-01:
Mine errors pretty regularly.  I have the most current version as far as I know, they call it version C.

timeshell, 2016-06-01:
What steps did you do to produce that output?  I'll go test right now.

Sniffle, 2016-06-01:
@thinkyhead that is normal endstop reporting form normal operation.  so at all times the endstop shows open.  if you look at the documentation, there is a notation of 5ms and this is the pulse when the endstop is triggered.

@timeshell 
M280 P0 S90
M119
M280 P0 S10
M119

Sniffle, 2016-06-01:
so unless the magnet gets stuck in just the right spot to show endstop triggered, there is no way to know what state the bltouch is in

thinkyhead, 2016-06-01:
@Sniffle It was my understanding, from previous comments, that if the BLTouch is in ALARM state, it will show as TRIGGERED, when it's not supposed to be (i.e., when not actually TRIGGERED).

Sniffle, 2016-06-01:
give me a sec I'll go pull it down by hand and send it into an alarm state and do an M119

thinkyhead, 2016-06-01:
> unless the magnet gets stuck in just the right spot to show endstop triggered…

I see. But when it goes into ALARM state (not just stuck), does it also show as TRIGGERED?

timeshell, 2016-06-01:
Apparently it doesn't.  I can confirm.  I'm pretty sure my original one did, but the C version does not.

timeshell, 2016-06-01:
I'll check, just a sec.

timeshell, 2016-06-01:
Yes it does show triggered.

thinkyhead, 2016-06-01:
Ok, so TRIGGERED is a signal for both ALARM state and MAYBE_STUCK state. But we can't test for a "stuck" probe. We can only know it is _either_ STUCK or ALARM.

timeshell, 2016-06-01:
And when m280 p0 s160 is sent it resets pin to closed/up.

timeshell, 2016-06-01:
If it is stuck it will go into alarm, so not really relevant.

thinkyhead, 2016-06-01:
I see. So after 160 we need to send another 10.

timeshell, 2016-06-01:
I think we should.  Doesn't hurt even it stays down.

Sniffle, 2016-06-01:
I can confirm, I just tested by hand that when in an alarm state that for me at least Z min was triggered

thinkyhead, 2016-06-01:
Ok. So I need to outline a procedure that we can "bolt onto" the current servo deploy/stow stuff in both homing and leveling, which after "servo deployment" takes a short pause, checks the state of the Z_MIN endstop, and if TRIGGERED, then goes into a routine to resolve the state. If the state can't be resolved, then we abort homing or leveling and halt the printer, requiring `M999` to reawaken.

timeshell, 2016-06-01:
Ideally that should include checking the state at EVERY probe point along the way during auto levelling.

Sniffle, 2016-06-01:
that sounds correct, I dont know that it would be useful for any other probe, so it could probably ifdefed to a bltouch specific probe safety define

timeshell, 2016-06-01:
Interestingly, all this had nothing to do with the original report.  :P

[34 more replies]

closed 2016-07-23 by jbrazio