MarlinFirmware/Marlin #3904: RCBugFix branch does miss a few directories..

lrpirlet, 2016-05-29:
@thinkyhead  I was trying to answer your suggestions for another issue when I observed the following.

Is it me or do we miss quite  a few files in RCBugFix Branch?? 

I miss the Documentation directory, the LinuxAddon/bin directory, the PlatformIOAddons when comparing with the RC branch...

Sorry, I have no deep knowledge of git and/or github nor the time to learn for the moment so I will not be able to search for a possible cause...

lrpirlet, 2016-05-29:
All right, maybe I should have searched a little longer... Seems that, at least the logo file, is under another directory...

I, personally, would have documented that change in the README.md file... I feel this to be a change that would need be documented immediately... I hate data loss... I felt it as a big one... Of course this is only my humble opinion...

jbrazio, 2016-05-29:
It's documented on the PR section of Github.

jbrazio, 2016-05-29:
In a nutshell, all support files have being moved into the build root folder.

closed 2016-05-29 by jbrazio

thinkyhead, 2016-05-30:
@lrpirlet Sorry about all that. The files have been reorganized pretty recently. I'm not sure why "buildroot" was chosen, but maybe that's a standard naming convention now…?

lrpirlet, 2016-05-30:
@thinkyhead No problem, any directory organization is valid... 

My only regret is that this move was hidden with the many information coming from MarlinDev... 

In fact, this is my fault and I should not have raised any issue.. I, not anyone else, decided to mark all my emails from marlin as read without reading and, at the same time, I assumed that the directory structure would not change...

jbrazio, 2016-05-31:
@thinkyhead It was the name I choose due to the "support" contents like scripts and such. If we find a better name we can rename it.

thinkyhead, 2016-06-01:
It's fine with me. I can understand users wondering, and I want to assure them it's not purely arbitrary.