MarlinFirmware/Marlin #3905: Dual X Carriage duplication mode

pbrgle1, 2016-05-29:
Hi. This is either a Support question or a Feature request concerning the Dual X Carriage feature in Duplication mode (M605 S2).

I was just wondering if there is (or could be) a possibility to stop one of the extruders in duplication mode if one of the two prints should fail. It is silly that you would have to cancel the whole print if only one print failed.

The main goal would be only to disable one of the extruder motors (via command?) and a more advanced option (if possible) would also be to park the failed extruder and disable the heater.

I'm just laying my idea out there. I'm sadly not capable of doing any modifications to the code.

caenan0, 2016-05-29:
For now, a work-around could be to manually send an M104 T1 to turn off the heater and the tool will keep moving around but should stop extruding (provided you have mintemp protection).

barus93, 2016-05-30:
If you must stop the T1 you can send m605 s1/s0

maukcc, 2016-05-31:
Is there already a request to extend on this feature. Like
I want extrudermotor 1 and 3 doing the same and 2 and 4 doing the same
or I want 1,2,3 and 4 doing the same

thinkyhead, 2016-05-31:
@maukcc To do things more generally, as you suggest, will require a systematic rewrite of various areas of the stepper, planner, and main code. We do have it in mind to do more "abstraction" on extruders, axes, steppers, etc., but we have to be careful not to over-generalize, because it tends to produce slower code.

closed 2016-07-11 by jbrazio