MarlinFirmware/Marlin #3910: E-step not respected

Scratchbuild52, 2016-05-30:
Hi there,

I have wasted a lot of time yesterday trying to calibrate my extruder. I have tried to config the e-steps through the config file and it did not work. I then after a while suspected that Marlin do not save the value and compared it with my lcd-values. The values were not nearly the same. After changing the value on my lcd , Marlin started responding to the e-step value and I could calibrate my extruder, luckily I had the LCD!!! So this is definitely a bug as I have tested it thoroughly. I did not check the code yet.

I did not upgrade to a bug release, because I could not see this fixed in one of the releases.

Regards 

Nic

Blue-Marlin, 2016-05-30:
You have EEPROM on. You just have to be aware of that.

thinkyhead, 2016-05-30:
@Scratchbuild52 As an extra measure, after flashing a new Marlin it can help to reset settings with `M502`, then check to see that they are what you want (they should be reset to the values set in your configuration), and then save to EEPROM with `M500`.

Scratchbuild52, 2016-05-30:
I will check tonight, but as far as I know, it is on, but I might be mistaken. I did load from memory again but I will try the reset tonight.

Thank You.

Regards

Nic

Scratchbuild52, 2016-05-30:
Sorry for the close, will first come back to you.

Scratchbuild52, 2016-05-30:
Hi guys,

I have checked , I did enable eeprom already in my config i.e.
#define EEPROM_SETTINGS

#if ENABLED(EEPROM_SETTINGS)
  // To disable EEPROM Serial responses and decrease program space by ~1700 byte: comment this out:
  #define EEPROM_CHITCHAT // Please keep turned on if you can.
#endif

So the problem could not have been caused by that. 

I did ran a M502 now and it suddenly updated. Do you always have to run M502, should it not update automatically on a reload?

Regards

Nic

Jnesselr, 2016-05-30:
I think you misunderstood. Having eeprom on means that your settings get backed up to eeprom. Since they were backed up to eeprom, you were seeing that older value. You have to run M502 to clear those older values.

thinkyhead, 2016-05-31:
@Jnesselr has explained it well.

closed 2016-05-31 by thinkyhead