MarlinFirmware/Marlin #3912: LCD contrast

maukcc, 2016-05-30:
We define our user interface as following in Conditionals.h:

#if defined (CartesioUI)
  #define DOGLCD
  #define SDSUPPORT
  #define ULTIPANEL
  #define NEWPANEL
  #define DEFAULT_LCD_CONTRAST               90
  #define ENCODER_PULSES_PER_STEP            2
  #define ENCODER_STEPS_PER_MENU_ITEM        1
  #define LCD_FEEDBACK_FREQUENCY_HZ          800  
  #define LCD_FEEDBACK_FREQUENCY_DURATION_MS 2000 
  #define REPRAPWORLD_KEYPAD
  #define REPRAPWORLD_KEYPAD_MOVE_STEP 1.0
 #endif

The   #define DEFAULT_LCD_CONTRAST setting does not work here.
It only works when we change:

  #if ENABLED(DOGLCD) && DISABLED(DEFAULT_LCD_CONTRAST)
    #define DEFAULT_LCD_CONTRAST 90
  #endif

Also the contrast setting in the lcd menu only goes to 60 or so which is not enough and always blanks the display

thinkyhead, 2016-05-30:
It looks like the `DISABLED` in that code block is causing the problem. Your code should work properly if you change that same block to this instead:
```cpp
  #if ENABLED(DOGLCD) &&!defined(DEFAULT_LCD_CONTRAST)
    #define DEFAULT_LCD_CONTRAST 32
  #endif
```
I will submit a patch for this issue shortly.

maukcc, 2016-05-30:
That worked. 
Now only the setting of contrast in the LCD menu goes to 63 and not higher

thinkyhead, 2016-05-30:
I'm not sure why the LCD contrast menu item is limiting values the way that it is. For `U8GLIB_LM6059_AF` it allows values from 0-255, but for all other displays it only allows values of 0-63. What type of "U8G" display would yours be? And what should be its full range of values?
```cpp
static void lcd_set_contrast() {
  ENCODER_DIRECTION_NORMAL();
  if (encoderPosition) {
    #if ENABLED(U8GLIB_LM6059_AF)
      lcd_contrast += encoderPosition;
      lcd_contrast &= 0xFF;
    #else
      lcd_contrast -= encoderPosition;
      lcd_contrast &= 0x3F;
    #endif
    encoderPosition = 0;
    lcdDrawUpdate = LCDVIEW_REDRAW_NOW;
    u8g.setContrast(lcd_contrast);
  }
  if (lcdDrawUpdate) {
    #if ENABLED(U8GLIB_LM6059_AF)
      lcd_implementation_drawedit(PSTR(MSG_CONTRAST), itostr3(lcd_contrast));
    #else
      lcd_implementation_drawedit(PSTR(MSG_CONTRAST), itostr2(lcd_contrast));
    #endif
  }
  if (LCD_CLICKED) lcd_goto_previous_menu(true);
}
```

thinkyhead, 2016-05-30:
(You may be able to just set `U8GLIB_LM6059_AF` in your CartesioUI block. This is what the `ELB_FULL_GRAPHIC_CONTROLLER` does:
```cpp
    #elif ENABLED(ELB_FULL_GRAPHIC_CONTROLLER)
      #define DEFAULT_LCD_CONTRAST 110
      #define U8GLIB_LM6059_AF
      #define SD_DETECT_INVERTED
    #endif
```

maukcc, 2016-05-30:
I changed the integer of 63 to 255 and that works for my DOGM128 display. I can not see why it would not work for others. Only thing is that the lcd menu only shows 2 characters. So if you go larger the 99 it looks strange 127 will show 27

#if ENABLED(HAS_LCD_CONTRAST)
  static void lcd_set_contrast() {
    ENCODER_DIRECTION_NORMAL();
    if (encoderPosition) {
      #if ENABLED(U8GLIB_LM6059_AF)
        lcd_contrast += encoderPosition;
        lcd_contrast &= 0xFF;
      #else
        lcd_contrast -= encoderPosition;
        **lcd_contrast &= 0xFF; // 0x3F;**
      #endif
      encoderPosition = 0;
      lcdDrawUpdate = LCDVIEW_REDRAW_NOW;
      u8g.setContrast(lcd_contrast);
    }
    if (lcdDrawUpdate) {
      #if ENABLED(U8GLIB_LM6059_AF)
        lcd_implementation_drawedit(PSTR(MSG_CONTRAST), itostr3(lcd_contrast));
      #else
        lcd_implementation_drawedit(PSTR(MSG_CONTRAST), itostr2(lcd_contrast));
      #endif
    }
    if (LCD_CLICKED) lcd_goto_previous_menu(true);
  }
#endif // HAS_LCD_CONTRAST

maukcc, 2016-05-30:
And change
lcd_contrast -= encoderPosition;

to
lcd_contrast += encoderPosition;

as otherwise the dial goes in the wrong direction

Sebastianv650, 2016-05-30:
Just a quick first thought of mine regarding the 63 max limit:
You said that the display can only show 2 digits. 63 is the biggest 2-digit number resulting from 255/x (x=4). Maybe the original intention was to have values from 0-63 in the display, but it should be multiplied by 4 for the real contrast value to give the full range?

thinkyhead, 2016-05-31:
> the dial goes in the wrong direction

If it goes the wrong direction in other places, we have a setting for that.

thinkyhead, 2016-05-31:
> the display can only show 2 digits

Look at the code. It displays 3 digits if you define `U8GLIB_LM6059_AF`. And this is the flag that also allows you to have values from 0-255. Try just setting that option and see that it works for your custom controller.
```cpp
#define U8GLIB_LM6059_AF
```

maukcc, 2016-05-31:
The dial does exactly what it should do except for the contrast, because the code is wrong.
so you say : use the patch(U8GLIB_LM6059_AF) as we are not going to take the bug out of the native code? 
If there is somebody out there with a "very special" display that can only go to 63 contrast, let them make the patch.
My answer to the bug is: take the whole patch and make that the native code.
FIY the answer to the 3 digits is here:
change
lcd_implementation_drawedit(PSTR(MSG_CONTRAST), itostr2(lcd_contrast));

to
lcd_implementation_drawedit(PSTR(MSG_CONTRAST), itostr3(lcd_contrast));

thinkyhead, 2016-05-31:
@maukcc There is no bug. If you examine the code, you can see that all it does is take the "normal" case with a range of 0-63. The larger range of 0-255 is a "special case." If your display is among those that work with a range of 0-255, requiring 3 digits, then we should add a new flag and make that part of the test that compiles the code. For example:
```cpp
#if ENABLED(U8GLIB_LM6059_AF) || ENABLED(U8GLIB_LM1234_XY)
. . .
```
…where `U8GLIB_LM1234_XY` would indicate the chipset that _your_ display uses. If you don't use the [LM6059](http://www.diytrade.com/china/pd/10011610/LM6059_for_128x64_dots.html) then which chip do you use?

maukcc, 2016-06-01:
I will have to disagree :) and say that 0-255 is the normal approach. Even my DOGM128 display starts at 40 and blanks out at 120. So am I saying that we need a special entry for 40-120 displays? no.
because the general 0-255 works for all. 0-63 does not work for all, so get that out of there completely, unless there actually is a display that breaks down when set to a higher value then 63

thinkyhead, 2016-06-01:
@maukcc Unfortunately I don't know the rationale for the original code. And I am not an expert on graphical displays. But yours appears to be the first non-U8GLIB_LM6059_AF display that anyone has presented with a range other than 0-63. At least, no one has ever complained about contrast on their graphical display up to now…

So we need to look at the history of that code and query the original author on their rationale before proceeding.

maukcc, 2016-06-01:
I am guessing the reason for no one complaining, is that no one actually cares :) 
Me either as I never use it. But if it is in there it should work properly.

thinkyhead, 2016-06-01:
#### Graphical Displays with software-based contrast:
- `DOGLCD` [generic] (default 32, range 0-63)
- `MAKRPANEL` (default 17)
- `MINIPANEL` (default 17)
- `miniVIKI` (default 95)
- `VIKI2` (default 40, range)
- `ELB_FULL_GRAPHIC_CONTROLLER` (default 110, range 0-255)

#### Graphical Displays with **no** software-based contrast:
- `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`
- `U8GLIB_SSD1306`
- `SAV_3DGLCD`
- `U8GLIB_SH1106`

thinkyhead, 2016-06-01:
@royco We were just wondering about the correct contrast settings for `miniViki`. Currently the default contrast value is set at **95** but the adjustable range is only **0-63**. Should the default be lower, or should the adjustable range be any larger, like **0-255**?

royco, 2016-06-01:
Default contrast for the miniViki is 95 but adjustment range in marlin is
0-63 and at that time I could not find where to adjust it. The miniViki was
released only to printer manufacturers, a different version (with probably
different contrast setting) will be available to the public soon.

thinkyhead, 2016-06-02:
Thanks @royco. I will extend the contrast range for miniVIKI to 0-255, as we're just cleaning up the contrast code generally.

thinkyhead, 2016-06-04:
@maukcc #3944 patches Marlin with more comprehensive contrast handling. See how contrast is handled now, and adjust your code to follow suit.

closed 2016-06-04 by thinkyhead