MarlinFirmware/Marlin #3915: REPRAPWORLD_KEYPAD

maukcc, 2016-05-30:
If you are moving the X or Y axis with the keypad and click the homing button as well on the keypad It will start moving in Y+ axis and not stop.
Z-axis is not affected

maukcc, 2016-05-30:
I also did:

        //#if ENABLED(DELTA) || ENABLED(SCARA)
          #define _KEYPAD_MOVE_ALLOWED (axis_homed[X_AXIS] && axis_homed[Y_AXIS] && axis_homed[Z_AXIS])
        //#else
        //  #define _KEYPAD_MOVE_ALLOWED true
        //#endif

as that is also good practice in cartesian machines

maukcc, 2016-05-30:
Also: the home button works while printing

jbrazio, 2016-05-30:
Are you using the latest `RCBugfix` ?

maukcc, 2016-05-30:
1.1.0-RC6

thinkyhead, 2016-05-31:
Please test `RCBugFix` so we can see if this is still an issue now.

maukcc, 2016-05-31:
If I have a lot of time I will switch to the RCBugFix. for now I looked at the code in ultralcd.cpp and it seems the same in RCBugFix.

maukcc, 2016-05-31:
I did : 
      #if ENABLED(REPRAPWORLD_KEYPAD)

        #define _KEYPAD_MOVE_ALLOWED (axis_homed[X_AXIS] && axis_homed[Y_AXIS] && axis_homed[Z_AXIS])
       
        //if (REPRAPWORLD_KEYPAD_MOVE_HOME)       reprapworld_keypad_move_home();
        //if (REPRAPWORLD_KEYPAD_MOVE_HOME_Z)     reprapworld_keypad_move_home_z();
        if (REPRAPWORLD_KEYPAD_MOVE_Z_UP)     reprapworld_keypad_move_z_up();
      
        if (_KEYPAD_MOVE_ALLOWED) {
          if (REPRAPWORLD_KEYPAD_MOVE_Z_DOWN)   reprapworld_keypad_move_z_down();
          if (REPRAPWORLD_KEYPAD_MOVE_X_LEFT)   reprapworld_keypad_move_x_left();
          if (REPRAPWORLD_KEYPAD_MOVE_X_RIGHT)  reprapworld_keypad_move_x_right();
          if (REPRAPWORLD_KEYPAD_MOVE_Y_DOWN)   reprapworld_keypad_move_y_down();
          if (REPRAPWORLD_KEYPAD_MOVE_Y_UP)     reprapworld_keypad_move_y_up();
        }
      #endif

so, no homing from keypad, all moves(except Z/extruder distance larger) blocked before homing. 
But safe homing (or setting current position to zero for milling) would be nice.
Or a notice "are you sure you want to do this now? as I have print commands in buffer"

Blue-Marlin, 2016-05-31:
How about:
```cpp
  #if ENABLED(REPRAPWORLD_KEYPAD)

    #define _KEYPAD_MOVE_ALLOWED (axis_homed[X_AXIS] && axis_homed[Y_AXIS] && axis_homed[Z_AXIS])

    if (REPRAPWORLD_KEYPAD_MOVE_Z_UP)     reprapworld_keypad_move_z_up();

    if (_KEYPAD_MOVE_ALLOWED) {
      if (REPRAPWORLD_KEYPAD_MOVE_Z_DOWN)   reprapworld_keypad_move_z_down();
      if (REPRAPWORLD_KEYPAD_MOVE_X_LEFT)   reprapworld_keypad_move_x_left();
      if (REPRAPWORLD_KEYPAD_MOVE_X_RIGHT)  reprapworld_keypad_move_x_right();
      if (REPRAPWORLD_KEYPAD_MOVE_Y_DOWN)   reprapworld_keypad_move_y_down();
      if (REPRAPWORLD_KEYPAD_MOVE_Y_UP)     reprapworld_keypad_move_y_up();
    }
    else {
      if (REPRAPWORLD_KEYPAD_MOVE_HOME)     reprapworld_keypad_move_home();
      if (REPRAPWORLD_KEYPAD_MOVE_HOME_Z)   reprapworld_keypad_move_home_z();
    }
  #endif
```

maukcc, 2016-06-02:
@Blue-Marlin : that worked 
I made some additional changes:

      #if ENABLED(REPRAPWORLD_KEYPAD)

        #define _KEYPAD_MOVE_ALLOWED (axis_homed[X_AXIS] && axis_homed[Y_AXIS] && axis_homed[Z_AXIS])
       
        if (REPRAPWORLD_KEYPAD_MOVE_Z_UP)     reprapworld_keypad_move_z_up();
        if (REPRAPWORLD_KEYPAD_MOVE_MENU)     reprapworld_keypad_move_menu();
      
        if (_KEYPAD_MOVE_ALLOWED) {
          if (REPRAPWORLD_KEYPAD_MOVE_Z_DOWN)   reprapworld_keypad_move_z_down();
          if (REPRAPWORLD_KEYPAD_MOVE_X_LEFT)   reprapworld_keypad_move_x_left();
          if (REPRAPWORLD_KEYPAD_MOVE_X_RIGHT)  reprapworld_keypad_move_x_right();
          if (REPRAPWORLD_KEYPAD_MOVE_Y_DOWN)   reprapworld_keypad_move_y_down();
          if (REPRAPWORLD_KEYPAD_MOVE_Y_UP)     reprapworld_keypad_move_y_up();
        }
        else {
          if (REPRAPWORLD_KEYPAD_MOVE_HOME)     reprapworld_keypad_move_home();
        }
      #endif

so homing homes all axis, and the other key (F1) gets you into the move menu to select 0.1 or 1 or 10mm

  #if ENABLED(REPRAPWORLD_KEYPAD)
    static void reprapworld_keypad_move_z_up() {
      encoderPosition = 1;
      lcd_move_z();
    }
    static void reprapworld_keypad_move_z_down() {
      encoderPosition = -1;
      lcd_move_z();
    }
    static void reprapworld_keypad_move_x_left() {
      encoderPosition = -1;
      lcd_move_x();
    }
    static void reprapworld_keypad_move_x_right() {
      encoderPosition = 1;
      lcd_move_x();
    }
    static void reprapworld_keypad_move_y_down() {
      encoderPosition = 1;
      lcd_move_y();
    }
    static void reprapworld_keypad_move_y_up() {
      encoderPosition = -1;
      lcd_move_y();
    }
    static void reprapworld_keypad_move_home() {
      enqueue_and_echo_commands_P(PSTR("G28")); // move all axes home
    }
    static void reprapworld_keypad_move_menu() {
      lcd_goto_menu(lcd_move_menu , 0, false);
    }
  #endif // REPRAPWORLD_KEYPAD

this also eliminates the 

#define REPRAPWORLD_KEYPAD_MOVE_STEP 10.0

The only thing now is that there is no default move step (0.1  or 1 or 10) at boot

thinkyhead, 2016-06-04:
I'm working up a PR based on these suggestions. How do you define `REPRAPWORLD_KEYPAD_MOVE_MENU` ?

thinkyhead, 2016-06-04:
How about…
```cpp
if (REPRAPWORLD_KEYPAD_MOVE_HOME)       reprapworld_keypad_move_home();
if (REPRAPWORLD_KEYPAD_MOVE_MENU)       reprapworld_keypad_move_menu();

#if DISABLED(DELTA) && Z_HOME_DIR == -1
  if (REPRAPWORLD_KEYPAD_MOVE_Z_UP)     reprapworld_keypad_move_z_up();
#endif

if (axis_homed[X_AXIS] && axis_homed[Y_AXIS] && axis_homed[Z_AXIS]) {
  #if ENABLED(DELTA) || Z_HOME_DIR != -1
    if (REPRAPWORLD_KEYPAD_MOVE_Z_UP)   reprapworld_keypad_move_z_up();
  #endif
  if (REPRAPWORLD_KEYPAD_MOVE_Z_DOWN)   reprapworld_keypad_move_z_down();
  if (REPRAPWORLD_KEYPAD_MOVE_X_LEFT)   reprapworld_keypad_move_x_left();
  if (REPRAPWORLD_KEYPAD_MOVE_X_RIGHT)  reprapworld_keypad_move_x_right();
  if (REPRAPWORLD_KEYPAD_MOVE_Y_DOWN)   reprapworld_keypad_move_y_down();
  if (REPRAPWORLD_KEYPAD_MOVE_Y_UP)     reprapworld_keypad_move_y_up();
}
```

thinkyhead, 2016-06-04:
To make the keypad handler block non-reentrant is pretty easy. At the top of the block we add a `static bool`:
```cpp
static bool keypad_move_handling = false;
```
Then at the top of the block…
```
if (keypad_move_handling) return;
keypad_move_handling = true;
```
…and at the end of the block:
```
keypad_move_handling = false;
```
This will prevent the LCD handler from processing any keypad keys during a **_re-entrant call_**, such as occurs when queueing many planner moves in a row.

thinkyhead, 2016-06-04:
@maukcc See if #3957 helps at all…

maukcc, 2016-06-06:
close this and let`s move over to #3957

closed 2016-06-06 by thinkyhead