MarlinFirmware/Marlin #3919: Looking for language maintainers

jbrazio, 2016-05-30:
We need your help, Marlin has a lot of language packs and we need to create a language team that will be responsible for having the translations up to date. Please answer to this issue with the language you would like to maintain, but be aware we prefer native speakers. We would like to have at least three people per supported language.

 - [ ] English (en)
 - [ ] Aragonese (an)
 - [ ] Basque-Euskera (eu)
 - [ ] Bulgarian (bg)
 - [ ] Catalan (ca)
 - [ ] Chinese (cn)
 - [ ] Czech (cz)
 - [ ] Dutch (nl)
 - [ ] Finnish (fi)
 - [ ] French (fr)
 - [ ] Galician (gl)
 - [ ] German (de)
 - [ ] Italian (it)
 - [ ] Japanese (kana)
 - [ ] Japanese (kana_utf8)
 - [ ] Polish (pl)
 - [ ] Portuguese (pt): @jbrazio
 - [ ] Portuguese (pt-br)
 - [ ] Portuguese (pt-br_utf8)
 - [ ] Portuguese (pt_utf8): @jbrazio
 - [ ] Russian (ru)
 - [ ] Spanish (es)

boelle, 2016-05-30:
if the Oled for the sav mk1 was supported i might (no gurantee) do Danish

it works best for me if i get a english file with string that i can translate and test right away... some stuff needs to be display as a whole before it can be certified to be correct translated

the oled is supported here: https://github.com/fmalpartida/Marlin/tree/SAV-MkI-config

emartinez167, 2016-05-30:
I can take on Spanish

jbrazio, 2016-05-30:
@boelle we never had a Danish translation ?
I can merge that PR on a branch in my repo so you can test with your display, is it OK ? If everything runs smooth we submit a new PR against RCBugfix.

paulusjacobus, 2016-05-31:
Happy to help out with Dutch

boelle, 2016-05-31:
we do have danish in bug fix

https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/language_da.h

jbrazio, 2016-05-31:
@barus93 can we count with you for Italian ?
@rafacouto can we count with you for Galician ?
@gege2b can we count with you for French ?
@koldoartola, @MatixYo can we count with you for Basque-Euskera ?
@RicardoGA, @fbarcenas can we count with you for Spanish ?
@AnHardt, @Blue-Marlin, @CONSULitAS can we count with you for German ?
@DavidBjerreBjoerklund, @benj4560 can we count with you for Danish ?
@pixatintes can we count with you for Catalan ?
@danielmartinez can we count with you for Aragonese ?

boelle, 2016-05-31:
made a few changes do da... will open the english to compare

you should have a PR

boelle, 2016-05-31:
hehe... now you have.... forgot to push the PR

danielmartinez, 2016-05-31:
Yes please, count with me for Aragonese, but in several days.

jbrazio, 2016-05-31:
@danielmartinez we're just building the teams.

boelle, 2016-05-31:
danish almost there.... i only have a very few defines i need to study more close

rafacouto, 2016-05-31:
Of course, I'll translate defines into Galician :-)

RicardoGA, 2016-05-31:
@jbrazio of course, I can help with Spanish

fbarcenas, 2016-05-31:
@jbrazio Of course!

barus93, 2016-05-31:
@jbrazio ok, i can help with italian..

jbrazio, 2016-05-31:
@boelle we have a script to do it, but has to be ran manually.
Have a look [here](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/buildroot/share/scripts/findMissingTranslations.sh).

Roxy-3DPrintBoard, 2016-05-31:
We have a lot of hard coded English strings.  Especially in Marlin_main.cpp.    (Just look at the M48: code.   I'm the guilty party on that one!)

clexpert, 2016-05-31:
We will maintain Czech language as we did.

clexpert, 2016-05-31:
@jbrazio We have complete Czech language with no problem. All translates on time. Please use @clexpert and @petrzjunior for Czech language as sufficient solution. You can tick this language.

Blue-Marlin, 2016-05-31:
Dear @Roxy-3DPrintBoard 
M48 has no display messages. Only display messages are translated. All serial output is in English. All strings used for the serial output 'should' be in 'language.h' - not 'language_en.h'. There is no need for immediate action - just if someone is bored he/she could clean up a bit.

thinkyhead, 2016-05-31:
@boelle Here you go…#3936. Looking forward to that Danish!

CONSULitAS, 2016-06-01:
@jbrazio 
German is fine for me :smile:

CONSULitAS, 2016-06-01:
@jbrazio 
BTW: There was a complain about not translated strings in German. 
Many technical terms are the same words in German like in English. 
So perhaps we should add a comment like 'intentionally untranslated' to those.

Roxy-3DPrintBoard, 2016-06-01:
This will result in a lot of clutter in the administrative pages.   By that same thinking, we will need Code-Maintainers-PreProcessor-Bugs as well as Code-Maintainers-Header-Files etc. etc.

AnHardt, 2016-06-01:
Pleas avoid that overhead.
If i see a need for a change in any file i'll do it, or will ask for help, anyway.
If it would be someones duty, i'd stay away.

Indeed the unbearable German language file, from ending 2014 was the first thing i wanted to change and was the reason for beginning with Marlin programming. :-)

jbrazio, 2016-06-02:
I plan to have all the people related with language on a "global" language team as I agree with Roxy and Andreas, it will be a clutter for now and usually when we need to add a new string, it must be added to all languages.

@AnHardt may I add you for German then ? :-)

AnHardt, 2016-06-02:
@jbrazio 
No. _Someone_ includes me.

rafacouto, 2016-06-02:
Should we translate in the RCBugFix branch? Marlin or MarlinDev repo? I miss a little translator guide (excuse me if it already exists O:) in order to assist new maintainers and non git-ninjas...

Roxy-3DPrintBoard, 2016-06-02:
MarlinDev is going away.   RCBugFix is the correct place because what ever is there is going forward.  And just for right now, during this release cycle, the RC branch is acting as 'Stable' and probably doesn't get the fix applied to it.

petrzjunior, 2016-06-02:
So now, I think it's the right time to discuss some future plans. 
Will the string be translated on every PR someone defines a new string? Or before every release (say every mothly release) we translate all the string that were added since the last one (maybe push against one fork and track progress in one PR)?

jbrazio, 2016-06-02:
Usually we need to translate a new string when a new feature is added.. up to now the developer adds the string into the English file and (which I do not agree) adds the same define in English on all other languages. The ideia is when a new string is added on the PR we just ping @MarlinFirmware/language-team informing there is a new string that must be translated.

boelle, 2016-06-03:
@jbrazio you can tick of danish... i just submitted a pr which should make it complete

i checked 2 times by hand

boelle, 2016-06-03:
@gege2b can we count with you for French ?
@koldoartola, @MatixYo can we count with you for Basque-Euskera ?
@AnHardt, @Blue-Marlin, @CONSULitAS can we count with you for German ?
@DavidBjerreBjoerklund, @benj4560 can we count with you for Danish ?
@pixatintes can we count with you for Catalan ?

closed 2016-06-03 by jbrazio

thinkyhead, 2016-06-04:
> and (which I do not agree) adds the same define in English on all other languages

This is specifically designed to annoy language maintainers into fixing it. 😝