MarlinFirmware/Marlin #3920: Small circles/arcs don't print well RC6

paulusjacobus, 2016-05-30:
@thinkyhead I am printing with RC6 (It2 prusa RAMPS1.4) which works well for normal sized objects but for small circles/cylinders like 2.5mm diameter, they all end up like like screws/staircases which easily break off. I tried various things before logging this issue: 
-reduced retraction from 4.5mm to 0.1mm (direct drive)
-slowed down the printing to 20mm/s
-increased the infill to 50%. 

Next step I can try is to print a few cylinders in various sizes 1 - 10 mm. The first time I noticed this was with the CTRLV test from Thingiverse. The dome and cone had the staircase/screw at the top layers.
Is this related to the printing arc limitations in Marlin or is it a hardware failure/limitation on my side?

thinkyhead, 2016-05-31:
It could be so many things. I'm not sure how you can isolate the cause.

paulusjacobus, 2016-05-31:
I have not seen this behavior before so I will try to isolate the case by
printing various sizes of cylinders and see when it happens. Than I can
investigate the slicer code and see if there is a correlation between the
used G codes for each print or not.

It is indeed a tough one (who? slicer, firmware or printer) and I hope a
few other users can try to print this example that I have attached
(conrod.stl) to see if they encounter the same issue or not so at least I
know where to look. The base of the model prints well but the two small
arcs on top don't, they become little staircases like the walls are missing.

brainscan, 2016-05-31:
Could you possible post a pic of how it printed? I've had some weird looking circles but thought it was mechanical as I'd just made some changes. I'll try printing the conrod and post the results in a bit.

Blue-Marlin, 2016-05-31:
A bit more input please.
DELTA or not?
G0/G1 or G2/G3?

brainscan, 2016-05-31:
@paulusjacobus I must have fixed my problem which was mechanical, just printed your conrod and its fine so I don't think its the .stl that's the problem.

paulusjacobus, 2016-05-31:
@brainscan thank you very much for testing, it does help me with the analysis. I imported the stl into tinkercad and replaced the little cylinder discs with a tinkercad ones which are low res with less segments and tatata it printed fine.
So it could be that the stepper motors in my prusa iteration2 are getting too much current to accurately print small segments. I will try to tune the pot meters on the step sticks (drv8255s 1/32). Maybe the stepsticks need 1/16 in order to print the segments?
I will try and report back and then close the issue.

brainscan, 2016-05-31:
No problem, I'm happy to help if I can, which isn't often. I can only do x16 atm so can't check if x32 would make a difference. What are your jerk settings? I set mine too low once and had some horrible prints.

paulusjacobus, 2016-05-31:
@brainscan my have my jerk settings set to the default Marlin values since
those settings prints well for the normal sized objects (99% of my prints).
I think I will set the step sticks back to 1/16. I bought the DRV8255
because my extruder stepper was a bit underpowered on the old stepstick
A4988. The DRV step stick just delivers a bit more current than the old
ones (1.5 A instead of 1A) and the clicking of my extruder is entirely
gone.
The segments on the stl for the circles was set to 100 according to the
designer of that stl.

paulusjacobus, 2016-05-31:
Issue closed i.e. Too many segments

closed 2016-05-31 by paulusjacobus

thinkyhead, 2016-06-01:
Very tiny lines may simply be dropped and ignored. I suggest that if printing very small circles, don't use too many facets. 6 or 8 might be fine for a 2mm circle.