MarlinFirmware/Marlin #3927: Tune - Change Filament not working in RC6

WZ9V, 2016-05-31:
If I try to do a filament change from the LCD it works right up until it goes to print again.  Once it gets into position if backs out the filament again rather than resuming extrusion.

I pick change filament on the menu.  It moves away from the print as expected.  It ejects the filament as expected.  I load new filament and it extrudes as expected.  Once the color runs clean I hit the button on the LCD and it stops extruding as expected.  It moves back into position and then instead of picking up where it stopped, it ejects the filament again and sits there.

I scanned through the issues and couldn't see if this had been addressed in RCBugFix.  Sorry if it's alread in there.

thinkyhead, 2016-05-31:
We have a couple of PRs in the queue that aim to reform Filament Change. Hopefully those will be integrated soon. In the meantime, you might want to give them a test. The relevant PRs are #3605 and #3662.

WZ9V, 2016-05-31:
I'm not really sure how to test specific PR's.  Is there a for dummies guide?  If not I can wait and do some testing once you integrate them into bugfix.  I'll be happy to give it a whirl on my Robo R1+

thinkyhead, 2016-05-31:
At the top of the PR page you can see where it comes from. For example, #3662 comes from "`thinkyhead:rc_fixup_auto_filament_change`". That means it comes from https://github.com/thinkyhead/Marlin/tree/rc_fixup_auto_filament_change and you can download it there.

jbrazio, 2016-06-23:
Will be fixed by #3662

closed 2016-06-23 by jbrazio

thinkyhead, 2016-06-24:
> Will be fixed by #3662

But oh how that needs more work. It currently requires 20x4 LCD.