MarlinFirmware/Marlin #3930: Pause command like change filament

maukcc, 2016-05-31:
I would like the pause printing command to be like the change filament command.
So when a pause command is given, retract Xmm, move to a position (preferably the back for pause and the front for change filament) and wait for another press of the dial, while playing irritating sounds : ). Then go back and adjust the retract and resume printing.

Pausing now will make for ill prints as it will ooze all over the place

thinkyhead, 2016-05-31:
I agree, this would be a good addition. The print head will need (ideally) to raise all the way to the top, to ensure that if you were printing a plate of several objects one-at-a-time, for example, the nozzle won't run into any of them during the pause.

maukcc, 2016-06-01:
Does anybody actually use the "print several objects one-at-a-time" approach, as I have not had any idea where that may be useful. I would love to hear some examples. 
I like it to be like M600 as you can set your own values of where to move and what to do when pause is being clicked.

thinkyhead, 2016-06-01:
@maukcc It is meant to always be supported. And yes, we have heard from a fair number of people who prefer it, because if one object doesn't come out, for some reason, the others usually still will.

brainscan, 2016-06-01:
Could it please be adjustable in some way? Never printed one at a time and can't see me ever using it.

thinkyhead, 2016-06-01:
@brainscan Yes, if we add retraction, Z lift, and parking to the "Pause print" function, rest assured that the amount of retraction and Z lift, and the park position will all be customizable.

brainscan, 2016-06-01:
That sounds excellent, thanks.

petrzjunior, 2016-06-01:
If I understeand you correctly, it's the matter of function I work on with @clexpert. You will have option to change filament (same way as you described) form LCD or via M600 code. Need only few days to test it before it will be available here to discuss, improve and maybe merge.

thinkyhead, 2016-06-01:
> change filament (same way as you described) form LCD or via M600 code

@petrzjunior See #3605 and #3662 for our current work on filament change and `M600`.

petrzjunior, 2016-06-01:
@thinkyhead Yes, the code is from my father @clexpert, I was helping him with it, so I know perfectly was is in that thread. :smile:

thinkyhead, 2016-06-01:
> I was helping him with it

Fantastic! I don't think we've ever had a father-son team contribute anything before.

petrzjunior, 2016-07-19:
@maukcc #3662 was merged some time ago, now in `RCBugFix`, is it fulfilling your idea? Do you have any suggestions?

maukcc, 2016-07-20:
NO, it is not fulfilling my idea :). This is the change filament feature and works as predicted.
I need the pause command to move the head to a park position (which may or may not be the same as the change filament position)
And I want to do this pause(and change filament) from my home with octoprint(or any other host), then drive to work, actually change the filament, and press resume, locally, on my LCD display.
Also the stop function still does not shut of the heaters in bugfix.
Back to RC6 :)

petrzjunior, 2016-07-20:
I hope this could be mor eor less done with M600 if you don't mind beeper beeping whole time while you're driving to work.
> stop function still does not shut of the heaters

You can shut them with g-code, can't you?

maukcc, 2016-07-20:
Let him beep all night long :)
With M600 I want the extruder to come to the front (for easy access to my extruder or other tool). With pause I want the extruder to go to the back (for easy access to my build platform).

>  You can shut them with g-code, can't you?

Yes, but this is not good enough for a CE certification. the operational stop(not to be confused with the emergency stop) needs to put the machine in a neutral safe mode. Heaters at 400degreesC is not that safe :)

jbrazio, 2016-07-20:
You speak always from your context.. "I want", "my printer" etc. we can't start building features for individual needs.

thinkyhead, 2016-07-20:
> We can't start building features for individual needs.

@maukcc We might not make this a formal addition just now, but if you just want to get some action now when you do SD pause and SD resume, make the following changes, adjusting for where you want your XY to move:

```diff
+ float pause_xy[2];
  static void lcd_sdcard_pause() {
    card.pauseSDPrint();
    print_job_timer.pause();
+  pause_xy[X_AXIS] = current_position[X_AXIS];
+  pause_xy[Y_AXIS] = current_position[Y_AXIS];
+  enqueue_and_echo_commands_P(PSTR("M83\nG1 E-3 F8000\nM82\nG1 X0 Y200"));
  }

  static void lcd_sdcard_resume() {
+   cmd[50];
+   sprintf_P(cmd, PSTR("G1 X%i Y%i F8000\nM83\nG1 E3\nM82\n"), pause_xy[X_AXIS], pause_xy[Y_AXIS]);
+   enqueue_and_echo_commands(cmd);
    card.startFileprint();
    print_job_timer.start();
  }
```

Drayson80, 2016-08-09:
Very good, but is there a way to set it up into standard - so as a formal addition??
somewhere in the configuration "define pause position" and those lines? 
I would really like to see get CE conformity in terms of safety so from ym point of view a propper stop (not emergency stop!) and pause behavior should be mandatory...

thinkyhead, 2016-08-18:
> I would really like to see get CE conformity

Are you able to create a branche, apply changes, commit, and submit pull requests? That's how things usually get into Marlin. Otherwise, it sounds like you're more-or-less demanding that other generous volunteers do this on your behalf.

Drayson80, 2016-08-22:
I would love to do so as I would not depend on somebody else - but I have no clue about programming, and beleve me, it´s good so....
Anyway, thanks for your reply - I thought it´s not only about doing things but also recommending things... my fault...

thinkyhead, 2016-09-14:
> but also recommending things

True. With the understanding that suggestions may be put off indefinitely or utterly rejected. We already have this feature on the long-term plans list. But it's non-trivial, and will need to be tested on a wide variety of setups. Time is just very limited right now, and I don't even have a 3D printer at my disposal if I wanted to work on this. So it just has to wait.

tximy, 2017-01-18:
This is an great feature for multiple extruders and serial runout filament sensors not indicate the attached extruder for filament change, the pause is ok for it.

thinkyhead, 2017-04-02:
We now have this. It requires also enabling `M600`.

closed 2017-04-02 by thinkyhead