MarlinFirmware/Marlin #3931: a friend in trouble

boelle, 2016-05-31:
> The symptom is under-extrusion on some layers
> not quite randomly - but the pattern is not clear
> Using a 3DBenchy, we get underextrusion halfway up the hull and around the cabin pillars especially

any that can give pointers on where to look?

jbrazio, 2016-05-31:
This could be a multitude of hardware problems:
- Broken thermistor
- Power supply issues
- Stepper motor overheating

Just to name a few.

boelle, 2016-05-31:
will pass that on

forgot to ask what hardware he uses... could be the omca

Roxy-3DPrintBoard, 2016-05-31:
Actually...   the first place I would check is if the nozzle is a little bit plugged.  If he is using PLA, tell him to heat the nozzle to 235C and flush a lot (20 cm) of filament through it.   If he doesn't notice any inconsistencies it is probably not this causing the problem.   (Or possibly he has cleared the blockage with this high heat and volume.   If he switches between ABS and PLA occasionally, this can be a problem.)   

But if he does notice changes in how much is coming out, he can focus on getting this fixed.

boelle, 2016-05-31:
> Print temperature is at 230 degrees, in between underextruded layers I'm getting several centimetres of good layers.  No ABS.  The extruder is e3D Titan/v6 ht-end, made right, but did experience a jam which was surprising - the PTFE tube has since been replaced.   
> Main areas being checked are the stepper driver current, and I'm testing with reduced speed firmware controlled retractions F1800 --> F900

boelle, 2016-05-31:
http://imgur.com/LuMKZGl

Blue-Marlin, 2016-05-31:
Is the layer height a exact multiple of the z step length, or better full (not micro) step lenght?
Is `DISABLE_Z false`?

boelle, 2016-05-31:
> Layer height is 200 micron, 1600 steps/mm on Z

boelle, 2016-05-31:
> Here are my acceleration and jerk settings
> #define DEFAULT_MAX_ACCELERATION      {6000,6000,100,6000}  // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot.
> 
> #define DEFAULT_ACCELERATION          3000    // X, Y, Z and E max acceleration in mm/s^2 for printing moves
> #define DEFAULT_RETRACT_ACCELERATION  3000   //
> #define DEFAULT_XYJERK                20.0    // (mm/sec)
> #define DEFAULT_ZJERK                 10.0      // (mm/sec)
> #define DEFAULT_EJERK                 5.0    // (mm/sec)

thinkyhead, 2016-06-01:
If acceleration or jerk are set too high, he should expect to see grinding on the filament where the wheel grips it, so the filament isn't getting pushed properly. But he would tend to only see that a lot when using retraction. If he's using AUTORETRACT he should check the retract and recover speeds. Of course he can eliminate retraction as being a factor by turning off retraction in his slicer settings.

Q: Is he using 1.75mm filament or 3.0mm filament? There's more pressure with 3.0mm, a tendency to gather more dust.

Q: Does he use a "dust wipe" on the filament before it enters the nozzle? That can help eliminate impurities that may alight on the filament and cling by static charge.

boelle, 2016-06-01:
he reduced the acc on retract and it helped... will have a full report when he shows up on hangouts

he failed to mention its a core xy machine

thinkyhead, 2016-06-01:
> he failed to mention it's a core xy

Fortunately, the COREXY customizations don't affect the E axis at all, so this would apply just as well to any type of robot.

Roxy-3DPrintBoard, 2016-06-01:
> If acceleration or jerk are set too high, he should expect to see grinding on the filament where the wheel grips it, so the filament isn't getting pushed properly. 

> he reduced the acc on retract and it helped... 

Once the filament starts to slip on the hobbled bolt, filament dust and shavings start to fill up the space between the teeth.  He will never have 'full traction' until that is removed.  It is a judgment call what to do.  It probably depends on how hard his extruder is to take apart and put together.

My extruder is very difficult and time consuming to take apart and I have to re-calibrate the Z-Probe if I do that.   So I retract the filament as much as I can (with the nozzle hot) and if I do it quickly enough the hot and soft filament rolls across the teeth and can get much of the filament dust to stick to it.  Then I cut that off and in very difficult cases I paint SciGrip #4 on the filament and feed it in.   That is the glue (solvent) I use for PLA.  The remaining filament dust will stick to the filament and get feed into the nozzle to be melted.

closed 2016-07-16 by jbrazio