MarlinFirmware/Marlin #3935: Compile error for SAV MK1 (rcbugfix)

boelle, 2016-05-31:
when i compile i get this

```
In file included from C:\Users\Bo\AppData\Local\Temp\build8d9b268018e4650011bb6d082bdba173.tmp\sketch\Conditionals.h:264:0,

                 from C:\Users\Bo\AppData\Local\Temp\build8d9b268018e4650011bb6d082bdba173.tmp\sketch\Configuration_adv.h:707,

                 from C:\Users\Bo\AppData\Local\Temp\build8d9b268018e4650011bb6d082bdba173.tmp\sketch\Configuration.h:1178,

                 from C:\Users\Bo\Desktop\Marlin-RCBugFix\Marlin\Marlin.ino:37:

pins.h:143: error: #error "Unknown MOTHERBOARD value set in Configuration.h"
   #error "Unknown MOTHERBOARD value set in Configuration.h"

    ^

In file included from C:\Users\Bo\AppData\Local\Temp\build8d9b268018e4650011bb6d082bdba173.tmp\sketch\Configuration_adv.h:708:0,

                 from C:\Users\Bo\AppData\Local\Temp\build8d9b268018e4650011bb6d082bdba173.tmp\sketch\Configuration.h:1178,

                 from C:\Users\Bo\Desktop\Marlin-RCBugFix\Marlin\Marlin.ino:37:

SanityCheck.h:505: error: #error "HEATER_0_PIN not defined for this board."
   #error "HEATER_0_PIN not defined for this board."

    ^

#error "Unknown MOTHERBOARD value set in Configuration.h"
```

>  #ifndef MOTHERBOARD
>   #define MOTHERBOARD SAV_MKI
> #endif
>

boelle, 2016-05-31:
teensduino installed and have selected teensy 2 ++

Blue-Marlin, 2016-05-31:
How about __BOARD_SAV_MKI__?

boelle, 2016-05-31:
dooh of course

my only problem left is that the boards.txt is made for an old version of IDE.... and i'm not sure if it matters at all

http://reprap.org/wiki/SAV_MKI

step3...

who have a clue if it makes any difference if i use the custom boards.txt in the zip or i select teensy 2 ++

boelle, 2016-05-31:
if i just overwrite like the instructions say i get these errors

> Board teensy:avr:cdcsav doesn't define a 'build.board' preference. Auto-set to: AVR_CDCSAV
> Board teensy:avr:usbtiny1286 doesn't define a 'build.board' preference. Auto-set to: AVR_USBTINY1286
> Board teensy:avr:avr1091286 doesn't define a 'build.board' preference. Auto-set to: AVR_AVR1091286
> Board teensy:avr:usbtsav doesn't define a 'build.board' preference. Auto-set to: AVR_USBTSAV
> exec: "C:\\Users\\Bo\\Desktop\\arduino-1.6.8\\hardware\\teensy/../tools/": file does not exist
> Error compiling for board [BootloaderCDC]SAV-MkI.

boelle, 2016-05-31:
@thinkyhead are you an "expert" in arduino IDE ?

my board is based on the teensy so as per normal i install teensyduino.. but in ide 1.0.6 i just took a custom boards.txt and smaked in it teensy dir and my board showed up and and the teensy ones vent away...

in ide 1.6.8 that does not work so well and its a bit over my head what i should do

thinkyhead, 2016-06-01:
> are you an "expert" in arduino IDE ?

Not really! And although I have a Teensy 3.0 I haven't messed with it in a long while. Otherwise I would have more practice installing the Teensyduino support.

boelle, 2016-06-01:
yep, 

that part is easy enough

hard part is the way i used to install it

boelle, 2016-06-01:
hmmmm

i remember that roxy said her printrboard is based on same chip...

i can select teensy 2 ++ like the pins file for printrboard says and it compiles.

but selecting teensy 2 gives option to select mhz and keyboard layout etc.... what to do here?

@Roxy-3DPrintBoard have you tried to upload yet?

boelle, 2016-06-01:
> Opening Teensy Loader...
> 
> 
> Sketch uses 57.276 bytes (44%) of program storage space. Maximum is 130.048 bytes.
> Global variables use 2.922 bytes (35%) of dynamic memory, leaving 5.270 bytes for local variables. Maximum is 8.192 bytes.
> Teensy did not respond to a USB-based request to automatically reboot.
> Please press the PROGRAM MODE BUTTON on your Teensy to upload your sketch.

Got that far.... as per normal i remove the programming jumper and press the reset button...

thinkyhead, 2016-06-01:
Apparently @paulusjacobus ended up using [Flip](http://www.atmel.com/tools/FLIP.aspx) to get it working, as detailed a little at https://github.com/MarlinFirmware/Marlin/pull/3182#issuecomment-201260335.

boelle, 2016-06-01:
will have a look

what is in the way right now i'm just assuming is the teensy loader

when i remove that programming jumper and reset com port changes and it will accept upload directly

boelle, 2016-06-01:
hmm.. just left a comment there... 

the big difference is the bootloader... and i guess big time that i just have to disable that to get it going... could be wrong thou

boelle, 2016-06-01:
http://blog.lincomatic.com/?p=548

that one says it all i think.... i have the CDC bootloader

boelle, 2016-06-01:
https://github.com/scwimbush/Printrboard-HID-Arduino-IDE-Support

might be on to something here

boelle, 2016-06-01:
well i tried to upload as is.... 

but it stays on upload... no done msg or timeout

boelle, 2016-06-01:
dinner.... what could be preventing it from doing the upload right? missing upload protocol??

boelle, 2016-06-01:
i have a small prayer that @fmalpartida has some spare time to make a boards.txt or something that would make things work in ide 1.6.8

as is @repetier is the only thing i can run

tiagom62, 2016-06-02:
do you use the usbtinyisp or the bootloader to program the board?

paulusjacobus, 2016-06-02:
@Boelle I ended up compiling the Marlin files in Arduino IDE with as
board/port settings Teensy2++ and TinyUSB.
The Teensy loader popup appears but I ignored that and took the hex output
file from app data directory into the Flip programmer. There I had to
select all 4 check boxes (erase, check, verify ...) in order to flash the
hex file to the Rev F Printrbot board. I did set in Flip to board processor
to 16MHz (it's the default). This worked for me after 1 day of
experimenting, so I know the pain you feel.

Roxy-3DPrintBoard, 2016-06-02:
I wish we had an honest to goodness real PrinteRboard file for Arduino!

StephS, 2016-06-04:
The default bootloader in the AT90USB128 uses the FLIP programmer and bootloader, but some boards overwrite the bootloader with the LUFA CDC. When all else fails, use an ISP. It's an Atmel and Arduino issue.

closed 2016-06-11 by thinkyhead