MarlinFirmware/Marlin #3938: When does EEPROM get set to defaults?

edwilliams16, 2016-05-31:
If I use the same version of Marlin and change a parameter - say the xyjerk setting - in Configuration.h and upload the modified firmware, then the parameter loaded into memory does not change as the old value in EEPROM over-rides it.  M502 will bring the value in from Configuration.h and M500 will then stuff it into EEPROM.  Although a potential trap for the unwary, I see this is a feature not a bug.
   What I'm unclear about is what happens when I change firmware versions.  Are any of the old EEPROM settings kept?  Are they wiped and null until an M500 is done, or are the new "factory default" settings pushed into EEPROM -- or what?

KevMags, 2016-05-31:
All settings stored in EEPROM stay there until EEPROM is erased via Arduino or M500.

KevMags, 2016-05-31:
In the LCD menu, Restore Failsafe writes firmware values to EEPROM.

edwilliams16, 2016-06-01:
You didn't really address my specific question.  What happens when you change Marlin versions?

KevMags, 2016-06-01:
Uploading Marlin firmware does nothing to the EEPROM.  None of the default parameters get pushed into EEPROM.  On a cleared EEPROM the EEPROM stays "empty" unless you push a value into it.  The old values stay in the EEPROM unless cleared or overwritten by M500. The values stored in the EEPROM for such items as e-steps and accel will be used with the new firmware.   From what I can tell (and I'm not the sharpest knife in the drawer) the major EEPROM addresses used in all of the firmware versions are the same.  Items not used in the new firmware will be ignored and newer feature parameters (if changed via USB or LCD and stored) will be stored in unique addresses (chosen by Marlin) that don't conflict with older firmware.   I always clear the EEPROM before uploading new or modified firmware since any value stored in an EEPROM address will override firmware settings.

Blue-Marlin, 2016-06-01:
```
/**
 * configuration_store.cpp
 *
 * Configuration and EEPROM storage
 *
 * IMPORTANT:  Whenever there are changes made to the variables stored in EEPROM
 * in the functions below, also increment the version number. This makes sure that
 * the default values are used whenever there is a change to the data, to prevent
 * wrong data being written to the variables.
 *
 * ALSO: Variables in the Store and Retrieve sections must be in the same order.
 *       If a feature is disabled, some data must still be written that, when read,
 *       either sets a Sane Default, or results in No Change to the existing value.
 *
 */

#define EEPROM_VERSION "V23"
```

edwilliams16, 2016-06-01:
To make sure I understood the IMPORTANT/ALSO comments.  Is it saying that if the EEPROM version number is changed, the default values are written into the EEPROM, but leaving unknown/disabled feature values either unchanged or sensible.  Or is KevMags correct and the the EEPROM is untouched even if you change EEPROM Version?

  One context of all this is that there's a fork of Marlin supporting a digipot in the Rigidbot. This firmware allows you to set the stepper currents.  The question is if these get set appropriately (I think writing into EEPROM in the digipot itself), can it get screwed up on switching to mainstream Marlin.

thinkyhead, 2016-06-01:
> the default values are written into the EEPROM

⬆️  Whenever the version number stored in the EEPROM is different from the Marlin's current EEPROM version number (as when you flash a newer version of Marlin where the EEPROM has been changed) Marlin will write the "factory settings" to EEPROM. This will occur when you first boot into the updated Marlin.

KevMags, 2016-06-01:
Well obviously I am full of bad information.  Sorry .

edwilliams16, 2016-06-01:
Got it.   It works exactly how I hoped it would!  Thanks for your patience.

closed 2016-06-01 by edwilliams16