MarlinFirmware/Marlin #3940: Repetier can't recognize RC6

otarkhan, 2016-06-01:
I was working with RC4 and repetier host was working perfect, I updated my firmware to RC6 to solve the SD card problem but now Repetier can't recognize the new firmware, it says " Unknown printer firmware "

jbrazio, 2016-06-01:
@repetier what is required from our side to get fully recognized within Repetier Host ?

thinkyhead, 2016-06-01:
@otarkhan Perhaps this issue has been fixed already. Can you try the `RCBugFix` branch and see if the problem still exists?

lrpirlet, 2016-06-01:
@otarkhan <https://github.com/otarkhan> I am using RCBugFix with Repetier
Host and have no problem... I downloaded Repetier a few weeks ago when I
re-installed my computer... Since a communication has 2 ends maybe you want
to verify with a new version of Repetier???

repetier, 2016-06-01:
I had already similar reports. The host uses M115 and searches for "marlin" to detect it. I think you have changed that into "Marlin" making it a new identifier string. I have added "Marlin" to the new variants list in host for next release - until then it will not recognice and Mac version probably will never recognice the new spelling.

In general it is a bad idea to change output that could be useful. In repetier firmware we e.g. write out changes done on lcd so hosts know new values. Changing strings here would break existing hosts parsing such things. And you never know which host already parses which answer, so never change output formats once you have introduced them or expect some problems.

repetier, 2016-06-01:
Ok, I'm a bit confused after checking mac host test. There it was always "Marlin" so what does RC6 now return on M115?

Blue-Marlin, 2016-06-01:
It's
```
+#define MSG_M115_REPORT "FIRMWARE_NAME:Marlin ...
```
With a upper case 'M' since at least 2012-08-12 when language.h was introduced

jbrazio, 2016-06-01:
[image]

@repetier Mine (RCBugfix) is missing the URL and I don't know if besides "Marlin" it was supposed to have the version as I don't see Marlin's version anywhere.

thinkyhead, 2016-06-02:
Here's where [Repetier Host detects Marlin](https://github.com/repetier/Repetier-Host/blob/5707d0bd8304e3785b085dbb82a406ebf0bfc27a/src/RepetierHost/model/PrinterConnection.cs#L739-L744) for reference.

tiagom62, 2016-06-05:
I use repetier host with RC6 without issues, very strange.

otarkhan, 2016-06-06:
@lrpirlet @tiagom62 @dteeter maybe you're using windows or Linux version and I'm using Mac version?

repetier, 2016-06-06:
That little hint would have helped a bit. Anyhow that the firmware is not written at the bottom is not a problem and changes nothing in the host, except extruder/bed power which is now not multiplied by 2 to get into 0-255 range so you see only 50% of what you really have. Anyhow what is the exact complete response to M115 in the log? Knowing this I might be able to fix it.

lrpirlet, 2016-06-06:
@otarkhan @repetier Yes, I am using Windows, however the startup code in marlin RCBugFix dispays:
```
22:14:36.850 : N1 M110*34
22:14:36.850 : N2 M115*36
22:14:36.850 : N4 M114*35
22:14:36.866 : N5 M111 S6*98
22:14:36.866 : N6 T0*60
22:14:36.866 : N7 M20*22
22:14:36.882 : N8 M80*19
22:14:40.663 : echo:SD init fail
22:14:40.663 : ok
22:14:40.773 : FIRMWARE_NAME:Marlin 1.1.0-RCBugFix (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Pi3 3D LabteK EXTRUDER_COUNT:1 **UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
22:14:40.773 : ok
22:14:40.773 : N11 M220 `S100*8122:14:36.850 : N1 M110*34
```

lrpirlet, 2016-06-06:
@otarkhan @repetier the version is defined in Version.h...

```
/**
 * Marlin release version identifier
 */
#define SHORT_BUILD_VERSION "1.1.0-RCBugFix"

/**
 * Verbose version identifier which should contain a reference to the location
 * from where the binary was downloaded or the source code was compiled.
 */
#define DETAILED_BUILD_VERSION SHORT_BUILD_VERSION " (Github)"

/**
 * The STRING_DISTRIBUTION_DATE represents when the binary file was built,
 * here we define this default string as the date where the latest release
 * version was tagged.
 */
#define STRING_DISTRIBUTION_DATE "2016-04-27 12:00"

/**
 * @todo: Missing documentation block
 */
#define PROTOCOL_VERSION "1.0"

/**
 * Defines a generic printer name to be output to the LCD after booting Marlin.
 */
#define MACHINE_NAME "3D Printer"

/**
 * The SOURCE_CODE_URL is the location where users will find the Marlin Source
 * Code which is installed on the device. In most cases —unless the manufacturer
 * has a distinct Github fork— the Source Code URL should just be the main
 * Marlin repository.
 */
#define SOURCE_CODE_URL "https://github.com/MarlinFirmware/Marlin"

/**
 * Default generic printer UUID.
 */
#define DEFAULT_MACHINE_UUID "cede2a2f-41a2-4748-9b12-c55c62f367ff"

/**
 * The WEBSITE_URL is the location where users can get more information such as
 * documentation about a specific Marlin release.
 */
#define WEBSITE_URL "http://marlinfw.org"
```
In Configuration.h one can find
```C++
#if ENABLED(USE_AUTOMATIC_VERSIONING)
  #include "_Version.h"
#else
  #include "Version.h"
#endif
```
Note that I cannot find ``` USE_AUTOMATIC_VERSIONING ``` defined... So you MAY want to change that code in configuration.h to

```C++
//#if ENABLED(USE_AUTOMATIC_VERSIONING)
//  #include "_Version.h"
//#else
  #include "Version.h"
//#endif
```

tiagom62, 2016-06-07:
On windows 10.

20:10:28.352 : Printer reset detected - initalizing
20:10:28.353 : start
20:10:28.354 : echo:Marlin 1.1.0-RC6
20:10:28.357 : echo: Last Updated: 2016-04-24 12:00 | Author: (Tiago, default config)
20:10:28.357 : Compiled: May 15 2016

Im using Repetier-Host V1.6.1

lrpirlet, 2016-06-07:
@otarkhan @repetier I knew I did read that somewhere... please have a look at #2232... I think the problem is to compile on a Mac OS x or from Linux without first setting it correctly...

Do NOT ask how to setup correctly, read #2232... I have NOT tried it myself...

otarkhan, 2016-06-07:
@repetier It seems that this error is happening only with mac version, does this make any sense?

thinkyhead, 2016-06-08:
@tiagom62 That's the generic boot message. I assume Repetier and other hosts are looking at the output of `M115` and not the boot message.

@otarkhan What output do you see from the `M115` command?

otarkhan, 2016-06-08:
Finally, it's working.
 he problem was with the Baud rate, It was "250000" and when I changed it for " 112500" it worked

thank you all

jbrazio, 2016-06-08:
> problem was with the Baud rate, It was "250000" and when I changed it for " 112500" it worked

How did that fixed the issue ?
Sounds like something is wrong at higher serial bps.

thinkyhead, 2016-06-08:
Higher baud rates tend to be less reliable. But also, if the stepper ISR takes too long to finish then it could potentially block the serial interrupt, causing dropped characters in the serial input or output.

tiagom62, 2016-06-10:
@thinkyhead  I still expect the versioning to be correct, it was just to show that i am indeed running RC6.

thinkyhead, 2016-06-11:
If there is a bug here, it is related to comm issues generally, and we need to keep optimizing the `Stepper::isr` (and the functions it calls) to address that, especially.

closed 2016-06-11 by thinkyhead