MarlinFirmware/Marlin #3941: Request: change servo value in real time while moving/printing?

arthursw, 2016-06-01:
I am working on a graffiti draw bot (like [this one][1]) powered by the Marlin firmware. 

A servo presses the spray to paint the wall. I need to change the servo up/down value while the robot is painting (moving) to adapt the pressure (by hand, with the keyboard arrows) in real-time.

How can I change the Marlin firmware so that the robot takes my command (for example `M340 P3 S1700`) into account while it is moving (while executing a `G1 X1440.0 Y2300.0`)

  [1]: http://juerglehni.com/works/hektor/

http://arduino.stackexchange.com/questions/24785/marlin-change-servo-values-while-moving

Roxy-3DPrintBoard, 2016-06-01:
That is so cool!    I think I would be tempted to treat the servo that controls the spray can to be an extruder.    I'm not sure this would really work, but your G1 X1440.0 Y2300.0 command might be able to be changed into G1 X1440.0 Y2300.0 E10.0      You would have to cut into the code so instead of moving an Extruder stepper motor you are positioning the servo at a 'reasonable' place given the command.    

Because at that point, you can take company logo's and/or text and project them into 3D and use the standard Slic3r's to send this machine start to write on the wall.    

I'll bet others will have better ideas.  But if I wasn't able to consult with people, that is the approach I would be trying.

Blue-Marlin, 2016-06-01:
Write the command into your g-code file for your painting.

thinkyhead, 2016-06-02:
For configured servos in Marlin (via `NUM_SERVOS`), you use the `M280 P<servo> S<angle>` command from GCode to set the servo position.

If you want to attach more than 4 servos, you can skip telling Marlin about them because you'll know the pin numbers your servos are attached to. Then you would use `M226 P<pin> S<value>` to adjust these servos from GCode by setting the pins directly.

For producing paintings, paths for laser cutters and CNC machines, and drawing-style control, some people have written plugins for programs like [InkScape](https://inkscape.org/en/download/addons/), which convert the vectors in your InkScape file into GCode. If you wrote something like an InkScape plugin to generate GCode for your paintings, you would simply have it generate the`M280` or `M226` commands automatically to adjust the spray where needed.

jbrazio, 2016-07-11:
Closing due to inactivity.

closed 2016-07-11 by jbrazio