MarlinFirmware/Marlin #3945: Request: Axis Homing dryrun

GreatGrizzly, 2016-06-02:
Something that might be helpful interim solution to auto mesh bed leveling would be to allow a G28 that moves the print head to the origin of a particular axis but not set that point as the origin.  I propose We make G28 <AXIS>0 and G28 <AXIS>  do different things. G28 <AXIS> will move to an endstop, but not set that as the origin. While G28 <AXIS>0 will behave normally. A use for this could be as follows:

;Home All Axis, and set the origins
G28 X0 Y0 Z0

;Combine mesh leveling of four points with proposed function (move until switch is triggered, dont set origin)
G29 S1 ;Go to first point
G28 Z ; Lower head until probe is triggered, just like a normal G28 Z0 call, BUT don't save new origin
G29 S2 ;Save the current printhead height which might be different from true origin.
G28 Z ;continued...
G29 S2
G28 Z
G29 S2
G28 Z
G29 S2
G28 Z
G29 S2

Blue-Marlin, 2016-06-02:
G30?

GreatGrizzly, 2016-06-02:
I am not familiar with G30, is there an example that does this? The [gcode wiki](http://reprap.org/wiki/G-code#G30:_Single_Z-Probe) is raising more questions then answers.

Roxy-3DPrintBoard, 2016-06-02:
There is no reason G30 can't work on Delta's.   It would be trickier if G30 allowed you to specify an X & Y but it doesn't (currently).  And actually...  Now that the movement for Delta's is all figured out at a lower level, the higher level functions don't need to worry about calculating the tower heights.

I can't test this right now, but I'm like 75% sure that if you just take that 
~~~~
    #endif //!Z_PROBE_SLED
  #endif // AUTO_BED_LEVELING_FEATURE
~~~~
at the end of the gcode_G30() and move it up to before the gcode_G30() it will be close to compiling clean so you can check it out and see if it works for you.

thinkyhead, 2016-06-04:
The intent of `G30` is to allow host software to probe the bed. But, I suppose it's okay to repurpose it so that when the MBL probing procedure is underway it sends the probe result to MBL. There's no other reason why you would do `G30` in the middle of MBL.

GreatGrizzly, 2016-06-07:
@Roxy-3DPrintBoard 
Until we have Automatic Mesh Bed Leveling, I think any command that allows Z probing will work. MBL allows you to store the location of hot end with G29 S2. If a G30 is ran, the probe is sitting at the right height after the probe is done, which can be read by MBL code. Originally I was going to use G28 Z, but that sets the new height to zero, and thus the point is saved as 0 when G29 S2 is issued.

Roxy-3DPrintBoard, 2016-06-08:
> The intent of G30 is to allow host software (or troubleshooters) to probe the bed. But, I suppose it's okay to repurpose it so that when the MBL probing procedure is underway it sends the probe result to MBL. There's no other reason why you would do G30 in the middle of MBL.

@ThinkyHead   If the G29 can do a single Z Probe operation, do we need G30?   I'm wondering how much of a legacy we have built up.    Because it would be nice to have G30 actually do things that compliment G29.   (We can chit chat in the Google Chat window about what I'm thinking.)

thinkyhead, 2016-06-08:
>  If the G29 can do a single Z Probe operation, do we need G30?

The reason for this topic is because we don't currently have automated mesh leveling, so this is just one proposal to accomplish it. But once automation is done for all forms of leveling (and a manual option for all forms of leveling), this concept will be obviated and automated mesh leveling will work in the same manner as automated grid and 3-point leveling.

Roxy-3DPrintBoard, 2016-06-08:
Yes.   I understand and agree.   From my perspective, the reason I'm asking if we can re-claim G30 is because we would have 3 GCodes all together.   You do a G28 to home and setup the printer.  You do a G29 to prepare the auto bed leveling system.   And I wanted to use G30 to fine tune and test the Auto Bed Leveling system.     (G29 would allow a single Z-Probe at a specified location)

thinkyhead, 2016-06-08:
However we choose to extend or repurpose `G30`, it should continue to work in the manner that it does in other firmware (if any other firmware implements it), and as documented in the RepRap wiki.

Roxy-3DPrintBoard, 2016-06-08:
I think that means this idea is a bad one.  Too many other firmware's have already implemented G30 as a Single Z-Probe.

GreatGrizzly, 2016-06-10:
@Roxy-3DPrintBoard  I did what you suggested and that didnt work. It figures that it wouldnt be that easy...

Roxy-3DPrintBoard, 2016-06-10:
Well...   Sit tight.   In a month there will be an Automatic High Resolution version of Mesh Bed Leveling.   And as it turns out, I'm working on the single Z-Probe part of that this morning.   The reason this is important is the Mesh gets loaded out of EEPROM, but we still need a way to accurately adjust for different pieces of glass.

GreatGrizzly, 2016-06-10:
I am really excited. :-) I wish I was knowledgeable enough about this stuff
to better assist you.

thinkyhead, 2016-06-11:
> In a month

I can't wait that long. Can I help out?

Roxy-3DPrintBoard, 2016-06-11:
> I can't wait that long. Can I help out?

You already are helping out by getting us close to Golden!    I wasted 1/2 a day yesterday because I didn't understand that you can't test for:
~~~~
f=NAN;
if (f==NAN){
   this code never executes;
}
~~~~
I'm thinking if this check is done in software, that adds a lot of expense to every floating point comparison.   If it is done in hardware, it might not be so bad.  But given it is an 8-Bit CPU, I'm betting it is done in software.

boelle, 2017-04-26:
@thinkyhead do we not have automatic mesh leveling ? if so we could close this one

closed 2018-03-16 by thinkyhead