MarlinFirmware/Marlin #3949: Best Practise: Marlin Buffer

openhardwarecoza, 2016-06-03:
I Tagged @thinkyhead over the last two days into a few threads on our host application (Laserweb)

However, i havent had any responses.  

Please review thread https://github.com/openhardwarecoza/LaserWeb3/issues/2  and reply - grbl and smoothie already got back to us, just Marlin holding us up a little...

boelle, 2016-06-03:
@thinkyhead has more than enough on his plate i think

currently a LOT of bugfixing is been done and next step is to split up the code in the sub folders to make it more easy to manage

bobc, 2016-06-03:
That's a long thread, it is not quite clear what is being asked. I _think_ the question is : "how many gcode lines can a host send to Marlin before waiting for an 'ok' response?".

jbrazio, 2016-06-03:
The answer is "one". :-)

openhardwarecoza, 2016-06-03:
Thanks.   @HakanBastedt that means we can use the code I wrote for
Smoothieware,  just slightly different error handling (Marlin prints
Resend,  everything else prints 'rs'  for example.

jbrazio, 2016-06-03:
@openhardwarecoza when you require feedback from Marlin team it's better to tag @MarlinFirmware/general-maintainers

openhardwarecoza, 2016-06-03:
(:Thanks,  will do!

closed 2016-06-03 by jbrazio

thinkyhead, 2016-06-04:
Only getting through about 8-10 issues a day, but picking up the pace…