MarlinFirmware/Marlin #3951: Looking for language maintainers

boelle, 2016-06-03:
- [ ] English (en)
- [ ] Aragonese (an): @danielmartinez 
- [ ] Basque-Euskera (eu)
- [ ] Bulgarian (bg)
- [ ] Catalan (ca)
- [ ] Chinese (cn)
- [ ] Czech (cz): @clexpert, @petrzjunior
- [ ] Danish (da): @boelle 
- [ ] Dutch (nl): @paulusjacobus 
- [ ] Finnish (fi)
- [ ] French (fr)
- [ ] Galician (gl): @rafacouto
- [ ] German (de): @CONSULitAS
- [ ] Italian (it): @barus93
- [ ] Japanese (kana)
- [ ] Japanese (kana_utf8)
- [ ] Polish (pl)
- [ ] Portuguese (pt): @jbrazio
- [ ] Portuguese (pt-br)
- [ ] Portuguese (pt-br_utf8)
- [ ] Portuguese (pt_utf8): @jbrazio
- [ ] Russian (ru)
- [x] Spanish (es): @emartinez167, @RicardoGA, @fbarcenas

boelle, 2016-06-03:
We made a new topic so that i can tick off when a language is done

PLEASE... keep on topic and dont start a book long debate. If needed create a new issue to keep this one short and clean

boelle, 2016-06-03:
@esenapaj

can you do Japanese?

boelle, 2016-06-03:
@clexpert, @petrzjunior

Czech language file is 21 days old but english is 16... are you sure that you have compared the 2 files close and got everything in there?

boelle, 2016-06-03:
@MatixYo can you do Polish?

boelle, 2016-06-03:
@ivan-galvez can you do Basque-Euskera ?

boelle, 2016-06-03:
@blubbfish can you do Catalan?

boelle, 2016-06-03:
@tommil can you do Finnish ?

blubbfish, 2016-06-03:
I only speak English and German

boelle, 2016-06-03:
oki... i just used the history on github to figure who did what language

blubbfish, 2016-06-03:
Interresting, i never did any with catalan^^ But i can also did german translation

boelle, 2016-06-03:
hehe... my search skills prob suck.... so if you can help me out who did the language files i can ask the right people

So i can add you to german?

boelle, 2016-06-03:
@gege2b can you do French?

blubbfish, 2016-06-03:
that will be fine :)

boelle, 2016-06-03:
ok, you are added

esenapaj, 2016-06-03:
>can you do Japanese? and maybe Chinese ?

I can only Japanese.
About linguistic structure, Japanese and Chinese are different absolutely.
About letter(Kanji), those were analogous until about 70 years ago, but those are half different currently.

And, if I'll be integrated into language team, I can't perform intricate giving and taking, and I can't discussion in real-time like a chat.
Because my skill of English is exceedingly poor.
For example, I've spent over 50 minutes for writing this response that it has only 6 lines.

tommil, 2016-06-03:
Yes I can do Finnish.

petrzjunior, 2016-06-03:
@boelle 
> Czech language file is 21 days old but english is 16... are you sure that you have compared the 2 files close and got everything in there?
Damn, you're right, unspotted changes. First job for Czech section of lang-team. Ready ASAP :smile:

Roxy-3DPrintBoard, 2016-06-03:
> And, if I'll be integrated into language team, I can't perform intricate giving and taking, and I can't discussion in real-time like a chat.
> Because my skill of English is exceedingly poor.
> For example, I've spent over 50 minutes for writing this response that it has only 6 lines.

Understood.   But reading English text is not as difficult, right?   Because we should be able to give you a good description of what is trying to be communicated, and under what circumstances it happens.

esenapaj, 2016-06-04:
>reading English text is not as difficult, right?

I'm sorry. No, it isn't. And It all depends on circumstances.
About your message,

>Understood.

I needed about 5 minutes for understanding that what does the word means in the message.

>But reading English text is not as difficult, right? 

I understand this part immediately.

> Because we should be able to give you a good description of

I understand this part by a minute.

>what is trying to be communicated, 

I understand this part by over 2 hours.

>and under what circumstances it happens.

I understand this part by about 15 minutes.

But when all parts of the sentence are combined to

>Because we should be able to give you a good description of what is trying to be communicated, and under what circumstances it happens.

, I've no idea what does this combined parts would like to say.

As a result, I spent about 4 hours for trying to read your short message and writing this response, nevertheless I wasn't able to understand that what you'd like to tell me.
I'm very sorry.

Roxy-3DPrintBoard, 2016-06-04:
OK!

We can be careful.        We can keep ideas very small.         That is not a problem!

Remember!   The LCD Messages need to be short.

So we can communicate that way.      We value your time!!!

esenapaj, 2016-06-05:
Thank you for all the kindness you've shown me.

gege2b, 2016-06-05:
Hi all, @boelle : OK for french translations

jbrazio, 2016-06-05:
@gege2b you should receive an invitation shortly.

jbrazio, 2016-06-05:
@esenapaj check your email. I have invited you to join the organization. You only need to translate Japanese.

Roxy-3DPrintBoard, 2016-06-05:
@esenapaj  We will make the English very simple.

tnw513, 2016-06-06:
I can translate strings into Italian, if needed

esenapaj, 2016-06-06:
@jbrazio 
I've joined.
I'm thankful for your explanation. And sorry.
I confirmed your mail 2 days ago, but I hesitated because I didn't rightly know that point.

@Roxy-3DPrintBoard 
I'm thankful to you.
I also will endeavor.

psavva, 2016-06-12:
Please add Greek, and I'll have it translated.

thinkyhead, 2016-06-12:
@psavva Would you do:
-  `el` — Greek
- `el-GR` – Greek (Greece)

jbrazio, 2016-06-13:
@tnw513, @psavva I've invited you to the team.

psavva, 2016-06-13:
Both el and el-GR will be identical.
I'll add both.

The language code 'el' ISO 639-1 is our base.
The region code 'GR' is Greece, which happens to be the only one.

Maybe Cyprus gets an official language code in the future 😃

tnw513, 2016-06-13:
@jbrazio thank you

Blue-Marlin, 2016-06-13:
Doesn't you need a special font? Is there a character based display out there with a Greek font?

robimarko, 2016-06-13:
I will do Croatian.
Are croatian specific characters:"č, ć, dž, đ, š, ž, lj, nj. " supported?

boelle, 2016-06-17:
are danish the only one comepleted ?

tnw513, 2016-06-17:
Italian is up-to-date #4029

thinkyhead, 2016-06-20:
> croatian specific characters

@robimarko Only some of those characters are supported. You'll need to test with your own display.

thinkyhead, 2016-06-20:
Language | Last Update
---------|------------
language.h | 7 hours
language_an.h | 2 months
language_bg.h | 2 months
language_ca.h | 2 months
language_cn.h | 2 months
language_cz.h | 13 days
language_da.h | 17 days
language_de.h | 2 months
language_en.h | a month
language_es.h | 2 days
language_eu.h | 2 months
language_fi.h | 2 months
language_fr.h | 2 months
language_gl.h | 17 days
language_hr.h | 3 days
language_it.h | 7 days
language_kana.h | a month
language_kana_utf8.h | a month
language_nl.h | 2 months
language_pl.h | a month
language_pt-br.h | 2 months
language_pt-br_utf8.h | 2 months
language_pt.h | a month
language_pt_utf8.h | a month
language_ru.h | 2 months
language_test.h | 2 months

psavva, 2016-07-03:
Hi, 

I need some help with the Mapper for the Greek Language.
Besides the actual translations, I also needed to include a font to print out UTF8...

So, I have the following problem.
Firstly, How do I debug my code?
How do I see program flow? How can I print debug messages, somewhere?

Now the problem.
I'm trying to display "ΑΒΓΔΕ" on the display, but instead it's printing "ώΑώΒώΓώΔώΕ"
I am getting the "0xce" character from my font.

This is my mapper function.

Any help on "how to"

#elif ENABLED(MAPPER_CECF)

  char charset_mapper(char c) {
    static uint8_t utf_hi_char; // UTF-8 high part
    static bool seen_ce = false;
    uint8_t d = c;
    if ( d >= 0x80 ) { // UTF-8 handling
      if ( (d >= 0xc0) && (!seen_ce) ) {
        utf_hi_char = d - 0xce;
        seen_ce = true;
        return 0;
      }
      else if (seen_ce) {
        d &= 0x3f;
        #ifndef MAPPER_ONE_TO_ONE
          HARDWARE_CHAR_OUT((char)pgm_read_byte_near(utf_recode + d + (utf_hi_char << 6) - 0x20));
        #else
          HARDWARE_CHAR_OUT((char)(0x80 + (utf_hi_char << 6) + d)) ;
        #endif
      }
      else {
        HARDWARE_CHAR_OUT('?');
      }
    }
    else {
      HARDWARE_CHAR_OUT((char) c );
    }
    seen_ce = false;
    return 1;
  }

Blue-Marlin, 2016-07-03:
You could try to define the hex constants as uint (0x80 -> 0x80u)
```
#elif ENABLED(MAPPER_CECF)
 char charset_mapper(char c) {
  static uint8_t utf_hi_char; // UTF-8 high part
  static bool seen_ce = false;
  uint8_t d = c;
  if ( d >= 0x80u ) { // UTF-8 handling
    if ( (d >= 0xc0u) && (!seen_ce) ) { //Did we see a font family but are not already in
      utf_hi_char = d - 0xceu; // utf_hi_char should be 0 or 1 now
      seen_ce = true; // last char seen was in the area we have a mapper for.
      return 0; // no chars printed
    }
    else if (seen_ce) { // did we see the family before
      d &= 0x3fu; // CE and CF map to a 0x3f big block
      #ifndef MAPPER_ONE_TO_ONE
        HARDWARE_CHAR_OUT((char)pgm_read_byte_near(utf_recode + d + (utf_hi_char << 6) - 0x20u));
      #else
        HARDWARE_CHAR_OUT((char)(0x80u + (utf_hi_char << 6) + d)); // set bit 8 to 1
                                                                   // utf_hi_char is bit 7
                                                                   // d the pointer into the block
      #endif
    }
    else {
      HARDWARE_CHAR_OUT('?'); // not a family we have a mapper for
    }
  }
  else {
    HARDWARE_CHAR_OUT((char) c ); // directly print a char < 0x80
  }
  seen_ce = false; // is printed - reset.
  return 1; // one char printed
}
```

How do your string defines look like?
It should simply look like
```
#define MSG_TEST "ΑΒΓΔΕ"
```

psavva, 2016-07-04:
Thank you for your advice @Blue-Marlin 

I have however found a different bug.

The way I figured is that my mapper simply was not getting executed. 
My assumption was right.

I have moved MAPPER_NONE below my mapper, and voila, Greek is now displaying as expected!

There is somewhere in the code which is Enabling MAPPER_NONE, even though it's not defined in my language files.

This is the change I did
utf_mapper.h

```
#if ENABLED(MAPPER_CECF)

  char charset_mapper(char c) {
    static uint8_t utf_hi_char; // UTF-8 high part
    static bool seen_ce = false;
    uint8_t d = c;
    if ( d >= 0x80u ) { // UTF-8 handling
      if ( (d >= 0xc0u) && (!seen_ce) ) {
        utf_hi_char = d - 0xceu;
        seen_ce = true;
        return 0;
      }
      else if (seen_ce) {
        d &= 0x3fu;
        #ifndef MAPPER_ONE_TO_ONE
          HARDWARE_CHAR_OUT((char)pgm_read_byte_near(utf_recode + d + (utf_hi_char << 6) - 0x20));
        #else
          HARDWARE_CHAR_OUT((char)(0x80u + (utf_hi_char << 6) + d)) ;
        #endif
      }
      else {
        HARDWARE_CHAR_OUT('?');
      }
    }
    else {
      HARDWARE_CHAR_OUT((char) c );
    }
    seen_ce = false;
    return 1;
  }

#elif ENABLED(MAPPER_NON)

  char charset_mapper(char c) {
    HARDWARE_CHAR_OUT( c );
    return 1;
  }
```

We we just need to understand why and where it's been defined?
perhaps an issue with if ENABLED()
???

AnHardt, 2016-07-04:
Sorry.

https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/language_en.h#L33_L35

```diff
-#if DISABLED(MAPPER_NON) && DISABLED(MAPPER_C2C3) && DISABLED(MAPPER_D0D1) && DISABLED(MAPPER_D0D1_MOD) && DISABLED(MAPPER_E382E383)
+#if DISABLED(MAPPER_NON) && DISABLED(MAPPER_C2C3) && DISABLED(MAPPER_D0D1) && DISABLED(MAPPER_D0D1_MOD) && DISABLED(MAPPER_E382E383)&&DISABLED(MAPPER_CECF)
  #define MAPPER_NON         // For direct ascii codes
#endif

```

psavva, 2016-07-05:
@AnHardt, awesome.

Thank you :)

boelle, 2016-08-11:
has anything been added to english since 20 Jun?

just wanted to make sure everything up to date is checkmarked and those that are not will have the checkmark removed again until they are updated

tnw513, 2016-08-11:
@boelle Yes, some strings. Italian language is up to date :)

rafacouto, 2016-08-11:
@boelle Check this useful bash command to look for changes from a date (YYYY-MM-DD): 

`export DATE='2016-06-20' ; git diff $(git log --since=${DATE}T00:00:00 | grep '^commit ' | tail -n1 | cut -d' ' -f2) language_en.h`

Powered by GNU utils! :)

thinkyhead, 2016-08-11:
…And I'm sure we can concoct a shell script to get all the English strings into a keyed-array, then run over the other language files to find those that are either missing or are still in English. @jbrazio has made a good start with [buildroot/share/scripts/findMissingTranslations.sh](https://github.com/MarlinFirmware/Marlin/tree/RC/buildroot/share/scripts/findMissingTranslations.sh).

petrzjunior, 2016-08-12:
@thinkyhead Since the last week, Windows Anniversary update got out and win10 user will be able to run bash scripts in the Ubuntu interpreter, right in the windows OS. I will test it ASAP, hopefully no more abusing git Shell as the only bash interpreter on win. :smile:

boelle, 2016-08-14:
danish updated @thinkyhead just need to merge my PR

@rafacouto that command does not work in win 8.1 :-P

@CONSULitAS, @blubbfish do you have time to update German?

boelle, 2016-08-14:
@thinkyhead last update table.... could we somehow mix that with the first post ?

would make it easy to see if english is updated and if the language of choice is behind etc

tnw513, 2016-08-14:
@barus93 is still an active maintainer for the italian translation?

boelle, 2016-08-14:
good you mention it... What i was after was getting all the major languages updated...
danish is not major but i cant hunt people if i don't do my part

ie practice what you preach :-D

boelle, 2016-08-14:
but yes if we dont hear from @barus93 we can remove him... but then italian will be 100% yours... always good with at least 2 persons pr language

boelle, 2016-08-14:
@gege2b can you check if french is up to date?

boelle, 2016-08-14:
and @emartinez167, @RicardoGA, @fbarcenas

can you check spanish?

tnw513, 2016-08-14:
@boelle I don't want it :-D
I've asked because I've some question about how to optimize some Italian strings and there aren't other Italian maintainers.

emartinez167, 2016-08-14:
Will do.

Regards,

Ernesto

RicardoGA, 2016-08-15:
@boelle i just check it and made a pull request

zenmetsu, 2016-08-17:
Which branch should we work against when updating the languages?  I'll go ahead and update language_ru.h

tnw513, 2016-08-17:
@zenmetsu RCBugFix ;)

[84 more replies]

closed 2018-05-11 by boelle