MarlinFirmware/Marlin #3953: Add 'percentage printed' to full graphic lcd screen

maukcc, 2016-06-03:
Change the main screen of a graphic display to make the printed percentage visible.
In dogm_lcd_implementation.h change

    if (IS_SD_PRINTING) {
      // Progress bar solid part
      u8g.drawBox(55, 50, (unsigned int)(71.f * card.percentDone() / 100.f), 2 - (TALL_FONT_CORRECTION));
    }

    u8g.setPrintPos(80,48);
    uint16_t time = print_job_timer.duration() / 60;
    if (time != 0) {
      lcd_print(itostr2(time/60));
      lcd_print(':');
      lcd_print(itostr2(time%60));
    }

to

    if (IS_SD_PRINTING) {
      // Progress bar solid part
      u8g.drawBox(55, 50, (unsigned int)(71.f * card.percentDone() / 100.f), 2 - (TALL_FONT_CORRECTION));
    // Percent Complete
      u8g.setPrintPos(55,48);
      u8g.print(itostr3(card.percentDone()));
      u8g.print("%"); 
    }

    u8g.setPrintPos(85,48);
    uint16_t time = print_job_timer.duration() / 60;
    if (time != 0) {
      lcd_print(itostr2(time/60));
      lcd_print(':');
      lcd_print(itostr2(time%60));
    }

lavato, 2016-06-06:
Approximate time left is probably what we all want. :)

jbrazio, 2016-07-16:
I will close this request as we already have a progress bar.
Nevertheless we keep it on the back of our heads if we do some kind of UI redesign.

closed 2016-07-16 by jbrazio

maukcc, 2016-07-18:
Please keep this open until it is merged

jbrazio, 2016-07-18:
I didn't say we will implement it, I said `IF` we redesign the UI we will take into consideration the percentage as a visual queue instead of a progress bar.

maukcc, 2016-07-19:
It is not the one or the other, its both. I am still trying to figure out why the percentage was taken out in the 1st place.

thinkyhead, 2016-07-20:
> why the percentage was taken out in the 1st place

I don't know why it was changed. Search the git history and see when it occurred.
Maybe there's a PR or discussion about it.

COOLover, 2017-07-27:
Hello,
sorry for my english.
I Would like to add the % information, but can you help me with that?
I found this code on ultralcd_impl_dogm.h
`    if (IS_SD_PRINTING) {

      //
      // Progress bar solid part
      //

      if (PAGE_CONTAINS(50, 51 - (TALL_FONT_CORRECTION)))     // 50-51 (or just 50)
        u8g.drawBox(
          PROGRESS_BAR_X + 1, 50,
          (uint16_t)((PROGRESS_BAR_WIDTH - 2) * card.percentDone() * 0.01), 2 - (TALL_FONT_CORRECTION)
        );

      //
      // SD Percent Complete
      //

      #if ENABLED(DOGM_SD_PERCENT)
        if (PAGE_CONTAINS(41, 48)) {
          // Percent complete
          u8g.setPrintPos(55, 48);
          u8g.print(itostr3(card.percentDone()));
          u8g.print('%');
        }
      #endif
    }`

what i must to write, that i get the text information?

Thanks!

SohaibSiddique, 2020-06-25:
it's late but I need this function for marlin 2.0.5.3, I try but did not the exact file,

JRitmeester, 2021-01-01:
Also looking for this feature. Cannot find a simple way to do this, which is weird.