MarlinFirmware/Marlin #3967: Dual Z motor connected to one driver broken

fabtopia, 2016-06-06:
I tried to upgrade my Prusa I3 from 1.0.2 to 1.1.0-RC6 but there was a problem. The Z axe motors which are both parallel connect tot de same stepper driver, are starting to murmur and not moving at some speeds. I checked al the way down and 1.0.2-1 was the last versie without the problem.
What happend?

My board has no option to split the two z motors over two stepper drivers.

Roxy-3DPrintBoard, 2016-06-06:
You may need to turn up the power on the Step Stick to have enough 'Juice' to drive both motors.  You may also have your Max Z Feedrate (and acceleration) set too high.   I'm extremely skeptical this is a problem with RC6.  My guess is you just saw it first with RC6.   

Look for the little potentiometer on  the step stick.  Be very gentle with it as you turn it.  It is very easy to damage the part if you turn it past the end.   If I remember correctly, you want to turn it clockwise to increase the drive power.   And if that doesn't fix things, consider bumping down your acceleration and feedrate.

fabtopia, 2016-06-06:
@Roxy-3DPrintBoard Why is it that when i put my 1.0.2 back in all problems are gone without changing te stepper driver. I checked the Max Z Feedrate. They are the same between 1.0.2 en 1.1.0-RC1  till RC6.

fabtopia, 2016-06-06:
I noticed that between 1.0.2-1 and 1.1 there are extra files stepper_indirection.h and cpp so i thing there are some changes.

Roxy-3DPrintBoard, 2016-06-06:
There have been some changes to Stepper.cpp but they are very minor.  (That is not to say they can't cause problems if you are right on the edge.)    Mostly, this was done to speed up the code a little bit and to instantiate it as a 'Singleton'.   The changes SHOULD be transparent to almost everybody.l

With that said...  Can you try turning up the power on your Z-Axis to get both motor stepping together?   If that fixes the problem, it is probably the right thing to do.

lrpirlet, 2016-06-06:
@fabtopia Could you please verify if the motor (or both motor) are running very slowly? Feedrate may NOT have been set correctly... It happened once to me and I was able to see the axis turning only looking at a mark done on the axis...

fabtopia, 2016-06-06:
Oke. I will try both and get back to it.

Roxy-3DPrintBoard, 2016-06-06:
> Feedrate may NOT have been set correctly... It happened once to me and I was able to see the axis turning only looking at a mark done on the axis...

Did you do a M502 followed by a M500 to initialize your EEPROM settings?  If not, you maybe picking up bad feedrate values out of the EEPROM when you reset the machine.    If you have not done this, do this first and see if the problem still exists.

fabtopia, 2016-06-06:
@Roxy-3DPrintBoard I turned the stepper up a little (0.5 V) bit but no difference. 

To be more precise When i select Prepare->Move axes from the menu and select 0.1mm with 1.02 and 1.1.0-rc6 both are ok. The 1mm Works fine with 1.0.2, with 1.1.0-RC6 They doe move a bot and then murmur.

@lrpirlet Current settings in 1.0.2
//// MOVEMENT SETTINGS
#define NUM_AXIS 4 // The axis order in all axis related arrays is X, Y, Z, E
#define HOMING_FEEDRATE {50*60, 50*60, 4*60, 0}  // set the homing speeds (mm/min)

// default settings

/*** iTopia ***/
//#define DEFAULT_AXIS_STEPS_PER_UNIT   {78.7402,78.7402,200.0*8/3,760*1.1}  // default steps per unit for Ultimaker
//#define DEFAULT_MAX_FEEDRATE          {500, 500, 5, 25}    // (mm/sec)
//#define DEFAULT_MAX_ACCELERATION      {9000,9000,100,10000}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot.
#define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,4000,590}  // default steps per unit for Ultimaker
#define DEFAULT_MAX_FEEDRATE          {300, 300, 3, 45}    // (mm/sec)
#define DEFAULT_MAX_ACCELERATION      {1000,1000,50,500}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot.
/*** iTopia ***/

/*** iTopia ***/
//#define DEFAULT_ACCELERATION          3000    // X, Y, Z and E max acceleration in mm/s^2 for printing moves
//#define DEFAULT_RETRACT_ACCELERATION  3000   // X, Y, Z and E max acceleration in mm/s^2 for retracts
#define DEFAULT_ACCELERATION          1000    // X, Y, Z and E max acceleration in mm/s^2 for printing moves
#define DEFAULT_RETRACT_ACCELERATION  1000   // X, Y, Z and E max acceleration in mm/s^2 for retracts
/*** iTopia ***/

1.1.0-RC6
//*** BEGIN iTopia
//#define HOMING_FEEDRATE {50*60, 50*60, 4*60, 0}  // set the homing speeds (mm/min)
#define HOMING_FEEDRATE {50*60, 50*60, 4*60, 0}  // set the homing speeds (mm/min)
//*** END iTopia

// default settings

//*** BEGIN iTopia
//#define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,4000,500}  // default steps per unit for Ultimaker
//#define DEFAULT_MAX_FEEDRATE          {300, 300, 5, 25}    // (mm/sec)
//#define DEFAULT_MAX_ACCELERATION      {3000,3000,100,10000}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot.
#define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,4000,590}  // default steps per unit for Ultimaker
#define DEFAULT_MAX_FEEDRATE          {300, 300, 5, 45}    // (mm/sec)
#define DEFAULT_MAX_ACCELERATION      {1000,1000,50,500}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot.
//*** END iTopia

//*** BEGIN iTopia
//#define DEFAULT_ACCELERATION          3000    // X, Y, Z and E acceleration in mm/s^2 for printing moves
//#define DEFAULT_RETRACT_ACCELERATION  3000    // E acceleration in mm/s^2 for retracts
//#define DEFAULT_TRAVEL_ACCELERATION   3000    // X, Y, Z acceleration in mm/s^2 for travel (non printing) moves
#define DEFAULT_ACCELERATION          1000    // X, Y, Z and E acceleration in mm/s^2 for printing moves
#define DEFAULT_RETRACT_ACCELERATION  1000    // E acceleration in mm/s^2 for retracts
#define DEFAULT_TRAVEL_ACCELERATION   1000    // X, Y, Z acceleration in mm/s^2 for travel (non printing) moves
//*** END iTopia

fabtopia, 2016-06-06:
@Roxy-3DPrintBoard I did try the M502 and M500 but the problem still exits.

Roxy-3DPrintBoard, 2016-06-06:
OK...   ThinkyHead will probably have some comments when he gets online later today.

lrpirlet, 2016-06-06:
OK, I think @thinkyhead will ask you to compile with ```#define DEBUG_LEVELING_FEATURE``` then issue M111 S255 and report the log..

I also, would suggest to do what I did... make  a mark on the Z motor axis (I used the sticking part of a "Post-It" fold in two around the axis to make a small flag... so that you can see the axis turning).

Then ussue the failing command and let the motor murmur for at least 15 second while oserving the post-it... Does the axis turn???

I bet it turns very slowly...

Roxy-3DPrintBoard, 2016-06-06:
> OK, I think @thinkyhead will ask you to compile with #define DEBUG_LEVELING_FEATURE then issue M111 S255 and report the log..

Yeah!  Almost for sure that will be the case!  Please do that as soon as you can.

RicardoGA, 2016-06-07:
@fabtopia try changing the feedrate to 2 or 2.5 and check if the problem persist

thinkyhead, 2016-06-07:
I agree that the suggestion to use a lower `DEFAULT_MAX_FEEDRATE` for Z is the right idea. I had to lower mine to 2.5 to avoid a slipping Z axis.

edwilliams16, 2016-06-08:
You are probably better off wiring your z motors in series rather than in parallel - almost certainly if you are using a 24V supply. See:
http://rigidtalk.com/wiki/index.php?title=Z-Axis_Stepper_Motor_-_Series_Connected_Splitter_Board

fabtopia, 2016-06-08:
Oke. Changing the DEFAULT_MAX-FEEDRATE for Z to 2.5 does the trick. Still don't understand why it worked with 5 on 1.0.2 and after RC1 so not after RC5 the problem appeared. But that's more for the programmars (@thinkyhead). My problem is solved Thanks you all.

closed 2016-06-08 by fabtopia