MarlinFirmware/Marlin #3971: Keep Power supply on startup

Grogyan, 2016-06-07:
Inside the configuration.h file is this little tidbit

``` // Define this to have the electronics keep the power supply off on startup. If you don't know what this is leave it.
//#define PS_DEFAULT_OFF ```

I'd like to know why, with both Repetier Host and OctoPrint, that on connection to to the board, the ATX power supply turns on?

I cannot find a setting inside OctoPrint or Repetier host that enables this feature.
Reson for this is that I have a d printer at the makerspace that i'd like to keep the power supply off when not in use or the Power on the Raspberry PI gets cycled on

thinkyhead, 2016-06-07:
The host must be sending the `M80` command to Marlin when it connects. You might be able to see this if you give Marlin `M111 S1` before connecting to the host. But connecting the host might actually reset the debug flags since doing a serial connection resets Marlin.

foosel, 2016-06-08:
OctoPrint sends no `M80` on connect (unless you've told it otherwise in "Settings" > "GCODE Scripts" > "After connection printer is established"). Commands sent automatically on connect are `M110`, `M105`and if SD support is enabled also `M21` and `M20`.

In fact, OctoPrint doesn't even know about `M80` besides that it should trigger an internal event when it sends one to the printer:

```
$ grep -r M80 *
octoprint/static/font/fontawesome-webfont.svg:<glyph unicode="&#xf18b;" horiz-adv-x="1920" d="M805 163q-122 -67 -261 -67q-141 0 -261 67q98 61 167 149t94 191q25 -103 94 -191t167 -149zM453 1176v-344q0 -179 -89.5 -326t-234.5 -217q-129 152 -129 351q0 200 129.5 352t323.5 184zM958 991q-128 -152 -128 -351q0 -201 128 -351q-145
 70 -234.5 218t-89.5 328 v341q196 -33 324 -185zM1638 163q-122 -67 -261 -67q-141 0 -261 67q98 61 167 149t94 191q25 -103 94 -191t167 -149zM1286 1176v-344q0 -179 -91 -326t-237 -217v0q133 154 133 351q0 195 -133 351q129 151 328 185zM1920 640q0 -201 -129 -351q-145 70 -234.5 218 t-89.5 328v341q194 -32 323.5 -184t129.5 -352z" />
octoprint/util/comm.py: "M80": Events.POWER_ON,
```

Grogyan, 2016-06-08:
Fortunate OctoPrint seems to be the only one of the Host interfaces that does a facility to enable forcing an M81 on connect to ensure that the power supply is off.
I have yet to try out good ol' Pronterface on the printer I have sporting the above feature

MoonshineSG, 2016-06-08:
maybe this helps ? 

[image]

Blue-Marlin, 2016-06-12:
@Grogyan 
```//// The following define selects which power supply you have. Please choose the one that matches your setup
// 1 = ATX
// 2 = X-Box 360 203Watts (the blue wire connected to PS_ON and the red wire to VCC)
// :{1:'ATX',2:'X-Box 360'}

#define POWER_SUPPLY 1
```
Try do `#define POWER_SUPPLY 2`. The only difference is the polarity of the pin.
So if `#define PS_DEFAULT_OFF` does not work for you, the polarity might be the other - even if you don't have a X-Box power supply.

Grogyan, 2016-06-12:
@Blue-Marlin 
``` #define POWER_SUPPLY 1 ``` is correct, using PNP transistor to turn on the power supply.
The issue is to ensure the power supply is always kept off when printer is not in use, thereby one can remotely turn on the printer and get a print going

thinkyhead, 2016-06-13:
Probably a good idea, if we haven't already done this, to clear `axis_known_position` when the power supply is turned off. Maybe it makes sense to disable heaters too, so they don't come on automatically, and possibly unexpectedly, when the power is restored.

Grogyan, 2016-06-17:
Just a follow up, due to some of the weirdness of the feature with an ATX power supply, i've gone back to not using this feature on the Makerspace 3d printer, and putting faith in OctoPrint to keep the power supply off, though this is isn't an optimal solution.
RAMPS1.4 with Full Graphic Display, and MX31855 thermocouple.

closed 2016-07-18 by jbrazio