MarlinFirmware/Marlin #3972: Z probe Offset

Darozas, 2016-06-07:
Hi all 

Just configured Marlin RC6 on my corexy all runs fine apart from Z probe offset

Got Inductive probe on.

Auto bed levelling on runs fine on G29

But no mater what I do on Z probe offset it just don't work

Got Z safe home enabled & Endstops only for homing.
It shows offset on lcd but cant get nozzle close to bed no matter why number I put in

Can any one help me here ???

Regards

Blue-Marlin, 2016-06-07:
temporarily deactivate min software endstops.

Darozas, 2016-06-07:
Thanks for reply 

Ok will do that. You mean comment them off?

At the minute I got Min software endstops false

Regards

Blue-Marlin, 2016-06-07:
ENDSTOPS_ONLY_FOR_HOMING ? (_adv.h)

Darozas, 2016-06-07:
ENDSTOPS_ONLY_FOR_HOMING  they are on in configuration_adv

Darozas, 2016-06-07:
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

#define AUTO_BED_LEVELING_FEATURE
#define Z_PROBE_OFFSET_FROM_EXTRUDER 1.38

#define FIX_MOUNTED_PROBE

that what I got but Z probe height gets ignored no matter what number I put in

Regards

Blue-Marlin, 2016-06-07:
What do you expect Z_PROBE_OFFSET_FROM_EXTRUDER to do?

Darozas, 2016-06-07:
lower the nozzle???

Regards

brainscan, 2016-06-07:
Why is your probe offset positive?

Blue-Marlin, 2016-06-07:
Ahh. No it does not and is not supposed to do so.
You should see the difference after homing/probing and going to Z0 (G1 Z0).

Darozas, 2016-06-07:
@brainscan  

On my printer 
Negative moves bed down positive moves it up
or I missing something here?

Darozas, 2016-06-07:
@Blue-Marlin 

G1 Z0 give me 1.38mm gap from bed to extruder?

So how do I deal with this gap?
On repetier Z probe offset worked that way was rising bed to extruder ?

I got lost here so what Z_PROBE_OFFSET_FROM_EXTRUDER actually do?

Regards

Blue-Marlin, 2016-06-07:
> or I missing something here?

Oh yes! Z is the difference of height between nozzle and bed - not the height of the bed.

brainscan, 2016-06-07:
Well positive would increase gap between bed and nozzle, negative would reduce it. Most probes have a negative value and as you have an inductive probe I'm sure it should have one too.

brainscan, 2016-06-07:
You z height is exactly your offset so it's set the wrong way around like I said

Darozas, 2016-06-07:
@brainscan

I have tried both negative and positive dimensions non of them worked I had -20 nothing still homes on 1.38 gap and +20 Same 1.38mm gap 

regards

brainscan, 2016-06-07:
Did you update eeprom with the new value? Plus you should really only put -1.38 as -20 could cause a crash

Darozas, 2016-06-07:
i did m851 z value then m500 then m501 to make sure its ok

Regards

brainscan, 2016-06-07:
It's M502 to update eeprom then M500 to save it. What have you set M851 value to? Have you tried RCBugFix as this might have been fixed?

brainscan, 2016-06-07:
Sorry just reread your last comment, I never used M851 but just checked the info on it and sorry your last comment makes total sense.

Darozas, 2016-06-07:
@brainscan

When i connect printer to printrun its show prob offset  (M851 z(Value)) 

I have used LCD to change values then press store them.That part works it shows me Z offset On LCD & Print run but in reality don't do nothing not changing gap.
Or i do understand this wrong (Z_PROBE_OFFSET_FROM_EXTRUDER)

when RCBugFix was released ??
I only downloaded Marlin on Saturday ?

Regards

brainscan, 2016-06-07:
Yeah sorry that was my bad, I had exactly the same problem you're having a couple of years ago but M851 wasn't being used then and had no documentation, so I never used it and got confused. 

RCBugFix is updated all the time but has less bugs than RC6. If you still have a problem it's easier for the developers to be working with the same up to date code. They're working really hard to get another stable version out soon.

Darozas, 2016-06-07:
Thanks for help ill try RCBugFix tonight and let you know results

Regards

brainscan, 2016-06-07:
Sorry I couldn't be more help, hopefully RCBugFix will be better. Someone will work it out and it will probably be something silly, it normally is in my case.

Darozas, 2016-06-07:
Worst case I have to go to 2 switches (Separate probe from endstop) witch is something I don't want to do.

Regards

Darozas, 2016-06-07:
Hi 
ok got RCBugFix installed now G1 Z0 brings the bed up but on auto level drops it off again???

regards

brainscan, 2016-06-07:
Sorry can you explain a little more. Do you mean that after G28 you can send G1 Z0 and the nozzle will move to touch the bed, but if you send G29 and then G1 Z0 it won't?

Does it try to print in mid air if you start a print?

thinkyhead, 2016-06-08:
> When i connect printer to printrun it shows the probe offset (`M851 z(Value)`)

@Darozas  Excellent. It should display `-1.38` if it is set properly. That is, if your nozzle is 1.38mm above the bed when your probe triggers, your `Z_PROBE_OFFSET_FROM_EXTRUDER` should be set to `-1.38` (a negative value).

> Negative moves bed down positive moves it up

That's backwards. As the Z coordinate increases the nozzle should get farther away from the bed. (The Z coordinate represents the distance between the nozzle and the bed.)

> when RCBugFix was released

Never. It is `1.1.0-RC6` plus several patches, which will eventually be released as `1.1.0-RC7`.

> cant get nozzle close to bed…

If your inductive probe your only Z endstop, or do you have another Z endstop?

Darozas, 2016-06-08:
@brainscan 
Installing RCBugFix sorted issue I think
M851 z -1.38 then M500 (Shows on LCD -1.38)
Then i do G28 (Its comes with gap at 1.38) 
Then G29 (Still with the gap at 1.38)
But once I Do G1 Z0 it ill bring bed to touch (So managed print cube yesterday)
So as far I understand that Z probe offset only get in to account when G1 Z Command is that right?

@thinkyhead 

Yes I do use Z probe For homing and Probing 
and as per above

Thank you all for help 
I think im good here now
Regards

brainscan, 2016-06-08:
I'm glad RCBugFix sorted out your problem and you've managed to print something. 

> So as far I understand that Z probe offset only get in to account when G1 Z Command is that right?
The offset is applied during homing or levelling, any command after that like M114 should show actual nozzle position.

Darozas, 2016-06-08:
@brainscan 

I doesn't apply offset on homing for some reason homes with gap.
That why I could not figure it out why offset not working.

Offset only works on  command G1 Z0
Then m114 shows Z0 and nozzle is touch to bed

If I G28 then M114 it shows Z-1.38 but after G1 Z0 its shows Z0 and its nozzle touch to bed.

its confusing me coz on repetier it accounted offset on homing but on Marlin I cant get it to do so.

Regards

brainscan, 2016-06-08:
Oh, sorry. I haven't used repetier in a long time but I think it's the only one to let you specify a home position, so it's normal for the nozzle to stay where it is at the end of a routine. Also never had an lcd, I use pronterface and am just going from the experience I've had with my printer. To be honest I haven't tried homing and sending M114 in a while so things might have changed. I did see someone else had an issue with current position not being updated but not sure if they'd be related. 

Does M114 definitely show -1.38 after G28? It should show 1.38 which would be normal.

Darozas, 2016-06-08:
@brainscan 

Its shows offset after G28 just don't remember positive or negative ill check once home.

But after G1 z0 it shows M114 z0

Regards

Darozas, 2016-06-08:
I have miss red this yesterday this clearly explain what Z prob Offset do
> Ahh. No it does not and is not supposed to do so.
 You should see the difference after homing/probing and going to Z0 (G1 Z0).

closed 2016-07-16 by jbrazio