MarlinFirmware/Marlin #3974: Print is hanging on bed temperature check

pauldesmet, 2016-06-07:
I installed Marlin RC6 on the Rumba board of my BigBox.
1. G-code print contained 
       M104 S210 T0
       M109 S210 T0
     Print did not start but hanged at the bed temperature check.
2. second trial. I set bed temperature manually on LCD at 210.
   I started the same G Code but deleted the line M109 S210 T0
   The print worked perfectly

Conclusion ??  Test done on M109 does not work correctly

manianac, 2016-06-07:
Maybe you have an impossible to reach WATCH_PERIOD?  Can you check the serial console?

RicardoGA, 2016-06-07:
@pauldesmet are youising rc6 or rcbugfix?? Another thing to check is in configuration is the residency time

thinkyhead, 2016-06-08:
@pauldesmet This might be fixed in `RCBugFix`, as we have done some patching around thermal handling. Be sure to test with that when you have a chance.

closed 2016-07-23 by jbrazio