MarlinFirmware/Marlin #3975: Brainstorming: Feedrates

AnHardt, 2016-06-07:
What feedrates do we need to define?
Does talking about or using broken feedrates make any sense?

(Not everywhere a description of what we have - more thought as a target)

* DEFAULT_MAX_FEEDRATE: Fastest usable feddrate without loosing steps. Fer the extruders this means while extruding.
* HOMING_FEEDRATE: Fastest usable feedrate not smashing the endstops.
* HOMING_BUMP_DIVISOR:  feedrate for final approach to the endstops. (being slow helps to be exact). Should be HOMING_BUMP_FEEDRATE.
* MANUAL_FEEDRATE; Fedrate used by the manual moves initiated by the menu. Could be the same as  DEFAULT_MAX_FEEDRATE: but sometimes slower is more convenient.
* XY_TRAVEL_SPEED: For traveling without Z and E. Could be the smaller of the two in DEFAULT_MAX_FEEDRATE.    
*    
* Z_TRAVEL_SPEED?:  For traveling without X/Y and E. Could be DEFAULT_MAX_FEEDRATE.
* E_TRAVEL_SPEED?:  For traveling without X/Y and Z. For transporting filament forward or backward, but not squeezing it through the nozzle.

The last 3 points could be unified to an array (TRAVEL_FEEDRATE)

Do we need more?
Do we have more?

fbarcenas, 2016-06-07:
Maybe my opinion is little drastic..
I only think than we need 2(3):
DEFAULT_MAX_FEEDRATE:
HOMING_FEEDRATE: 
      HOMING_BUMP_FEEDRATE.

and if use LCD:
MANUAL_FEEDRATE XYZ E

The rest i think than is stupid, because in the software slicer you config this...

Roxy-3DPrintBoard, 2016-06-07:
> Do we need more?

My vote would be we carefully consolidate and have less.   Without looking at the code, it isn't clear when one over rides another.

jbrazio, 2016-06-07:
I would suggest a bit of consolidation.

- `MAX_FEEDRATE`: mm/s clips anything above this value
- `TRAVEL_FEEDRATE`: mm/s
- `HOMING_FEEDRATE`: mm/s
- `HOMING_BUMP_DIVISOR`: units i.e. 2 = `HOMING_FEEDRATE /2`

`XY_TRAVEL_SPEED`, `Z_TRAVEL_SPEED`, `E_TRAVEL_SPEED`: Discarded as we have all the values on `TRAVEL_FEEDRATE`. We could even go further and make `HOMING_FEEDRATE` and `TRAVEL_FEEDRATE` a proportional of the `MAX_FEEDRATE`.. but that maybe an overkill.

I  am a bit hesitant to consolidate `MAX_FEEDRATE` and `TRAVEL_FEEDRATE_DIVISOR` because people may want the robot not to move all the time at max speed.

If `MANUAL_FEEDRATE_DIVISOR` is used when moving from the LCD, then we could kill it as the user can move 0.1mm per encoder tick.

We could make proportional units as a percentage (0-100) rather than a direct divisor.

Roxy-3DPrintBoard, 2016-06-07:
> XY_TRAVEL_SPEED, Z_TRAVEL_SPEED, E_TRAVEL_SPEED: Discarded as we have all the values on TRAVEL_FEEDRATE.

Agreed.

> I am a bit hesitant to consolidate MAX_FEEDRATE and TRAVEL_FEEDRATE_DIVISOR because people may want the robot not to move all the time at max speed.

I think you are saying you are hesitant to consolidate MAX_FEEDRATE and TRAVEL_FEEDRATE because people may not want the robot to move at full speed all the time.   I agree with this also.   But the reason for my confusion is TRAVEL_FEEDRATE_DIVISOR does not exist in the code.

jbrazio, 2016-06-07:
Never mind, take out the divisor part ! lol

fbarcenas, 2016-06-07:
Why not join MAX_FEEDRATE and TRAVEL_FEEDRATE?
Can anybody give me a explication.
Is supposed than in the travel movement we want the high speed and like i say before, THIS CAN be changed in the slicer software

Roxy-3DPrintBoard, 2016-06-07:
I think what you are trying to suggest is:  

Question:  Why define a MAX_FEEDRATE if we are going to limit its use (and full potential) but respecting the values in TRAVEL_FEEDRATE.   

I think this is a valid question.   But I do like having the Maximum feedrates documented as being the Maximum.

Blue-Marlin, 2016-06-07:
@fbarcenas 
There is no way you can adjust the speed of travels to get from one probe point to the other - in the slicer.

fbarcenas, 2016-06-07:
@Blue-Marlin 
"from one probe point to the other"

You want say when do G29 auto bed leveling?

Blue-Marlin, 2016-06-07:
Or the way to the save homing point, or the way to a point for filament change and back, ... .

jbrazio, 2016-06-15:
@AnHardt what is the outcome of this discussion ?

closed 2016-08-03 by AnHardt