MarlinFirmware/Marlin #3977: RC5 extruder coordinate overflow on LCD

orcinus, 2016-06-07:
If the extruder coordinates exceed 999, the coordinate output on the LCD seems to overflow into non-numeric characters:

https://www.youtube.com/watch?v=SwJ95ZivCC4

The actual coordinates were:

https://dl.dropboxusercontent.com/u/1702513/Photo%2007-06-16%2021%2000%2054.jpg

Suggested fix - for E axis, display relative position (since entering the Move Axis submenu) instead of absolute position, which is irrelevant and not very useful anyways.

Blue-Marlin, 2016-06-07:
Try
```diff
          #endif //EXTRUDERS > 3
        #endif //EXTRUDERS > 2
      }
    #endif //EXTRUDERS > 1
-    lcd_implementation_drawedit(pos_label, ftostr31(current_position[E_AXIS]));
+    lcd_implementation_drawedit(pos_label, ftostr51(current_position[E_AXIS]));
  }
  if (LCD_CLICKED) lcd_goto_previous_menu(true);
  #if EXTRUDERS > 1
    active_extruder = original_active_extruder;
  #endif
}
```
in `ultralcd.cpp`

Relative to what?

orcinus, 2016-06-07:
Relative to 0 set when you've entered the submenu.

orcinus, 2016-06-07:
I really don't have a use for the absolute number of mm extruded since the printer was turned on. I doubt anyone else does, especially when manually jogging the extruder.

jbrazio, 2016-07-16:
For reference: 45ea8749

closed 2016-07-16 by jbrazio

orcinus, 2016-07-16:
Why close it instead of pushing the milestone further?
This is a bizarre way of managing issues.

orcinus, 2016-07-16:
Additional comment re: question on "relative to what" and my previous answer of "relative to 0 set when you've entered the submenu"...

That is the exact same way Babystepping menu item is currently behaving. It displays relative state since selecting that menu item. So the behaviour i'm proposing as a fix already exists in the UI (so shouldn't be too confusing UX-wise).

jbrazio, 2016-07-16:
Wasn't this fixed by the referenced commit ?

thinkyhead, 2016-07-17:
The line has long been changed. Now:
```cpp
lcd_implementation_drawedit(pos_label, ftostr41sign(current_position[E_AXIS]));
```

thinkyhead, 2016-07-17:
@orcinus Thing is, the E axis is an actual axis. It has relevance because it corresponds to GCode you could send from the console alongside it.

orcinus, 2016-07-17:
The Z axis is an actual axis too, yet - relative babystepping : )
But i get your point - in the current move menu, it would not be consistent with other axes.