MarlinFirmware/Marlin #3980: SINGLENOZZLE: EXTRUDERS versus HOTENDS

MoonshineSG, 2016-06-08:
The pull request #3895  from thinkyhead/rc_singlenozzle_part_2 to modify extruders vs hotends to support single nozzle added a ton of changes. Previously I was able to create a custom M890 to allow me to change between Single to Dual nozzle as and when I needed. With the new changes, that is not possible (easily) 
Am I the only one that would swap between single and dual nozzle often enough to be annoyed by the need to flash a new firmware every time ? (10000 is a large number, but eventually the flash will fail...)

Can the switch between single/multiple be done using a M code ?

thinkyhead, 2016-06-08:
@MoonshineSG What about the recent changes is getting in your way? As long as `SINGLENOZZLE` is disabled, `HOTENDS` is the same as `EXTRUDERS`.

If you're able to rebase your custom Marlin onto `RCBugFix` at a point just prior to the recent `SINGLENOZZLE` changes, please make a branch out of that. Starting there I'm sure we can figure out a way to adapt your code and `M890` to the recent changes.

MoonshineSG, 2016-06-08:
With the changes I made after you posted the initial SINGLENOZZLE codes, I leave the SINGLENOZZLE enabled and using M890 N(1|2) i change the nozzle X offsets 9,9 or 18 (in my case) and add a single bool variable `singlenozzle` which then is used in `Temperature::get_pid_output`;

```
      bool nozzle_test = (e == active_extruder);
      uint8_t nozzle_extruder = e;
      int cterm_index = e;
[...]
        if (single_nozzle) {
            nozzle_test = true;
            nozzle_extruder = active_extruder;
            cterm_index = 0;
        }
```

With the new changes I see alot of checks ` #if ENABLED(SINGLENOZZLE)` and I am sure that not all can be simply replaced with a bool variable check...

MoonshineSG, 2016-06-08:
@thinkyhead does this helps ?
 https://github.com/MoonshineSG/Marlin/commits/single_nozzle_M890

thinkyhead, 2016-06-08:
> the initial SINGLENOZZLE codes

Well of course that code was totally incomplete. So, the main difference that `HOTENDS` introduces is that temperature checks only apply to the one nozzle. But `EXTRUDERS` still applies to the E steppers. The conditions based on `ENABLED(SINGLENOZZLE)` are not that many. Almost all changes pertain to the use of `HOTENDS` where `EXTRUDERS` used to apply.

> does this helps ?

It should! From that commit I can perhaps figure out how to adapt `M890` to this new environment. I'll mess with it shortly…

thinkyhead, 2016-06-08:
Ok, so… How are your steppers and extruders related? Or, what are the two setups that you're switching between, exactly?

With `SINGLENOZZLE` the main thing is that there is only a single nozzle, but multiple steppers are connected to that one nozzle. When you switch "extruders" you are still using the same heaters and thermistors, but a different (single) E stepper motor.

With multiple extruders (`EXTRUDERS >= 2`), there's more than one nozzle on the same carriage, each with its own (single) E stepper. Of course, this is entirely incompatible with the `SINGLENOZZLE` feature as we have implemented it (based on MarlinKimbra), because it needs to manage more than one heater/thermistor.

So what you need is a new feature, essentially, that allows you to switch your particular setup. But it can't rely on `SINGLENOZZLE`.

MoonshineSG, 2016-06-08:
There's nothing really special about my setup. Imagine a dual extruder prusa with e3d chimera hotend. 
This setup is used when printing 2 materials (PLA/PVA mostly). This of course has 2 extruders, 2 heaters, 2 thermistors and 2 nozzles.

When I want to print dual colours I take out the heat breakers & 2 heater blocks and replace them with the 2 in 1 out E3D cyclops heater block. That has 2 extruders, 1 heater, 1 thermistor and one nozzle

Currently when I use cyclops I add an additional thermistor to the empty slot so Marlin doesn't throw a thermal error. 

unless you have 2 separate printers, I am very sure this is common setup 

M890 allow to change the offset between `--v--v--` and  `----v----` nozzles.

thinkyhead, 2016-06-09:
So the two configurations you need to be able to swap:
- Chimera: 2 stepper, 2 heater/thermistor, 2 nozzle
- Cyclops: 2 stepper, 1 heater/thermistor, 1 nozzle (i.e., "single nozzle")

Are you using the same set of 2 steppers, or 4 steppers total?

I'm not sure that we can (or want to) create a general feature to support changing setups. But if you were able to do it with the incomplete `SINGLENOZZLE` feature, then I'm certain you could make this as a general feature. Basically, by imitating that first set of `SINGLENOZZLE` changes, but giving your custom setting a different name.

MoonshineSG, 2016-06-09:
Using the same set of 2 steppers (left-right). Basically I just change the heater block (with thermistors).

Ah... so I take latest marlin with the new ("complicated") `SINGLE NOZZLE` support (so I can stay up to date with code changes) and set it to 2 extruders, 2 hotends, plus my own existing changes but renamed to CHIMERA_CYCLOPS_AWSOME_SETUP ?

that might work... 

what features from  the complete SINGLENOZZLE would I be missing ?

thinkyhead, 2016-06-09:
I'll see if I can illustrate what I mean, using your branch, reworking it so it doesn't collide with `SINGLENOZZLE` anyplace.

>  ("complicated")

Well the idea of `SINGLENOZZLE` (which has been in MarlinKimbra and some other forks for well over a year) was to be able to simply set the single option, then your multi-extruder would work with no other rigamoarole required.

thinkyhead, 2016-06-09:
> what features from the complete SINGLENOZZLE would I be missing?

It depends on how much of `SINGLENOZZLE` you decide to borrow. But basically, if you look at all the places in the current code where `EXTRUDERS` is still being used (rather than `HOTENDS`) you can see where the number of extruder steppers still applies. And every place where `HOTENDS` is used, you will see where only the number of hotends matters.

For your customization, you don't want the "feature" of `SINGLENOZZLE` where it always assumes you only have a single heater. You want to be able to turn that on and off with a flag.

Basically, if you do a global search on `SINGLENOZZLE`, `EXTRUDERS`, and `HOTENDS` you will find all the places where behavior _may_ need to be tuned. And you can decide in each instance whether that is needed.

MoonshineSG, 2016-06-09:
I did that... thats why I called it "complicated" :D But I guess I can give it another try. 

For now I will go with the renamed simple version (probably find a better name than `CHIMERA_CYCLOPS_AWSOME_SETUP` ) on top of the latest RCBugFix

I'll shout for help if need be...

thinkyhead, 2016-06-09:
Sounds good. I will be working in parallel on this also, because it intrigues me to see what is required.

thinkyhead, 2016-06-09:
Integrating your changes into RCBugFix… https://github.com/thinkyhead/Marlin/tree/rc_singlenozzle_M890
…without any extra work to check on other places where `single_nozzle` might come into play.

MoonshineSG, 2016-06-09:
:D You're fast. I was only half way there... 

I'll port the changes to my repo and run some tests.

thinkyhead, 2016-06-09:
I've just cleaned it up a bit. Probably will leave it where it's at for now…

MoonshineSG, 2016-06-09:
No worries, I can take over from here. The idea of having a separate setting for this case was what saved me. I don't need to care about extruder/hotend code changes.... 
Thanks again.

MoonshineSG, 2016-06-09:
I cleaned up my code, added some extra checks and seams like everything is ok. I managed to do an update to the latest RCBugFix version without conflicts, so that's good. 

https://github.com/MoonshineSG/Marlin/commit/8843e978ed56bb4c4e067e6b905fe7e707b90c78

(I managed to mess my github and now by default it compares with RC instead of RCBugFix... No idea how to revert that... grrrr..... )

thinkyhead, 2016-06-09:
> mess my github and now by default it compares with RC instead of RCBugFix

That's how it has been for a long while. In fact, one of the main reasons I made the `mfpr` shell script.

MoonshineSG, 2016-06-09:
I am pretty sure previously it was reporting "This branch is X commits ahead, Y commits behind MarlinFirmware:RC**BugFix**".

thinkyhead, 2016-06-09:
> This branch is X commits ahead, Y commits behind MarlinFirmware:RCBugFix

When? Where? How? Why? The default branch is `RC` and when you do a compare (using the button on your Github page), it's supposed to compare to the default branch, and for me it has worked this way for a long, long time.

MoonshineSG, 2016-06-09:
[image]

here... But I am an idiot!!! I renamed my own branch `prusa_i3` so now it compares to the default branch in MarlinFirmware... Previously _my_ branch was called `RCBugFix` as well. Github is smart. I am not. 

Sorry for the confusion. :D

jbrazio, 2016-07-11:
@MoonshineSG Can we close this issue ?

MoonshineSG, 2016-07-11:
yes. it's all good. Thanks.

MoonshineSG, 2016-08-04:
@thinkyhead I was away for a while and now I looked at the changes and noticed "SWITCHING_EXTRUDER"  and "MIXING_EXTRUDER" on top of "SINGLENOZZLE"... 

Would this conflict or replace the functionality my implementation of "SWAPPABLE_EXTRUDER" ?

thinkyhead, 2016-08-06:
The new options try to compartmentalize their behavior, so they shouldn't get too much in the way. The main changes they apply are in the `stepper_indirection.h` file, since they each affect how steppers behave.

- `SWITCHING_EXTRUDER` drives a single stepper motor that reverses direction for T1.
- `MIXING_EXTRUDER` moves 2 or more steppers proportionally, with enable/direction applying to all.

This block in `Conditionals_LCD.h` is the part that you need to pay attention to:
```cpp
/**
 * Extruders have some combination of stepper motors and hotends
 * so we separate these concepts into the defines:
 *
 *  EXTRUDERS    - Number of Selectable Tools
 *  HOTENDS      - Number of hotends, whether connected or separate
 *  E_STEPPERS   - Number of actual E stepper motors
 *  TOOL_E_INDEX - Index to use when getting/setting the tool state
 *  
 */
#if ENABLED(SINGLENOZZLE)             // One hotend, multi-extruder
  #define HOTENDS      1
  #define E_STEPPERS   EXTRUDERS
  #define E_MANUAL     EXTRUDERS
  #define TOOL_E_INDEX current_block->active_extruder
  #undef TEMP_SENSOR_1_AS_REDUNDANT
  #undef HOTEND_OFFSET_X
  #undef HOTEND_OFFSET_Y
#elif ENABLED(SWITCHING_EXTRUDER)     // One E stepper, unified E axis, two hotends
  #define HOTENDS      EXTRUDERS
  #define E_STEPPERS   1
  #define E_MANUAL     1
  #define TOOL_E_INDEX 0
  #ifndef HOTEND_OFFSET_Z
    #define HOTEND_OFFSET_Z { 0 }
  #endif
#elif ENABLED(MIXING_EXTRUDER)        // Multi-stepper, unified E axis, one hotend
  #define HOTENDS      1
  #define E_STEPPERS   MIXING_STEPPERS
  #define E_MANUAL     1
  #define TOOL_E_INDEX 0
#else                                 // One stepper, E axis, and hotend per tool
  #define HOTENDS      EXTRUDERS
  #define E_STEPPERS   EXTRUDERS
  #define E_MANUAL     EXTRUDERS
  #define TOOL_E_INDEX current_block->active_extruder
#endif
```
For your feature, just add `#elif SWAPPABLE_EXTRUDER` with the values that apply to your feature. Then peek at the places where the new `E_STEPPERS` and `E_MANUAL` values are utilized. You probably don't need to make very many changes, compared to previously.

MoonshineSG, 2016-10-15:
I did have an attempt to introduce my own `SWAPPABLE`, but al the above cases the `HOTENDS` is defined at compile time, while mine would change at run time. So I'm going with the easy way of having 2 compiled firmwares and burn them as and when I change the hotend... Not ideal, but workable for now...

thinkyhead, 2016-10-22:
> the above cases the HOTENDS is defined at compile time

True, but this is actually not too hard to work around. You can add your own (yet another) macro, define it as `#define ACTIVE_HOTENDS hotends_count` for `SWAPPABLE_EXTRUDER` but as `HOTENDS` for others, and then replace `HOTENDS` throughout the code, in just the places where `hotends_count` should apply.

boelle, 2017-04-26:
@thinkyhead so we can close this one again??

closed 2017-11-01 by thinkyhead