MarlinFirmware/Marlin #3981: Printer randomly Hangs while printing from SD

RicardoGA, 2016-06-08:
Hi everybody, with the last rc bugfix the printer hangs randomly sometimes prints ok  and another times hangs with differents files and at differents heights, and the printer freezes and dont show any error message in the lcd or via serial 

things i´ve tried:

1.- Test with another Sd card
2.- Compile with Arduino 1.6.9 and 1.6.8
3.- Print from sd  with the printer connected with repetier to check if some error appear and not the only message is  echo:busy: processing 
4.- Check connections

Printer Details 
Prusa i3
Ramps 1.4
Firmware Marlin RC6 Bug fix
Auto bed leveling with Servo 
Filament Widht Sensor 

Thanks in advance

thinkyhead, 2016-06-08:
Did you test with `RC6` or did you test with the latest code in `RCBugFix`?

thinkyhead, 2016-06-08:
> the printer freezes and don't show any error message

That might be hard to track down. It implies that there is a buffer overflow perhaps. Are you able to get the printer to freeze outside of printing? Do you have any GCode that _**always**_ causes the machine to freeze? Since Marlin is a more-or-less deterministic system, you _should_ be able to find some GCode that always causes a freeze, assuming the problem is software-related.

RicardoGA, 2016-06-08:
@thinkyhead Yes i have a gcode that i've tried to print 3 times and the 3 times the printer freeze, currently i'm trying to print this file with other version of marlin (marlin kimbra) to discard a gcode error.

in a couple of hour i'll report if the print was successfull and upload the file.

RicardoGA, 2016-06-09:
@thinkyhead The print just finished without any problem with marlin krimbra, the gcode it´s ok. here's the link with the gcode that hangs with the rc bugfix and also the rcbugfix with the configuration i use.
  
https://www.dropbox.com/sh/3dlhqk8k1vjr25r/AABjL79SQPHsQr1ky8iclBaka?dl=0

thinkyhead, 2016-06-09:
That's quite a big model, and probably not easy to use for determining the issue here. How far into the print does the freeze occur?

What I'm hoping for is a very short piece of GCode that always causes a freeze.

RicardoGA, 2016-06-09:
@thinkyhead  that's right it's a big model i'll try to find a small model that repeat the problem like this and the freeze happen between 29 mm and 33 the 3 times at different height but in this range

RicardoGA, 2016-06-15:
@thinkyhead just to update after the test with marlin kimbra, I  re downloaded the rcbugfix to ensure i have the latest bugfix and i've been printing without any freeze even i printed the file that previously cause the freeze

thinkyhead, 2016-06-16:
@RicardoGA Well, if the problem pops up again, please post an issue again about it. Mysterious crashing bugs are the worst!

closed 2016-06-16 by thinkyhead