MarlinFirmware/Marlin #3982: Second temp sensor for ambient temp

Kev0123, 2016-06-08:
Hi

I'm using a GT2560 and wanted to use the left over temp-in to show the ambient temp.
So I would be able to log it and you could may implement a minimum print temp?!
Is it possible?

Regards 
Kev0123

thinkyhead, 2016-06-08:
Currently all of Marlin's temperature sensing code is tied to the heated bed and hotends. So you would need to implement this as an add-on feature (perhaps `AMBIENT_TEMP_SENSOR`), making sure not to step on any settings that "imply" more hotends.

maukcc, 2016-06-10:
That would be a nice feature.
We are using the 3rd extruder to display the ambient temp at the moment, which is not ideal as you get a lot of 3rd extruder additions.

thinkyhead, 2016-06-11:
If I were to implement this (which I may) then it would automatically utilize the first unused E thermistor plug. So if you had E0 and E1 connected, you would only need to set `EXTRUDERS 2` and connect the ambient sensor to the E2 thermistor plug.

Kev0123, 2016-06-11:
Wow this would be great! Thanks for giving a thought!

maukcc, 2016-06-11:
That will be better then it is now. 
What will happen when 4 extruders are hooked up? will the ambient temp just be lost some how?(I have no problem with that for now)
Can we also set (in the pins file): 
 
> #define TEMP_CHAMBER_PIN           13

thinkyhead, 2016-06-13:
> What will happen when 4 extruders are hooked up?

If all the E thermistor pins are full then an override would be needed.

> Can we also set (in the pins file): `TEMP_CHAMBER_PIN`

If that were to be included in all the pins files then it would be the _only option_, and would not allow automated use of the next unused E thermistor plug. So I suggest instead to make it a disabled-by-default sub-option of the feature, then it will only be used if it set as an override. If there is no override set, then we fall back to the next available `TEMP_n_PIN`. If the fallback can't be done, then it will print a compile error "Please set an explicit TEMP_AMBIENT_PIN."
```cpp
//#define AMBIENT_TEMP_SENSOR
#if ENABLED(AMBIENT_TEMP_SENSOR)
  //#define TEMP_AMBIENT_PIN TEMP_1_PIN
#endif
```

boelle, 2017-04-26:
maybe add this to milestone 1.1.1, but would ambient not be the same as chamber temp?

if i remember correct i suggested that before this one.

maukcc, 2017-05-03:
I think we need to change this to CHAMBER_TEMP_SENSOR and TEMP_CHAMBER_PIN
as ambient would suggest the temp outside of the printers volume.

fiveangle, 2017-06-26:
Doesn't @MagoKimbra 's fork have this already ?

(https://github.com/MagoKimbra/MarlinKimbra):

> * Added Cooler and Hot Chamber

fiveangle, 2017-06-26:
> would ambient not be the same as chamber temp?

To me, "Ambient" implies temperature of the universe outside of the printer, while "Chamber" implies temperature inside heated/cooled chamber.

thinkyhead, 2018-02-06:
Dupe of #973.

closed 2018-02-06 by thinkyhead