MarlinFirmware/Marlin #3984: Is endstops.enable() a valid call?

lrpirlet, 2016-06-08:
Sorry if my question is stupid... I do not know too well C++

In Marlin_main.cpp I found the folowing code 
```C++
static void setup_for_endstop_move() {
  saved_feedrate = feedrate;
  saved_feedrate_multiplier = feedrate_multiplier;
  feedrate_multiplier = 100;
  refresh_cmd_timeout();
  #if ENABLED(DEBUG_LEVELING_FEATURE)
    if (DEBUGGING(LEVELING)) SERIAL_ECHOLNPGM("setup_for_endstop_move > endstops.enable()");
  #endif
  endstops.enable();
```
In the endstops.cpp, I could not find **enable(**... I only found 
```C++
Endstops::Endstops() {
  enable_globally(
```
Is ``` endstops.enable(); ``` a valid call??? the compiler does NOT choke.

lrpirlet, 2016-06-08:
OK, I found in endstops.h the following 
```C++
static void enable(bool onoff=true) { enabled = onoff; } 
```
I that enough to define a function to enable or disable the use of the endstops? Is that function nothing more than setting the boolean named _onOff_ ??

thinkyhead, 2016-06-08:
> Is that function nothing more than setting the boolean…

When the boolean `Endstops::enabled` is true, Marlin does not respond to endstops. It still updates them from the hardware so it knows the current states.

thinkyhead, 2016-06-09:
It should be noted that in some cases it's more appropriate to call `endstops.not_homing()`. This will make sure endstops are enabled if they should be left enabled, as you get when `ENDSTOPS_ONLY_FOR_HOMING` is disabled, or as set by `M120` / `M121`.

lrpirlet, 2016-06-14:
@thinkyhead. Thanks for the explanation, it is clear now... and sorry for the late answer.

closed 2016-06-14 by lrpirlet