MarlinFirmware/Marlin #3987: Print offsets in two 2 hot end configuration

lavato, 2016-06-09:
I just came across this issue  when I started printing larger object yesterday.

**Printer Configuration**
X carriage has following attached (from left to right):
HotEnd2, HotEnd1 and a Sensor (each taking 30mm space)

**Relevant Configuration.h details**
#define HOTEND_OFFSET_X {0.0, -30.20}
#define HOTEND_OFFSET_Y {0.0, -1.20}
#define X_PROBE_OFFSET_FROM_EXTRUDER 30
#define Y_PROBE_OFFSET_FROM_EXTRUDER 0
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.90
#define Z_SAFE_HOMING
#define Z_SAFE_HOMING_X_POINT 45
#define Z_SAFE_HOMING_Y_POINT 5
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 190
#define Y_MAX_POS 190
#define Z_MAX_POS 180

**How to reproduce**
1) Generate gcode so HotEnd 1 is used (the one in the middle, which can cover the biggest area)
2) Start print
3) The print offsets by 15mm to the right 
So instead of print starting e.g. at X20, Y20, it starts at X35 & Y20

**Some more points**
When i issue G28, Hot end one goes to X0,Y0 (as it should) and the LCD displays X0,Y0
Because Hotand 1 is in the middle of the carrage, it should cover the most space on the hotbed,However because of the issue it offsets to X+15mm and therefore if I want to print an object covering full plate, I cannot as the X Max is of the bed.

I guess this is happening because Hot and 2 is <X0 so it sets X MIN to a wron value.
I also noticed that I cannot compile Marlin if I set  LEFT_PROBE_BED_POSITION  to 20 (currently 30), possibly because of the same issue.

**Versions**
BugFix from 08/Jun
RAMPs1.4
PrusaI3

thinkyhead, 2016-06-09:
What is your output from `M503`?

lavato, 2016-06-10:
I am not sure what  happening now, some funny behavior with my EEPROM but I cannot reproduce the X-15,Y-5,Z1.9 after g28 and the print starts at x0,y0

This "correction" happened after I applied offsets and stored it to EEPROM. I recompiled the Firmware uploaded it it to Arduino but still X0 & Y0, **which is correct**

Here is the output.
>>>M503
SENDING:M503
echo:Steps per unit:
echo:  M92 X71.14 Y71.37 Z400.30 E94.29
echo:Maximum feedrates (mm/s):
echo:  M203 X500.00 Y500.00 Z3.00 E35.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B0 F0
echo:  M145 S1 H240 B80 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00 C100.00 L20
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 T1 D3.00
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-1.90

Thank you anyway

closed 2016-06-10 by lavato

thinkyhead, 2016-06-11:
Good news @lavato. I was concerned that your `M206` might not be zeroed.