MarlinFirmware/Marlin #3988: Z Probe deploy delay added

modwadd, 2016-06-09:
I have a Z Probe what is tiggered if it is not Deployed.
To make that work I add a delay at the end of deploy_z_probe().
That works fine for me for Homing and Auto bed leveling.
Now I try to use bed mesh leveling, there this change is not working.
I could not find where the probe will deployed, any advice?
If use  RCBugFix Branch, should mesch leveling work there?

BR

thinkyhead, 2016-06-09:
The Mesh Bed Leveling in Marlin can only be done manually right now, so it doesn't use a probe. You must use a piece of paper to feel under the nozzle at each point, manually adjusting Z with the LCD controller.

thinkyhead, 2016-06-09:
> a Z Probe that is triggered if it is not Deployed

What kind of Z probe is that, by the way?

closed 2016-06-09 by thinkyhead

modwadd, 2016-06-10:
[image]

The Steel needle goes on the bed.
The Servo lift it thrue the optical Endswich if retracked -> switch triggered. 
If it is deployed the servo provides pressure on the Spring to hold the needle down during probing.
Givs me a Standard Deviation of 0.0034 mm ! with 10 Probes including E.(retrac /deploy for each probe).
Ususal due to the glitches in the servo control pwm and the servo itself the enstop is not that acurate.

I guess its even more acurate than inductive proxemety sensors because I messure the glas and the blue tape also (If I use it)

It would be perfect if the autolevel compensate the "bowl" of the glas what happen if the bed (30*30cm) is heated up.

thinkyhead, 2016-06-11:
Very nice probe solution. Is this design published on Thingiverse or elsewhere?

If there is a bow in the glass on heating, then, yes, Mesh leveling is best, and you would want to probe when the bed is heated up.

modwadd, 2016-06-11:
Hello, not yet Published.,,, But youre right it shoud be. 
I will publish it soon and post a link here.

modwadd, 2016-06-11:
It will be http://www.thingiverse.com/thing:1620579
as soon as I could enable published :-)