MarlinFirmware/Marlin #3989: Creating Menu Options w/ variable input.

timmit99, 2016-06-09:
I'm truing a menu option for controlling 3 transistors with three of the PWM servo pins on my Ramps 1.4. I have created a submenu and option for controlling an LED with the M42 command ut this allows for only discrete values. 

`static void lcd_led_menu()
{
    START_MENU();
    MENU_ITEM(back, MSG_PREPARE, lcd_prepare_menu);
    MENU_ITEM(gcode, MSG_ENABLE_LED_4, PSTR("M42 P4 S255"));
    MENU_ITEM(gcode, MSG_DISABLE_LED_4, PSTR("M42 P4 S00"));
    END_MENU();`

How would i go about creating an option to use the encoder to select a number between 0 and 255 for a certain variable and then apply that value to the M42 gcode? I am seeing this line: 
`MENU_ITEM_EDIT(int3, MSG_SPEED, &feedmultiply, 10, 999);`
showing up whever the rotory encoder is used and i have made it look like this:
`MENU_ITEM_EDIT(int3, MSG_LED_4, &PWMled4, 0 , 255);` 
but i do not know where to go from here. I'm trying to trace the values used in &feedmultiply but i'm getting lost.

timmit99, 2016-06-09:
I have a workaround where i made a second menu item that was a function that just wrote the value to the desired pin  with `analogWrite
```

int PWMledstrip[3] ={0};

void set_led_strip()
{
  analogWrite(4, PWMledstrip[0]);
  analogWrite(5, PWMledstrip[1]);
  analogWrite(6, PWMledstrip[2]);
}

void toggle_led_strip()
{
  int PWMvalue =0;
  if (PWMledstrip[0] == 0 && PWMledstrip[1] == 0 && PWMledstrip[2] == 0)
  PWMvalue = 255;
  for (int i=0; i<3;i++){
    PWMledstrip[i] = PWMvalue;
    }
    set_led_strip();
}`

static void lcd_led_menu()
{
    START_MENU();
    MENU_ITEM(back, MSG_PREPARE, lcd_prepare_menu);
    MENU_ITEM(function, MSG_TOGGLE_LED_STRIP, toggle_led_strip);
    MENU_ITEM_EDIT(int3, MSG_LED_STRIP_RED, &PWMledstrip[0], 0, 255);
    MENU_ITEM_EDIT(int3, MSG_LED_STRIP_GREEN, &PWMledstrip[1], 0, 255);
    MENU_ITEM_EDIT(int3, MSG_LED_STRIP_BLUE, &PWMledstrip[2], 0, 255);
    MENU_ITEM(function, MSG_SET_LED_STRIP, set_led_strip);
    MENU_ITEM(submenu, MSG_LED_STRIP_PRESET  , lcd_move_menu)
    
    END_MENU();
  }

```
However i can't seem to pass values into the functions with the menu item control. the line below was my attempt to create a function that would take an input from within the MENU_ITEM() block and send that to the function. 

If i change set_led_strip to:

```
void set_led_strip(int Red,int Green,int Blue)
{
  analogWrite(4, Red);
  analogWrite(5, Green);
  analogWrite(6, Blue);
}
```

and try to use:

`MENU_ITEM(function, MSG_SET_LED_STRIP, set_led_strip(PWMledstrip[0],PWMledstrip[1],PWMledstrip[2]));`

I get _Invalid use of void expression_ with the following error message

```
Arduino: 1.6.5 (Windows 8.1), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

ultralcd.cpp: In function 'void lcd_led_menu()':
ultralcd.cpp:141: error: invalid use of void expression
             menu_action_ ## type ( args ); \
                                         ^
ultralcd.cpp:693:5: note: in expansion of macro 'MENU_ITEM'
     MENU_ITEM(function, MSG_SET_LED_STRIP, set_led_strip(PWMledstrip[0],PWMledstrip[1],PWMledstrip[2]));
     ^
ultralcd.cpp:151: error: expected ';' before 'if'
     if (encoderPosition / ENCODER_STEPS_PER_MENU_ITEM >= _menuItemNr) encoderPosition = _menuItemNr * ENCODER_STEPS_PER_MENU_ITEM - 1; \
     ^
ultralcd.cpp:697:5: note: in expansion of macro 'END_MENU'
     END_MENU();
     ^
invalid use of void expression
```

thinkyhead, 2016-06-09:
You want the `MENU_ITEM_EDIT_CALLBACK` macro. Take a look at the menu items that use it, and it should be obvious how it works.

jbrazio, 2016-07-11:
Closing due to inactivity.

closed 2016-07-11 by jbrazio