MarlinFirmware/Marlin #3990: PID autotune FAILED

focamonca, 2016-06-09:
When I Try to Give the PID AUTOTUNE for the bed this is the result: 

`SENT: M303 E-1 S60 C10
RECEIVED: PID Autotune failed! Bad extruder number
RECEIVED: ok
`

Roxy-3DPrintBoard, 2016-06-09:
Extruder -1 ?

focamonca, 2016-06-09:
>  #define  DEFAULT_bedKp 234
  #define  DEFAULT_bedKi 43.28
  #define  DEFAULT_bedKd 316.28

  //120V 250W silicone heater into 4mm borosilicate (MendelMax 1.5+)
  //from pidautotune
  //#define  DEFAULT_bedKp 97.1
  //#define  DEFAULT_bedKi 1.41
  //#define  DEFAULT_bedKd 1675.16

  // FIND YOUR OWN: "M303 E-1 C8 S90" to run autotune on the bed at 90 degreesC for 8 cycles.
#endif // PIDTEMPBED

This is as is written on Config.h

Blue-Marlin, 2016-06-09:
```
    if (false
      #if ENABLED(PIDTEMP)
         || hotend >= HOTENDS
      #else
         || hotend >= 0
      #endif
      #if DISABLED(PIDTEMPBED)
         || hotend < 0
      #endif
    ) {
      SERIAL_ECHOLN(MSG_PID_BAD_EXTRUDER_NUM);
      return;
    }
```
Is `PIDTEMPBED` active?

Blue-Marlin, 2016-06-09:
Ups.
```
    if (false
      #if ENABLED(PIDTEMP)
         || hotend >= HOTENDS
      #else
         || hotend >= 0
      #endif
      #if DISABLED(PIDTEMPBED)
         || hotend < 0
      #endif
+     || hotend < -1
    ) {
      SERIAL_ECHOLN(MSG_PID_BAD_EXTRUDER_NUM);
      return;
    }
```

thinkyhead, 2016-06-09:
> Ups.

Oops! 😵  Yes, I guess that would be good to add. It can replace the `false` clause at the top.

focamonca, 2016-06-10:
I did't understood how to solve the problem LOL

Blue-Marlin, 2016-06-10:
If you have `PIDTEMPBED` not activated you will get the error. No PID for the bed -> No tuning the PID for the bed.

closed 2016-06-11 by thinkyhead

Anton73rus, 2016-09-22:
**Please correct source:**
if (hotend >=
        #if ENABLED(PIDTEMP)
          HOTENDS
        #else
          0
        #endif
      || hotend <
        #if DISABLED(PIDTEMPBED) //(**It was Enabled**) 
          -1
        #else
          0
        #endif
    ) {
      SERIAL_ECHOLN(MSG_PID_BAD_EXTRUDER_NUM);
      return;
    }

Blue-Marlin, 2016-09-22:
Please rethink,try and close the issue.
-1 is the number of the bed!

Anton73rus, 2016-09-23:
if I'm wrong ? then why after the code has changed - I have gone error : "PID Autotune failed Bad extruder number!", and began autotune? = ) I apologize if the text with errors - I translate through Google translator :D

Blue-Marlin, 2016-09-23:
Likely because you have not activated 'PIDTEMPBED'.

thinkyhead, 2016-09-24:
@Anton73rus

[The code is correct.](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/temperature.cpp#L227)

Do you have the option set…?
```cpp
#define PIDTEMPBED
```

micahobaca, 2019-03-12:
I finally got it to work, you have to heat your bed withing 10 degrees of your target temperature then run the pid autotune, then it will work, as long as you have the pid autotune enabled in your marlin code and that other option below it commented out you are good to go, try that and let me know if it worked for you, if you haven't already figured it out.