MarlinFirmware/Marlin #3999: Compilation error with BEEPER_PIN -1 or #undef BEEPERPIN

esenapaj, 2016-06-10:
After PR #3995(New feature: Non blocking tone queue) was meged, I've gotten compilation errors.
If I set the BEEPER_PIN to -1 for disable beep, it causes compilation error.
[error log.txt](https://github.com/MarlinFirmware/Marlin/files/309369/error.log.txt)

If I undefine the BEEPER_PIN by #undef BEEPERPIN, I got another error.
[error log2.txt](https://github.com/MarlinFirmware/Marlin/files/309378/error.log2.txt)

Settings:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/309382/Configuration.txt)
[pins_RAMPS_14.txt](https://github.com/MarlinFirmware/Marlin/files/309383/pins_RAMPS_14.txt)

jbrazio, 2016-06-10:
Seems an error with `#define HAS_BUZZER`, will check ASAP.

jbrazio, 2016-06-10:
@esenapaj could you please try the following diff ?

```diff
diff --git a/Marlin/ultralcd.h b/Marlin/ultralcd.h
index 5cb1ecc..e855835 100644
--- a/Marlin/ultralcd.h
+++ b/Marlin/ultralcd.h
@@ -27,8 +27,6 @@
 
 #if ENABLED(ULTRA_LCD)
 
-  #include "buzzer.h"
-
   #define BUTTON_EXISTS(BN) (defined(BTN_## BN) && BTN_## BN >= 0)
   #define BUTTON_PRESSED(BN) !READ(BTN_## BN)
```

esenapaj, 2016-06-10:
>could you please try the following diff ?

Thanks for the quick fix.
I've confirmed just now that compilation error is gone by the diff.

commit 2c34ec2b (2016-06-11): Closes #3999: Compilation error with BEEPER_PIN -1
commit 93ffa028 (2016-06-11): Merge pull request #4003 from jbrazio/bugfix/3999

Closes #3999: Compilation error with BEEPER_PIN -1
closed 2016-06-11 by thinkyhead