MarlinFirmware/Marlin #4006: latest BF bf8710d

ruggb, 2016-06-11:
In the interest of uncomplicating my files I have deleted all files that I do not use.
watchdog.h & watchdog.ccp are included in the delete list since watchdog is disabled in my config
The latest rel bf8710d adds things to the buzzer file and #include watchdog.h
But I have watchdog disabled so why isn't this include done with a test as to whether watchdog is enabled???

jbrazio, 2016-06-11:
"uncomplicating" by deleting source files ?..
I'm sorry @ruggb I will close this ticket, but if you want just apply the following patch to your source tree:
```diff
diff --git a/Marlin/buzzer.h b/Marlin/buzzer.h
index 2e43625..85dec2a 100644
--- a/Marlin/buzzer.h
+++ b/Marlin/buzzer.h
@@ -24,7 +24,9 @@
 #define __BUZZER_H__
 
 #include "fastio.h"
+#if ENABLED(USE_WATCHDOG)
 #include "watchdog.h"
+#endif
 #include "circularqueue.h"
 
 #define TONE_QUEUE_LENGTH 4
```

closed 2016-06-11 by jbrazio

ruggb, 2016-06-11:
my fix was to comment out the #include
the post was to bring the issue to your attention.