MarlinFirmware/Marlin #4007: latest build bf8710d

ruggb, 2016-06-11:
the error during compile is:
```
In file included from sketch\Configuration_adv.h:709:0,

                 from sketch\Configuration.h:1184,

                 from D:\Users\Bill\Documents\#Arduino\libraries\Marlin-RCBugFix-bf8710d\Marlin\Marlin.ino:37:

SanityCheck.h:591: error: #error "EXTRUDER_OFFSET_[XY] is deprecated. Use HOTEND_OFFSET_[XY] instead."

   #error "EXTRUDER_OFFSET_[XY] is deprecated. Use HOTEND_OFFSET_[XY] instead."

    ^

exit status 1
#error "EXTRUDER_OFFSET_[XY] is deprecated. Use HOTEND_OFFSET_[XY] instead."
```

however there is no note to that effect where I enabled EXTRUDER_OFFSET and I have no idea where HOTEND_OFFSET is.
Please make changes a little easier on users. thank you.

ruggb, 2016-06-11:
actually, after a search, I find that HOTEND_OFFSET,,,  does not exist
Not a good thing...........

jbrazio, 2016-06-11:
```
jbrazio@dagobah:~/Documents/src/Marlin/Marlin (RCBugFix) $ grep HOTEND_OFFSET *
Conditionals.h:    #undef HOTEND_OFFSET_X
Conditionals.h:    #undef HOTEND_OFFSET_Y
Conditionals.h:    #define HOTEND_OFFSET_X { 0 }
Conditionals.h:    #define HOTEND_OFFSET_Y { 0 }
Configuration_adv.h:      // However: In this mode the HOTEND_OFFSET_X value for the second extruder provides a software
Configuration.h://#define HOTEND_OFFSET_X {0.0, 20.00} // (in mm) for each extruder, offset of the hotend on the X axis
Configuration.h://#define HOTEND_OFFSET_Y {0.0, 5.00}  // (in mm) for each extruder, offset of the hotend on the Y axis
grep: example_configurations: Is a directory
grep: fonts: Is a directory
language.h:#define MSG_HOTEND_OFFSET                   "Hotend offsets:"
Marlin_main.cpp:  #ifndef HOTEND_OFFSET_X
Marlin_main.cpp:    #define HOTEND_OFFSET_X { 0 } // X offsets for each extruder
Marlin_main.cpp:  #ifndef HOTEND_OFFSET_Y
Marlin_main.cpp:    #define HOTEND_OFFSET_Y { 0 } // Y offsets for each extruder
Marlin_main.cpp:    HOTEND_OFFSET_X,
Marlin_main.cpp:    HOTEND_OFFSET_Y
Marlin_main.cpp:    SERIAL_ECHOPGM(MSG_HOTEND_OFFSET);
Marlin_main.cpp:        SERIAL_ECHOPGM(MSG_HOTEND_OFFSET);
SanityCheck.h:  #error "EXTRUDER_OFFSET_[XY] is deprecated. Use HOTEND_OFFSET_[XY] instead."
```

jbrazio, 2016-06-11:
i.e. just rename `EXTRUDER_OFFSET_*` to `HOTEND_OFFSET_*`.

closed 2016-06-11 by jbrazio