MarlinFirmware/Marlin #4008: M105  What if I have different power supplyfrom 12V?

Amachete, 2016-06-11:
Hello

I have a Power Supply for my hot End which has 12V, but my HotBed has a Power supply with 220V.
So the M105 does not give me an accurate information..

I suggest to include in the Configuration.h the following lines:
`// If you want the M105 heater power reported in watts, define the BED_WATTS, and (shared for all extruders) EXTRUDER_WATTS`

`// Voltage is in VOLTS`
`//#define HOTEND_VOLTAGE 12`
`//#define BED_VOLTAGE 220`

`//Resistance in Ohms`
`//define HOTEND_RESISTANCE 6,7`
`//define BED_RESISTANCE 1`

`//#define HOTEND_WATTS (HOTEND_VOLTAGE*HOTEND_VOLTAGE/HOTEND_RESISTANCE) // P=U^2/R`
`//#define BED_WATTS (BED_VOLTAGE*BED_VOLTAGE/BED_RESISTANCE)    // P=U^2/R`

These changes will give more accurate information, and it is not difficult to get the resistance of both Hotend and Bed, with a Voltimeter.

You can add at the end of a Gcode File, a M105code  to get how much power your printer has spend during the last print.

I hope you implement this change!!!

If you want I could submit a pach!!

Thank you all

Blue-Marlin, 2016-06-11:
What's so difficult in
```
// If you want the M105 heater power reported in watts, define the BED_WATTS, and (shared for all extruders) EXTRUDER_WATTS
//#define HOTEND_WATTS (220*220/6.7) // P=U^2/R
//#define BED_WATTS (12.0*12.0/1.1)    // P=U^2/R
```
Ah, yes - mainly to write a decimal '.' instead of a ','.

Finding the correct resistance is a bit more complicated. The hot heaters have a higher resistance than the cold ones.

For the same reason a total power consumption value would be a pure fake. (Or it would need much more work)

thinkyhead, 2016-06-11:
We encourage users to use #define as you have to let the compiler calculate values like this. But we don't need to make this a standard addition to Marlin.

jbrazio, 2016-07-11:
Closing due to inactivity.

closed 2016-07-11 by jbrazio