MarlinFirmware/Marlin #4009: Crash after setting Flowrate

Massa0815, 2016-06-11:
When i adjust the Flowrate in the menu during printing sometimes Marlin crashes completely. In my special case i changed the flowrate from 85% to 90%

I installed RC Bugfix #3110

thinkyhead, 2016-06-11:
Now that #3110 is merged, you can just test with the current `RCBugFix`.

I'm not sure how we can narrow this down. Can you try doing some other things in the menus, such as value editing, and see if anything else causes a crash?

Massa0815, 2016-06-13:
Ok. Give me a few days. I disassambled my Hetabed for a new cable chain. When i reassambled it i will test it. Thanks

Massa0815, 2016-06-20:
I tested the actual bugfix. The bug is still present. I think its only the flowrate wich produces the crash. It occurs when i confirm the changed flowrate. The display freezes. But there are still some moves the printer performs until complete crash.

thinkyhead, 2016-06-20:
I can find nothing wrong with the menu function that sets the flow rate in particular. So perhaps it has something to do with the function to print the value, or the display update.

closed 2016-07-11 by jbrazio