MarlinFirmware/Marlin #4016: Move E axis for second tool moves first tool

Alex9779, 2016-06-12:
I tried RCBugFix today and realised that moving the E axis with the LCD always moves the E axis of the current selected tool regardless the menu entry (Extruder 1 or Extruder 2) I choose.

To reproduce:

- I power on my machine.
- I heat up the hotends.
- I go to "Prepare"->"Move axis"->"Move 1mm"->"Extruder1" and move --> Extruder 1 moves.
- I go to "Prepare"->"Move axis"->"Move 1mm"->"Extruder2" and move --> Extruder 1 moves (!).
- I send T1 to the machine.
- I go to "Prepare"->"Move axis"->"Move 1mm"->"Extruder1" and move --> Extruder 2 moves (!).
- I go to "Prepare"->"Move axis"->"Move 1mm"->"Extruder2" and move --> Extruder 2 moves.

Alex9779, 2016-06-12:
BTW regarding #4000 if moving an E axis with the LCD the head should not move. No idea how the LCD call works but having tool 0 active and moving E1 with the LCD must not move the head and vice versa...

thinkyhead, 2016-06-12:
> if moving an E axis with the LCD the head should not move

Does it do that now? It should just move E and nothing else, assuming current position is correct.

Alex9779, 2016-06-13:
> Does it do that now?

I just wrote that I don't know how the code to move from LCD works and because the head didn't move until that other fix I have no idea if it would now IF it would move the correct extruder.

I will test newest RCBugFix and report...

Alex9779, 2016-06-15:
Moving E axes works again from the LCD with the fix from #4047...

closed 2016-06-16 by thinkyhead