MarlinFirmware/Marlin #4018: Question:  Why are we not using run_z_probe() for the Z-Axis in G28 ?

Roxy-3DPrintBoard, 2016-06-12:
Is there a reason we are not using run_z_probe() to handle the Z-Axis when we have Bed Leveling (when we know we have a Z-Probe) ?    It seems like we have a bunch of redundant code because of that.  We have to have the probe_pt() code.  

So shouldn't we end up in this same code if we are doing a G28 Z ????

thinkyhead, 2016-06-12:
The `run_z_probe` function is shared code. The `probe_pt` function calls it to do a single probe (so `gcode_G29` uses it). And the `gcode_G30` function also calls it.

That isn't to say that the code in `gcode_G29` and `gcode_G30` are ideal, or that the probing functions are as clean as they could be. For the most part all these functions were adapted from code that already existed.

Ideally you'd want a probe-management class that encapsulates each common step in the probing process, and then adapts each of those putative steps to the particular probe and endstops that you have.

thinkyhead, 2016-06-12:
> shouldn't we end up in this same code if we are doing a G28 Z?

Possibly. We should see where the special-cases lie in `G28 Z` when the Z probe is being used as the Z endstop.

Roxy-3DPrintBoard, 2016-06-12:
The reason I'm asking is run_z_probe()  and probe_pt() were behaving correctly.  But G28 was not doing a Z_RAISE_AFTER_PROBE.   My probe was OK just because of the direction the roller was on the micro-switch.   But I ended up having to add a raise_z_after_probing(); in to homeaxis() to fix the problem:
~~~~
    // Retract Servo endstop if enabled
    #if HAS_SERVO_ENDSTOPS
      if (_Z_SERVO_TEST && servo_endstop_id[axis] >= 0) {
        #if ENABLED(DEBUG_LEVELING_FEATURE)
          if (DEBUGGING(LEVELING)) SERIAL_ECHOLNPGM("> SERVO_ENDSTOPS > Stow with servo.move()");
        #endif
  	raise_z_after_probing();

        servo[servo_endstop_id[axis]].move(servo_endstop_angle[axis][1]);
        if (_Z_PROBE_SUBTEST) endstops.enable_z_probe(false);
      }
    #endif
~~~~

thinkyhead, 2016-06-12:
Looks like you could use `raise_z_for_servo()` instead. That's what other things use when not `G29` probing.

Roxy-3DPrintBoard, 2016-06-12:
OK.  I'll switch to raise_z_for_servo().    But do you agree this is a bug?  I can't be the only person seeing this behavior, right?

thinkyhead, 2016-06-12:
Yes, the homing, probing, stowing, deploying, etc. code as I scan through it is spaghetti.

I suggest being methodical and consider encapsulating each little step, at least as an exercise. Or maybe your UBL code will save us from all of this!

Roxy-3DPrintBoard, 2016-06-12:
> Yes, the homing, probing, stowing, deploying, etc. code as I scan through it is spaghetti.

To be honest...  That is why I was asking the question "Why are we not using run_z_probe() for the G28?"   I was hoping we could just delete all that spaghetti code in G28 and get rid of #ifdef's nested 13 layers deep.   As a side benefit, both G28 and G29 would start agreeing on where the Z-Axis was positioned!

> I suggest being methodical and consider encapsulating each little step, at least as an exercise. Or maybe your UBL code will save us from all of this!

No...  The Unified Bed Leveling won't directly clean this up.  It sits on top of the Z-Probe functions.   But when we do get a Z-Probe object, it should slip in place without any pain.   And a whole bunch more code with nested #ifdef's will disappear!   :)

thinkyhead, 2016-06-12:
Hmm. Well, I get this feeling like, if we just took each segment of each of these processes, and placed them into smaller functions, then after doing this a while they would start to merge and converge, and we'll end up with something minimal. It's a roundabout (or maybe straightforward) way of accomplishing both analysis and consolidation at the same time.

thinkyhead, 2016-06-12:
I think the main complication to doing this (up to now) is just the special-cases, and in a few spots, not quite understanding what is necessary vs what is provisional (for example, when measuring Z heights in a probe process, we should preserve "known position" so Z is always true).

Roxy-3DPrintBoard, 2016-06-12:
I agree that approach would work.   But we have the Z-Probe routines working and doing the right thing.   I wonder if it would be easier to just delete all the Z-Axis code in G28 that doesn't relate to a simple endstop and replace it with a probe_pt() ????

thinkyhead, 2016-06-12:
You can try that, and it might work on your robot, but then we don't know if it's good across all robots without more testing.

Roxy-3DPrintBoard, 2016-06-12:
Agreed!    This is such a drastic change, we probably don't even want to think about it in an RC branch.

thinkyhead, 2016-06-12:
Truly. Whereas, moving snippets out of a big function into smaller functions can be accomplished without any disruption, as part of the "general cleanup" stuff that we more or less always do along the way.

thinkyhead, 2016-06-12:
Before diving into too much of that, a good first step is to just do more liberal code-commenting and documentation throughout these functions. Explaining the rationale for each piece of logic. Out of that we will gain a better high-level view of this spaghetti monster.

thinkyhead, 2016-06-12:
Not too fond of that `raise_z_for_servo()` function. It's not as smart as it should be.

Roxy-3DPrintBoard, 2016-06-12:
I can switch back to raise_z_after_probing();    After all...  That is really what needs to happen in that part of the G28 code.

thinkyhead, 2016-06-12:
At one point I proposed replacing all the "raise z" values with a single `Z_PROBE_CLEARANCE` value. This is the space the probe needs to deploy/stow, or the height needed to keep an inductive probe from triggering early.

Roxy-3DPrintBoard, 2016-06-12:
Yeah...  The problem with the Z_RAISE numbers is we don't always want to raise the Z-Axis.  What we want is a minimal height of that number.  At a minimum, we should consider renaming those #defines.   

But I do think some people need the Z_RAISE_BETWEEN number just because the amount of clearance needed to deploy and stow is different than what you need to run the nozzle (probe) around on the bed.    And given how slow my Z-Axis is, I want the control to let that be different and smaller than the BEFORE and AFTER number.

thinkyhead, 2016-06-12:
We also talked about opting for "minimum height" rather than in terms of "raise" – which might tend to simplify things a little.

thinkyhead, 2016-06-12:
Well, this short conversation could easily lead to a whole day of tweaking, haha. But I should comb through my other 20 notifications first, see if there are any little things to fix first….

Roxy-3DPrintBoard, 2016-06-12:
No hurry on any of this...  Other than maybe you have 21 notifications and not 20 if you believe that I stumbled on a bug.

thinkyhead, 2016-06-12:
I think you stumbled on a weird oversight that might be in a few places. Surely we have lots of servo users. Maybe they all figured out workarounds in their settings, and when we fix this it will mess them up, haha!

thinkyhead, 2016-06-12:
See what you think of #4021

jstefanop, 2016-06-14:
Does this have anything to do with why G92 does not work (to set the Z offset of the prob in relation to nozzle after it does a G28 and G29) in RC6?

Roxy-3DPrintBoard, 2016-06-14:
G92 doesn't work any more?   Can you tell us which RC\# it quit working at?

jstefanop, 2016-06-14:
It was working on RC3 and tried out RC6, and normal G92 Z1.3; command I have been using for nozzle offset no longer has an effect. The machine has a Z-probe for both endstop and ABL if that makes a difference.

thinkyhead, 2016-06-15:
G92 is not the way to do this. It not only sets your current position, but it also moves the coordinate space (the software endstops). You will have to use the probe z offset (`M851`) or the home offsets (`M206`) instead.

jbrazio, 2016-07-18:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

PS: Lately huge development has been put into G28, ABL and MBL.

closed 2016-07-18 by jbrazio