MarlinFirmware/Marlin #4019: Mesh Bed Leveling loses state randomly

orcinus, 2016-06-12:
Apologies if this has already been reported or fixed (couldn't find it).
On RC5, MBL sometimes loses its setting completely.

As in, on power-on, the printer will home, then *won't* bring the Z up to the calibrated MBL value, and will not do MBL corrections during printing.

Don't ask me to reproduce it, i can't. It's utterly random.
Reloading settings and/or restarting fixes it, until the next time it happens.

epatel, 2016-06-12:
This has probably been fixed. See this https://github.com/MarlinFirmware/Marlin/issues/3750

orcinus, 2016-06-12:
Ah! Sorry, somehow missed that issue.
FWIW, i don't have G28 in my end g-code, but i *did* issue a G28 manually today, so that makes perfect sense.

closed 2016-06-12 by jbrazio