MarlinFirmware/Marlin #4022: Feature request - allow persistent z-offset change after babystepping.

DjDemonD, 2016-06-12:
So I measure my z offset, then enter m851 zx where x is the nozzle to bed distance after homing. If when I begin to print my nozzle is slightly too close/far from the bed I babystep it to the correct level (the gap changes due likely to thermal expansion in the hotend etc..).

Can this offset modification by babystepping be automatically added to the m851 offset, perhaps at the end of the print job to avoid the effetc being doubled, so that on the next print job my z height is correct, and I do not have to babystep it again. I can then do m500 and save this if I am happy with it. I understand I can manually alter the m851 offset afterwards for subsequent prints, but I have to guess as the babystepping does not change the z coordinates.

Apologies if something like this already in the pipeline.

thinkyhead, 2016-06-13:
Babystepping is only available during active printing, so far as I know. So it sounds like what you really want is just a way to adjust your Z probe offset from the LCD, while moving the axis up and down.

thinkyhead, 2016-06-13:
Babystepping doesn't keep track of the distance moved, and is applied in step units rather than millimeters, so this would be a minor overhaul. Of course, this would only apply to machines with a Z probe. Do you find that the distance from nozzle tip to the probe changes a lot for you?

DjDemonD, 2016-06-13:
Why would it only apply to machines with z Probes? If I manually level z then begin printing and I'm slightly too high to get a good first layer, I'd babystep it down a little. What I then need is for this to become the new z=0 position in the next print. Or if this is a leap too far just to have some output to know how far it moved so I can manually adjust the offset for the next print. As you say it's in steps so easy enough to output z changed by - x mm. Babystepping is really good it saves a lot of time not having to abandon the print, adjust offset or nozzle height and then reheat and try again, but it's only realising half its potential, rather than see it as a quick fix for nudging something back into alignment why not see it as the method by which a user sets their z offset, once they have a level bed.

thinkyhead, 2016-06-14:
Sounds like a fine idea. But how shall we implement it? Just a single menu item… "`Apply Babysteps Offset`". To keep it persistent it would simply be applied to the `home_offset` – the same as `M206`. This offset already adjusts the coordinate space persistently, so it would be easy to just add in the babysteps converted to millimeters (for DELTA and other custom kinematics we have to be cautious about these offsets – DELTA can only do Z offset, for example).

DjDemonD, 2016-06-14:
That would work for me and I think,  it would be a really great feature.

adamfilip, 2016-06-14:
how would this work with mesh levelling?

thinkyhead, 2016-06-15:
@adamfilip It would offset the print position by the amount adjusted, regardless of your leveling settings.

sarf2k4, 2016-06-16:
@DjDemonD it looked like that you have a bit of trouble on each of your print on the z-axis, if you have the need to keep adjusting babystepping on each print then you should try to change the z offset from the control>motion menu there by about 0.05mm or so then test them at the next print.

Don't forget to use the "save settings" at the "controls" menu there to save the settings changed into the eeprom. On my side here a continuous print right after the last one would sometimes require me to use babystepping a bit but most of the time I don't have to meddle with babysteppings. 

If this feature are to be implemented however, hopefully that during babystepping adjustment I didn't accidentally applied the babystepping to z-offset though

DjDemonD, 2016-06-16:
I am not having any problems with the consistency of the z-height/offset, its just I have made a lot of changes to the printer lately, each change requires this tuning to be performed. Before babystepping came along the process of measuring the z-offset entering that using m851 or worse having to re-upload the modified firmware, printing a calibration object, not being happy with the z-height on the first layer, going back and doing it all again, was very time consuming. Now I can measure z offset, enter it using m851, babystep to the perfect z height whilst I print a skirt around my object. But then I do not know what the change in offset has been. So I have to make a guess, change the offset and print again. We do not need to guess, the firmware can tell us by how much we moved the axis (with this change implemented), we can then either manually or automatically use that data to fine tune our printer quickly and easily. What's not to love?

Why would it be a problem for you if you accidentally applied the babystep offset to your z-offset. If your first layer height is not right and you correct it, why would you not want that correction to be persistent? Perhaps if you babystep later during a print, maybe try to get a bit better top surface or correct for some issue such as a slight shift after a filament change? I am referring only to use of babystepping for first layer improvement.

DjDemonD, 2016-06-17:
Thinking further on this it might be nice for the Lcd menu to automatically switch to babystepping as printing starts after heating, homing and autolevelling complete if babystepping is defined. We could then adjust z if required, then a click of the button would go back to the info screen perhaps with an option to make the change persistent (with no as default option), cover this with a 5 second timeout switching back to the info screen for those who have no wish to babystep. Of course this functionality would be useful to all who define it and would not irritate those who don't define it (those with very precise printers who calibrate/level once per year). Also this approach would introduce the feature to those who haven't tried it and don't know it exists. I say this as for a small print it takes time to find the babystepping menu option by which point I've already printed a fair bit of skirt or am maybe even printing the object before meaningful adjustment can be made.  There is a reprap forum thread about assigning the  babystepping function to two buttons, this is probably too much as it is an adjustment that wouldn't need to be made that often.

superlou, 2016-06-17:
Would this overlap/achieve the functionality requested in #4061?

Roxy-3D, 2016-06-17:
> Thinking further on this it might be nice for the Lcd menu to automatically switch to babystepping as printing starts after heating, homing and autolevelling complete if babystepping is defined. We could then adjust z if required

Are you saying you want the default encoder wheel action to be babystepping instead of feedrate?
Or are you saying you want the LCD Menu to automatically go to the BabyStepping menu once it starts printing?   I think I would prefer it went to the menu instead of taking priority over the feedrate adjustment.   But I would also like the menu to time out after a few layers and go back to the normal status display.  The reason is after 3 or 4 hours of printing, I don't want the babystep menu active where I can accidentally bump the encoder wheel and mess up the print.

DjDemonD, 2016-06-17:
No definitely not taking over the feed rate on the encoder wheel. I'm suggesting exactly the second case, printing starts the menu switches to babystepping, with a short time out and then back to the info screen. But I am really evangelising for the idea of being able to retain the babystepped offset for subsequent prints and be able to save it to eeprom.

DjDemonD, 2016-06-17:
Regarding #4061 I am not sure these are related. There must be a way to set manual z home position by gcode. Rich Cattel's fork does this when autocalibrating. I'll take a look on my delta I a minute I've got to calibrate it anyway.

thinkyhead, 2016-06-18:
> retain the babystepped offset for subsequent prints and be able to save it to eeprom

Now that it displays the offset in mm you can just use `M206` to save the value. Perhaps later we'll add an option to save the babystep offset as the `M206` offset. It's not too hard to implement. But I'm not sure this will end up being a standard feature.

DjDemonD, 2016-06-18:
Thank you that's going to be really useful. Will download latest bugfix and test.

DjDemonD, 2016-06-18:
Just flashed the latest bugfix from this morning, with the babysteps reading out in mm. Very pleased -that enables exactly the functionality I need.

closed 2016-06-18 by DjDemonD

gordo3di, 2018-01-12:
We **love** seeing the babystep value as mm! I think it would be very helpful if it were possible to apply this offset automatically without having to go to the control > z offset menu item, calculating the change, applying the change then saving it.

What we tell people now in our manual is to do the steps outlined above. Is there a way to recall the babystep value and run a custom gcode script to apply this change? Basically take the old offset value (ie. -1.9mm) add the babystep value (ie -1.9 + 0.23 = -1.67) then save the result as the new offset?

We find we have to babystep the first layer every so often especially if the first layer heights change per print.

thinkyhead, 2018-01-22:
> Is there a way to recall the babystep value and run a custom gcode script to apply this change?

Use `BABYSTEP_ZPROBE_OFFSET` and babystepping will be applied to `zprobe_zoffset`, which can then be saved with `M500`.

gordo3di, 2018-01-22:
@thinkyhead Thanks Scott. This new firmware is amazing.