MarlinFirmware/Marlin #4026: Marlin end-user survey results

jbrazio, 2016-06-12:
These are the results of the survey we did which had a total of 372 answers.
PS: If you'd like other views on the data just leave a comment.

--

[image]

[image]

[image]

[image]

bobc, 2016-06-16:
Does that mean most users (who responded) are using Arduino Mega + RAMPS and majority of users are running IDE 1.6.x?

Either way, interesting results.

fbarcenas, 2016-06-23:
Really interesting...

closed 2016-07-16 by jbrazio